MichaIng/DietPi #777: DietPi - Requests Internet access during installation when internet connection is functional

LexiconCode, 2017-02-24:
Dietpi: 144
Odroid C2

During the installation process DietPi Internet access during installation when internet connection is functional.

"Run the Internet Connection Test"  it report  "Internet     : Online "

Additionally verified by...
```
root@test:~# wget -O /dev/null http://speedtest.wdc01.softlayer.com/downloads/test10.zip
--2017-02-24 16:49:17--  http://speedtest.wdc01.softlayer.com/downloads/test10.zip
Resolving speedtest.wdc01.softlayer.com (speedtest.wdc01.softlayer.com)... 208.43.102.250, 2607:f0d0:3001:78::2
Connecting to speedtest.wdc01.softlayer.com (speedtest.wdc01.softlayer.com)|208.43.102.250|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 11536384 (11M) [application/zip]
Saving to: ‘/dev/null’

/dev/null                         100%[===============================================================>]  11.00M   437KB/s   in 13s

2017-02-24 16:49:31 (847 KB/s) - ‘/dev/null’ saved [11536384/11536384]
```
Initially I thought this may be related to Pihole

 Current Details:                                            
                                   │  - Mode    = STATIC                                           
                                   │  - Ip      = 192.168.1.118                                    
                                   │  - Mask    = 255.255.255.0                                   
                                   │  - Gateway = 192.168.1.1                                  
                                   │  - DNS     = 127.0.0.1  

```
root@test:~# ifconfig
eth0      Link encap:Ethernet  HWaddr 00:1e:06:33:9b:41
          inet addr:192.168.1.118  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::21e:6ff:fe33:9b41/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:9595 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3544 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:12335084 (11.7 MiB)  TX bytes:443761 (433.3 KiB)
          Interrupt:40

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:4096  Metric:1
          RX packets:306 errors:0 dropped:0 overruns:0 frame:0
          TX packets:306 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:27133 (26.4 KiB)  TX bytes:27133 (26.4 KiB)
```

Fourdee, 2017-02-24:
@LexiconCode 

Thanks for the report, this is a known issue:
https://github.com/Fourdee/DietPi/issues/755#issuecomment-282203823

We'll get this fixed for v145, in the mean time, quick solution is here:
http://dietpi.com/phpbb/viewtopic.php?f=11&t=1373#p5810

I'll mark this as closed (dupe)

closed 2017-02-24 by Fourdee

LexiconCode, 2017-02-24:
@Fourdee 
The suggested fix may cause the following error when running 'apt-get update'

W: GPG error: http://www.deb-multimedia.org jessie InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 5C808C2B65558117

Which can be fixed by
apt-get install deb-multimedia-keyring

Fourdee, 2017-02-24:
@LexiconCode 

> W: GPG error: http://www.deb-multimedia.org jessie InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 5C808C2B65558117

Yep, this doesn't exist on our image, it appears its pulled in during ```apt-upgrade``` possibly from one of Meveric's packages:
[image]

LexiconCode, 2017-02-24:
@Fourdee
Strange, So I initially tested the suggested fix. Via 'apt-get update' and 'apt-get 'upgrade' which everything works as expected after a reboot.

But when I go to install software such as Nextcloud.
 ```
Mode: Update & upgrade APT
 Please wait...

Err http://security.debian.org jessie/updates InRelease

Err http://ftp.debian.org jessie InRelease

Err http://ftp.debian.org jessie-updates InRelease

Err http://ftp.debian.org jessie-backports InRelease

Err http://www.deb-multimedia.org jessie InRelease

Err http://oph.mdrjr.net all InRelease

Err http://oph.mdrjr.net jessie InRelease

Err http://ftp.debian.org jessie Release.gpg
  Temporary failure resolving 'ftp.debian.org'
Err http://www.deb-multimedia.org jessie Release.gpg
  Temporary failure resolving 'www.deb-multimedia.org'
Err http://ftp.debian.org jessie-updates Release.gpg
  Temporary failure resolving 'ftp.debian.org'
Err http://oph.mdrjr.net all Release.gpg
  Temporary failure resolving 'oph.mdrjr.net'
Err http://security.debian.org jessie/updates Release.gpg
  Temporary failure resolving 'security.debian.org'
Err http://ftp.debian.org jessie-backports Release.gpg
  Temporary failure resolving 'ftp.debian.org'
Err http://oph.mdrjr.net jessie Release.gpg
  Temporary failure resolving 'oph.mdrjr.net'
Reading package lists... 4%
```
```
Unable to Install NextCloud: your very own cloud (eg: dropbox)     
 https://download.nextcloud.com/server/releases/nextcloud-11.0.0.zip is offline and/or unreachable.                                   
```

LexiconCode, 2017-02-24:
@Fourdee I updated to reflect the full output.

Fourdee, 2017-02-25:
@LexiconCode 

> Fyi: nextcloud-11.0.1.zip is available

Thanks 👍 

Lets use: 
https://download.nextcloud.com/server/releases/latest-11.zip