MichaIng/DietPi #778: Update all images to v145

Fourdee, 2017-02-25:


Fourdee, 2017-02-26:
OPi Zero board failure. No boot, serial, leds. Unable to complete.

taner1, 2017-02-27:
will this update rolls with 4.10 mainline kernel for opi zero? im waiting for it

Fourdee, 2017-02-27:
@taner1 

Unfortunately not from my end as my board is dead. Would need someone else to generate image while I wait for replacement.

Fourdee, 2017-02-27:
@taner1 

Although saying that, ARMbian may release 4.10 as a package install via their repo.

taner1, 2017-02-27:
I tried to install it via apt but my goal is getting wifi ap+sta mode so will this update also install mainline ported xradio module or does it conflicts?  I cant figure it out because kernel headers filled my 1gb disk.  btw how can i delete old kernel files i tried cleaners and apt autoremove clean things but nothing done :)

Fourdee, 2017-02-28:
@taner1 

> how can i delete old kernel files i tried cleaners

Manually is best way. 
- Remove the oldest version from (if 2 or more entries):
```
ls -lha /lib/modules
```

On some images, also check boot for duplicate files with different versions:
```
ls -lha /boot
```

> wifi ap+sta mode so will this update also install mainline ported xradio module or does it conflicts?

If you are modifying the kernel, you may wish to put the ARMbian packages on hold. This would allow you to used the ported xradio_wlan driver without kernel being updated.

Not sure which package exactly, but you can list them with.
```
dpkg -l | grep linux
```

Then to hold:
```
apt-mark hold thispackageplease
```

Fourdee, 2017-02-28:
Although DEV (4.9) offered excellent results with Nano Pi Neo's, the recent 4.10 update via ARMbian repo renders it unusable.

Making the decision to revert all DEV (4.x mainline) images to STABLE (3.x). We may offer DEV (4.x) as an alternative, but not for our main image.

Fourdee, 2017-03-02:
Completed

Fourdee, 2017-03-10:
- OPi Zero arrived, rebuild and update image

Fourdee, 2017-03-11:
https://i.imgur.com/KpeIJP0.gifv lol

Fourdee, 2017-03-11:
OPi Zero v145 image live: http://dietpi.com/downloads/images/DietPi_OrangePiZero-armv7-(Jessie).7z

closed 2017-03-11 by Fourdee