MichaIng/DietPi #780: Pine64 | Incorrectly reporting No internet connection

midwan, 2017-02-25:
Fresh install of v142, updated to v144 on a Pine64.
I've enabled both Ethernet and Wifi, the Interent Test in the Adapters config screen works fine.
Apt-get works fine as well, and I was able to update the firmware from the menu also.

However, Dietpi-Software reports "No internet connection" when it attempts to start installing my selected applications, and refuses to continue.

Is this a known issue? Any workarounds?

midwan, 2017-02-25:
I guess this is due to #755 then ?

midwan, 2017-02-25:
Confirmed, the fix mentioned helped resolve the issue. Sorry :)

closed 2017-02-25 by midwan

Fourdee, 2017-02-26:
@midwan 

Thanks for report, your spot on 👍 

Aiming to have the Pine A64 image updated today, with the fix applied:
https://github.com/Fourdee/DietPi/issues/778

Fourdee, 2017-02-26:
@midwan 

Pine image updated:
http://dietpi.com/downloads/images/DietPi_PineA64-arm64-(Jessie).7z

Contains fix for this ticket, and latest firmware/kernel from longsleep (104)

midwan, 2017-02-26:
Awesome, thanks!