MichaIng/DietPi #783: Emby Server

were55, 2017-02-28:
Hello, I'm trying to install Emby server on RPi 2 but I have some problems, I tried different sources with the same error:
Apt-get has failed:
- Apt string = emby-server embymagick 
- Error code = 100 
- Software Title = Emby Server: web interface media streaming server

Fourdee, 2017-02-28:
@were55 

Hi,

Thanks for the report 👍 

I'll try to replicate.

Fourdee, 2017-02-28:
Unable to replicate on fresh install.

@were55 

Issue may be limited to your current installation/package combination. We'll need your assistance to debug this manually to find a solution.

Please can you run the following commands, and paste results of the errors (or take a picture):
```
apt-get install emby-server embymagick
```

were55, 2017-02-28:
Hi Fourdee,

thanks for reviewing that. Here is the output:

root@DietPi:~# apt-get install emby-server embymagick
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 emby-server : Depends: mono-runtime (>= 3.0~) but it is not installable
               Depends: libmono-corlib4.5-cil (>= 4.0.0~alpha1) but it is not going to be installed
               Depends: libmono-microsoft-csharp4.0-cil (>= 1.0) but it is not going to be installed
               Depends: libmono-posix4.0-cil (>= 4.0.0~alpha1) but it is not going to be installed
               Depends: libmono-system-configuration4.0-cil (>= 4.0.0~alpha1) but it is not going to be installed
               Depends: libmono-system-core4.0-cil (>= 4.0.0~alpha1) but it is not going to be installed
               Depends: libmono-system-data4.0-cil (>= 4.0.0~alpha1) but it is not going to be installed
               Depends: libmono-system-drawing4.0-cil (>= 3.0.6) but it is not going to be installed
               Depends: libmono-system-io-compression4.0-cil (>= 3.2.1) but it is not going to be installed
               Depends: libmono-system-net-http4.0-cil (>= 1.0) but it is not going to be installed
               Depends: libmono-system-runtime-serialization4.0-cil (>= 4.0.0~alpha1) but it is not going to be installed
               Depends: libmono-system-servicemodel4.0a-cil (>= 3.2.3) but it is not going to be installed
               Depends: libmono-system-transactions4.0-cil (>= 1.0) but it is not going to be installed
               Depends: libmono-system-xml4.0-cil (>= 3.12.0) but it is not going to be installed
               Depends: libmono-system4.0-cil (>= 4.0.0~alpha1) but it is not going to be installed
               Depends: mono-devel (>= 4.6) but it is not installable
E: Unable to correct problems, you have held broken packages.
root@DietPi:~#

Maybe the problem is related to ""TVheadend" installation that I performed before. I followed the next https://tvheadend.org/boards/5/topics/21528?r=23457#message-23457 . 

Regards

Fourdee, 2017-02-28:
@were55 

> Maybe the problem is related to ""TVheadend" installation that I performed before. I followed the next https://tvheadend.org/boards/5/topics/21528?r=23457#message-23457 .

Yep, possibly repo your using for install of TVheadend is effecting this install.

Ok, we can try to modify the repos and see if we get success with install of Emby.

First things 1st, backup your system and userdata
```
dietpi-backup
```

We'll need to see your apt sources next:
```
ls -lha /etc/apt/sources.list.d
cat /etc/apt/sources.list
```

were55, 2017-02-28:
Fourdee,

Sorry for delay, I was out. Here is the output:

root@DietPi:~# ls -lha /etc/apt/sources.list.d
total 20K
drwxr-xr-x 2 root root 4.0K Feb 28 09:02 .
drwxr-xr-x 6 root root 4.0K Feb 27 17:38 ..
-rw-r--r-- 1 root root   74 Feb 28 09:02 emby-server.list
-rw-r--r-- 1 root root   59 Feb 28 09:02 mono-xamarin.list
-rw-r--r-- 1 root root  193 Nov 25 11:55 raspi.list
root@DietPi:~# cat /etc/apt/sources.list
deb http://mirror.umd.edu/raspbian/raspbian jessie main contrib non-free rpi

Fourdee, 2017-03-01:
@were55 

Thanks

> emby-server : Depends: mono-runtime (>= 3.0~) but it is not installable
Depends: mono-devel (>= 4.6) but it is not installable

Ran another test with ```deb http://mirror.umd.edu/raspbian/raspbian jessie main contrib non-free rpi``` as per yours, still installs fine.

Ok, lets check contents of:
```
cat /etc/apt/sources.list.d/raspi.list
cat /etc/apt/sources.list.d/emby-server.list
cat /etc/apt/sources.list.d/mono-xamarin.list
```

And, do you receive any errors when updating apt?
```
apt-get update
```

Lets also check if those packages exist in apt-cache after the update:
```
apt-cache search emby-server
```

were55, 2017-03-01:
Fourdee,

apt-get update runs fine, here is the output of the others commands:

root@DietPi:~# cat /etc/apt/sources.list.d/raspi.list
deb http://archive.raspberrypi.org/debian/ jessie main ui
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://archive.raspberrypi.org/debian/ jessie main ui
root@DietPi:~# cat /etc/apt/sources.list.d/emby-server.list
deb http://download.opensuse.org/repositories/home:/emby/xUbuntu_14.04/ /
root@DietPi:~# cat /etc/apt/sources.list.d/mono-xamarin.list
deb http://download.mono-project.com/repo/debian beta main
root@DietPi:~# apt-cache search emby-server
emby-server - Emby Server is a home media server.
emby-server-beta - Emby Server is a home media server.
emby-server-dev - Emby Server is a home media server.
libembysqlite3-0 - SQLite 3 emby-server library

thanks

Fourdee, 2017-03-01:
@were55 

> root@DietPi:~# apt-cache search emby-server
emby-server - Emby Server is a home media server.
emby-server-beta - Emby Server is a home media server.
emby-server-dev - Emby Server is a home media server.

Looks fine, packages exist, should install, lets try reinstall of Emby:
```
dietpi-software reinstall 41
```

were55, 2017-03-01:
@Fourdee 

output:

root@DietPi:~# dietpi-software reinstall 41
 [Info] Initializing database, please wait...
 [Info] Reading database, please wait...

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Automated reinstall
 Please wait...

 [Info] 41: Emby Server is not currently installed
 [Info] The program must be installed, before reinstall can be used
 [Ok] No changes applied for: Emby Server

regards

Fourdee, 2017-03-01:
@were55 

😃 
```
dietpi-software install 41
```

were55, 2017-03-01:
@Fourdee 

Is the same, (I don't remember we changed something to expect anotehr output, maybe I missed something):

`(Reading database ... 38168 files and directories currently installed.)
Preparing to unpack libjpeg8.deb ...
Unpacking libjpeg8:armhf (8d1-2) over (8d1-2) ...
Setting up libjpeg8:armhf (8d1-2) ...
Processing triggers for libc-bin (2.19-18+deb8u7) ...
 [Info] Apt-get is processing, please wait...

E: Unable to correct problems, you have held broken packages.
 [Failed] The apt cache may be corrupt, apt mirror offline, or you have held broken pacakages. DietPi-Software will now exit.

 [Info] Please try:
 [Info] - Fixing apt package errors with 'apt-get install -f'
 [Info] - Change apt mirror: 'dietpi-config' > 'Networking: NAS/MISC' > 'Apt Mirror'
 [Info] - Clear and refresh apt 'apt-get update'
 [Info] - If problems persist, please create a Git ticket so DietPi can investigate the issue:
 - https://github.com/Fourdee/DietPi/issues
`

Fourdee, 2017-03-01:
@were55 

> Is the same, (I don't remember we changed something to expect anotehr output, maybe I missed something):

Nope we didnt, everything is in order as far as I can see, just want to see if something else is causing this to fail.

> E: Unable to correct problems, you have held broken packages.

Run and paste results of:
```
apt-get install -f
```

were55, 2017-03-01:
@Fourdee 

output:

root@DietPi:~# apt-get install -f
Reading package lists... Done
Building dependency tree
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 5 not upgraded.

Fourdee, 2017-03-02:
@were55 

Thanks.

> emby-server : Depends: mono-runtime (>= 3.0~) but it is not installable

Really strange, everything is setup correctly as far as I can see.

> root@DietPi:~# cat /etc/apt/sources.list.d/mono-xamarin.list
deb http://download.mono-project.com/repo/debian beta main

So, issue appears to be mono repo, complaining packages are not available for install, yet, repo is active on your system.

Silly question, but, can you access http://download.mono-project.com/repo/debian on your network?

And also please paste results:
```
apt-cache search mono-runtime
```

If this still fails on your current installation, i believe a remote SSH session would be required to allow me to debug this further.

were55, 2017-03-02:
@Fourdee 

I have access to repo inside my network. The result of apt-cache search mono-runtime is nothing :-(

Let me know when you can take a look to allow you access.

thanks

Fourdee, 2017-03-04:
@were55 

> Let me know when you can take a look to allow you access.

If you can setup external SSH access to the device in question, then send the details to:
daniel.knight@dietpi.com

I should be able to take a look today/tomorrow.

Fourdee, 2017-03-08:
Notes:
- All ```mono-*``` packages are not available via apt, yet, should be.
- Apt sources are spot on.
- I had to manually pull in the required ```mono-*``` packages from http://download.mono-project.com/repo/debian/pool/main/m/mono/ and ```dpkg -i *.deb```
```
root@DietPi:~/test# ls -lha
total 22M
drwxr-xr-x  2 root root 4.0K Mar  8 10:19 .
drwxr-xr-x 13 root root 4.0K Mar  8 10:17 ..
-rw-r--r--  1 root root  28K Feb 22 12:59 mono-4.0-gac_4.8.0.495-0xamarin1_all.deb
-rw-r--r--  1 root root  18M Feb 22 12:59 mono-devel_4.8.0.495-0xamarin1_all.deb
-rw-r--r--  1 root root  23K Feb 22 12:59 mono-gac_4.8.0.495-0xamarin1_all.deb
-rw-r--r--  1 root root 656K Feb 22 12:59 mono-mcs_4.8.0.495-0xamarin1_all.deb
-rw-r--r--  1 root root  19K Feb 22 12:59 mono-runtime_4.8.0.495-0xamarin1_armhf.deb
-rw-r--r--  1 root root 319K Feb 22 12:59 mono-runtime-common_4.8.0.495-0xamarin1_armhf.deb
-rw-r--r--  1 root root 1.3M Feb 22 12:59 mono-runtime-sgen_4.8.0.495-0xamarin1_armhf.deb
-rw-r--r--  1 root root 964K Feb 22 12:59 mono-xbuild_4.8.0.495-0xamarin1_all.deb

#Pull in required lib-mono packages via apt:
apt-get install -f
```
- RPi 2

Fourdee, 2017-03-10:
Unable to replicate on RPi 2. Although, we cant find the cause on user's system, I believe issue is limited/unique to users current installation.

Marking as closed.

closed 2017-03-10 by Fourdee