MichaIng/DietPi #785: Webmin Install issue

9H5G, 2017-02-28:
I have a new installation (today) with Mosquitto, Node Red and a few nodes on a PI3 with USB drive configured.  I thought I had selected Webmin when installing NR etc but it wasn't running or responding to commands so I decided I had forgotten and decided to install it from the Software menu.  It stopped when 'unpacking webmin'.  Eventually, after some hours, I took the chance of a reboot.  Now, when I go to install webmin the installer stops at the apt simulation phase.  I have tried all the suggested actions to no avail.   Thx

9H5G, 2017-03-01:
Problem solved by running: dpkg --remove --force-remove-reinstreq webmin and then reinstalling

Fourdee, 2017-03-01:
@9H5G 

Hi,

Thanks for the report, i'll see if we can replicate your installation and the issue on a test system.

Fourdee, 2017-03-01:
@9H5G 

I was unable to replicate issue on fresh install (+ Webmin/Node red/Mosq), all software installs and functions successfully.

> It stopped when 'unpacking webmin'. Eventually, after some hours, I took the chance of a reboot.

Very strange, I assume no errors, simply system hangs on unpacking? If so, from my experience, this is an indication of a possible hardware failure (PSU / SD card).

Which:
- PSU used?
- SD card used?

9H5G, 2017-03-01:
Hi Dan,
Yes, very odd.  I have a 3A PSU and use new(ish) genuine 16gb Sandisk
ultra.  I suspect maybe a bad download followed by a power reset did the
damage. Was really flagging up that the advice given when the apt
simulation fails could include the line above if all else fails?

Thx for your quick reply.

Kind Regards
John

Fourdee, 2017-03-03:
@9H5G 

> Was really flagging up that the advice given when the apt
simulation fails could include the line above if all else fails?

Yep, good idea,

I'll see if we can include the apt output in the error message/log file.

Fourdee, 2017-03-03:
@9H5G 

Ticket: https://github.com/Fourdee/DietPi/issues/791

I'll mark this ticket as closed. Please reopen if required.

closed 2017-03-03 by Fourdee

TuRDMaN, 2017-05-10:
I'm running DietPi on a Pine A64. I am having this same issue with Webmin. 

```
Selecting previously unselected package webmin.
(Reading database ... 19660 files and directories currently installed.)
Preparing to unpack package.deb ...
Unpacking webmin (1.840) ...
```

It sticks there forever.... however the system is not frozen, as I can connect in another terminal session and htop shows processes running
```
  1  [||                                                                                       1.4%]     Tasks: 21; 1 running
  2  [||                                                                                       1.4%]     Load average: 1.44 1.51 1.14
  3  [                                                                                         0.0%]     Uptime: 01:02:32
  4  [                                                                                         0.0%]
  Mem[||||||||||||||||||||||||||||||||||||||||||||||||||                                 177/1988MB]
  Swp[                                                                                        0/0MB]

USER       NI CPU% MEM%   TIME+  Command
root        0 27.9  1.4  0:00.74 apt-get install -f -y --force-yes
root        0  0.5  0.1  0:05.37 htop
root        0  0.0  0.0  0:03.14 /bin/sh /usr/local/sbin/pine64_corekeeper.sh
root        0  0.0  0.1  0:04.37 /usr/sbin/dropbear -d /etc/dropbear/dropbear_dss_host_key -r /etc/dropbear/dropbear_rsa_host_key -p 22 -W 65536
root        0  0.0  0.1  0:02.10 /usr/sbin/dropbear -d /etc/dropbear/dropbear_dss_host_key -r /etc/dropbear/dropbear_rsa_host_key -p 22 -W 65536
root        0  0.0  0.1  0:00.50 /lib/systemd/systemd-journald
root        0  0.0  0.2  0:04.35 /sbin/init
root        0  0.0  0.1  0:00.42 /lib/systemd/systemd-udevd
root        0  0.0  0.1  0:00.12 /lib/systemd/systemd-logind
messagebu   0  0.0  0.1  0:00.39 /usr/bin/dbus-daemon --system --address=systemd: --nofork --nopidfile --systemd-activation
root        0  0.0  0.1  0:00.08 /usr/sbin/rsyslogd -n
root        0  0.0  0.0  0:00.01 /usr/sbin/dropbear -d /etc/dropbear/dropbear_dss_host_key -r /etc/dropbear/dropbear_rsa_host_key -p 22 -W 65536
root        0  0.0  0.4  0:00.00 dhclient -v -pf /run/dhclient.eth0.pid -lf /var/lib/dhcp/dhclient.eth0.leases eth0
root        0  0.0  0.0  0:00.03 /sbin/agetty --noclear tty1 linux
root        0  0.0  0.2  0:00.10 -bash
root        0  0.0  0.1  0:00.09 /bin/bash /DietPi/dietpi/dietpi-launcher
root        0  0.0  0.1  0:00.21 /usr/sbin/dropbear -d /etc/dropbear/dropbear_dss_host_key -r /etc/dropbear/dropbear_rsa_host_key -p 22 -W 65536
root        0  0.0  0.2  0:00.15 -bash
root        0  0.0  0.6  0:02.16 /bin/bash /DietPi/dietpi/dietpi-software
root        0  0.0  0.0  0:00.00 tee /var/log/dietpi-software_apt.log
root        0  0.0  0.0  0:00.00 sleep 5
```

I can leave it running for hours and it never completes. This happened once already, and I thought it was actually frozen, so I rebooted the device, and after forcing removal of the incomplete Webmin install with dpkg, I tried again now, and the same thing is happening.

ReanimationXP, 2017-07-22:
Had this problem for ages, thanks so much to 9H5G for suggesting:
dpkg --remove --force-remove-reinstreq webmin

Only thing that worked for me. Tried countless suggestions for the cleaning of apt-get and its' cache to no avail.

Fourdee, 2017-07-23:
@ReanimationXP 

> Update: Decided to reinstall webmin via the dietpi-software app as well and ended up with the same issue that started this mess. Required the first fix I mentioned above as well as apt-get --purge webmin, etc. Installing via apt-get directly had no issues. Something seems to be wrong with how dietpi-software is installing the 'Optimized' version of webmin.

thanks for the report 👍 

Is this the issues with webmin on a fresh DietPi image?

ReanimationXP, 2017-08-21:
That I'm not sure of, I used an image from Andreas Spiess on youtube.  I'll be making a fresh one soon and will let you know if I run into issues again.  I also had another probably unrelated issue related to installing stuff which turned out to be Andreas having changed the timezone to something european.