MichaIng/DietPi #786: Clock check?

9H5G, 2017-03-01:
This definitely an enhancement.  DietPi is attractive since its leaner all round.  I also chose it since I have very poor/unreliable network connectivity as I travel.  Often this means that NTP is inaccessible.

On the first boot, Dietpi checks for NTP time and if not received, the system clock is set to the time and date of the OS release. DietPi then rushes off to get updates automatically and you are led to the config and software menu.  Unfortunately much of the update and installation will fail since the date of the update and packages is in the future - sometimes by a long way.  

Would it be possible to check if the system time is valid (valid NTP update?) before launching into the update cycle to prevent this, albeit quite rare, situation?

Fourdee, 2017-03-01:
@9H5G 

Great idea 👍  i'll flag this for v146 milestone.

WolfganP, 2017-03-01:
If it fails on first attempt with currently configured servers, maybe a second try with the global servers at 0.pool.ntp.org or 1.pool.ntp.org is worthy.

Fourdee, 2017-03-09:
We actually run ntpd with the following cmdline:
```
   -g no  panicgate      Allow the first adjustment to be Big
                                - may appear multiple times
```

This should always allow for large adjustments to times, that are much further in the future. I believe the issue is simply ntpd not completeing this, prior to ```dietpi-software``` doing its thing.

I'll add a system which checks for the exit status of ```ntpd```, delay ```dietpi-software``` starting any installations accordingly.

9H5G, 2017-03-09:
OK Dan,

I suspect that the occasional user will still want a way to carry on if his
network does not pernit ntpd to operate correctly though.

Kind Regards
John

Fourdee, 2017-03-09:
Notes:

- If ```dietpi-software``` is launched and ```ntpd``` process is not completed, or, has a non zero exit code (eg: failed):
[image]

Fourdee, 2017-03-09:
Completed

Fourdee, 2017-03-09:
@WolfganP 

> 0.pool.ntp.org or 1.pool.ntp.org

If we can use 100.x 101.x lets add these in by default in patch, else, could be tricky to safely patch.

closed 2017-03-09 by Fourdee