MichaIng/DietPi #787: RPi Zero W | Add Support onboard WiFi

Fourdee, 2017-03-01:
>

Fourdee, 2017-03-01:
Waiting for response/stock details from PiSuppy:
https://twitter.com/DietPi_/status/837020769502826505

Fourdee, 2017-03-07:
- Update DietPi RPi Kernel required

Fourdee, 2017-03-08:
Fresh compile of RPi Kernel 4.4.50 (https://github.com/raspberrypi/linux/tree/rpi-4.4.y), no WiFi. Even when installed on the latest Raspbian image, which works prior to kernel replacement.

Fourdee, 2017-03-08:
- 🈯️ BT test:
```
root@DietPi:~# bluetoothctl
[NEW] Controller B8:27:EB:03:A5:41 DietPi [default]
[bluetooth]# power on
Changing power on succeeded
[bluetooth]# scan
Missing on/off argument
[bluetooth]# scan on
Discovery started
[CHG] Controller B8:27:EB:03:A5:41 Discovering: yes
[NEW] Device 54:60:09:D7:E1:87 54-60-09-D7-E1-87
[NEW] Device A4:77:33:B5:B9:8D A4-77-33-B5-B9-8D
```

Fourdee, 2017-03-08:
New image live, completed. Will update Stretch image at a later date: https://github.com/Fourdee/DietPi/issues/475#issuecomment-285044983

closed 2017-03-08 by Fourdee

rulenma, 2017-03-18:
Hi,

doesn`t work for me.
Iam running a fresh installation of 146 on the RPi zero w and it says "wlan0    No such device"
But in the config Wifi is enabled.
Anything else that I can do?
Iam searching for the right command to see if I can see that device, but I don't know the command and the name of the device.

Anyone has experience with the wifi modul on the zero w?
They say it`s the same modul as on the RPi3 but I compared both and on the zero w its just bad.
The Access Point can be in the same room and the zero w is 10m away and the signal quality is worst or it can't see the wifi.

Any ideas?
regards :)