MichaIng/DietPi #794: /usr/bin/sudo and setid missing

WolfganP, 2017-03-04:
I just burned the latest img for Raspberry Pi available on the site (it boots as v144) and noticed that on 1st boot sudo fails on "sudo: /usr/bin/sudo must be owned by uid 0 and have the setuid bit set".
The executable is actually owned by root:root, but setid flag is not set (solved via 'chmod 4755 /usr/bin/sudo')
I wonder if it's not a side effect of the recent reworking of the images?
Thx, Wolf

BTW: wont it be possible to include a unprivileged user preconfigured (ie 'pi' as in raspbian but with password 'dietpi' for easier onboarding on people migrating from raspbian). I know about the discussion at http://dietpi.com/phpbb/viewtopic.php?f=9&t=22 but that thread is locked (I support the enhancement BTW).

Fourdee, 2017-03-05:
@WolfganP 

> The executable is actually owned by root:root, but setid flag is not set (solved via 'chmod 4755 /usr/bin/sudo')
I wonder if it's not a side effect of the recent reworking of the images?

Strange, seems this isn't the 1st report:
https://github.com/Fourdee/DietPi/issues/775#issuecomment-282286311

We actually use the official RPi Jessie Lite image as a base, then prep it for DietPi. As far as I know, we make no changes to ```/usr/bin/sudo```.

> BTW: wont it be possible to include a unprivileged user preconfigured (ie 'pi' as in raspbian but with password 'dietpi' for easier onboarding on people migrating from raspbian). I know about the [...]

I've been thinking of creating a user ```dietpi``` and ```dietpi```, replacing the root user, however, its a big change (code wise) and everything would need testing (all 131 software installations in ```dietpi-software```). Would also require a new image, and us resetting the update system, possibly not something we could patch with certain success.

Fourdee, 2017-03-05:
@WolfganP 

> (solved via 'chmod 4755 /usr/bin/sudo')

Great stuff 👍 I'll get this patched for v146

Fourdee, 2017-03-05:
Lets apply it to all devices, as Pine seems to be effected also: https://github.com/Fourdee/DietPi/issues/775#issuecomment-282286311

Fourdee, 2017-03-05:
Completed. 

I'll create another ticket for dietpi user and link it here.

Fourdee, 2017-03-05:
@WolfganP 

https://github.com/Fourdee/DietPi/issues/796

closed 2017-03-05 by Fourdee

FWeissenb, 2018-05-27:
I installed Dietpi V6.8 on odroid XU4 and this bug still exists. :/

I used auto-install to create the system.

Fourdee, 2018-05-27:
@FWeissenb 

Thanks for the report 👍 

This 'should' be applied to our images by default, however, please try running the following command:
```
chmod 4755 $(which sudo)
# then verify with
ls -lha $(which sudo)
```

Should result in:
```
root@DietPi:~# ls -lha $(which sudo)
-rwsr-xr-x 1 root root 134K Jun  5  2017 /usr/bin/sudo
```

If this resolves, we'll run this as a patch for v6.9.

FWeissenb, 2018-06-04:
Hi Fourdee,

sorry for late repsonse. It doesn't work. I get  a "not allowed"

dietpi@Terminus:~$ sudo dietpi-update 
sudo: /usr/bin/sudo muss dem Benutzer mit UID 0 gehören und das »setuid«-Bit gesetzt haben
dietpi@Terminus:~$ chmod 4755 $(which sudo)
chmod: Beim Setzen der Zugriffsrechte für '/usr/bin/sudo': Die Operation ist nicht erlaubt
dietpi@Terminus:~$ ls -lha $(which sudo)
-rwxr-xr-x 1 root root 101K Mai  4 15:40 /usr/bin/sudo
dietpi@Terminus:~$ sudo dietpi-software 
sudo: /usr/bin/sudo muss dem Benutzer mit UID 0 gehören und das »setuid«-Bit gesetzt haben

WolfganP, 2018-06-04:
make user user _dietpi_ is listed as sudoer

> $ cat /etc/sudoers

Fourdee, 2018-06-04:
@WolfganP @FWeissenb 

Sudoer should exist here:
```
root@DietPi:~# cat /etc/sudoers.d/dietpi
dietpi ALL=NOPASSWD: ALL
```

@FWeissenb 
Try running the command as root login:
```
sudo su
chmod 4755 $(which sudo)
exit
```

FWeissenb, 2018-06-04:
I can't login as root :/

If i try he askes a password. And if i try "dietpi" it doesn't work.
I think there is something very broken. 
I will try to reflash my sd tomorrow.