MichaIng/DietPi #795: Resolution not set correctly when using display_rotate

Grayda, 2017-03-05:
I'm building a smart mirror using an old Raspberry Pi 1 I have laying about. The display needs to be rotated 90°, so I set `display_rotate=1` in `/boot/config.txt`. When I boot up, it displays as expected (the boot screen text reaches from the top of the display right down to the bottom) but as soon as it hits the DietPi part, the display only stretches a small portion, as though the display has been rotated, but the aspect ratio has remained the same (e.g. 16:9 instead of 9:16).

You can see a photo here: http://imgur.com/a/Dh959 . Sorry for the quality, but I don't feel like pulling apart my mirror yet. The green line indicates my LCD screen, while the red line indicates where the "display" starts and stops.

I'm sure I can fix this by setting my own resolution (haven't looked into it yet) but I mostly wanted to let you know about this. I'm running DietPi_RPi-armv6-(Jessie).7z, downloaded last night (4/3/2017). If I use Raspbian, it works exactly as expected (but Raspbian runs like a dog on the RPiv1)

Fourdee, 2017-03-05:
@Grayda 

> display_rotate=1 in /boot/config.txt

You'll need to apply this change in ```/DietPi/config.txt```. This is the DietPi-RAMdisk location and will save to ```/boot/config.txt``` on reboot.

I also believe ```display_rotate=1``` may require increased GPU memory to function, so if it still fails, try increasing GPU memory share to 128MB.

Grayda, 2017-03-05:
I changed the line outside of DietPi (plugged the SD card into my computer and made the change on the /boot partition), so the necessary change was already there, and in /DietPi/config.txt. Just to be on the safe side, I formatted the card, installed DietPi again and made the change on /DietPi/config.txt only. Rebooted, confirmed the image was rotated.

I also upped the memory and ran the firmware update in dietpi-config and still get the same thing.

I also realised that I made an error in my first post. The boot screen text doesn't go along the entire height of the screen, it appears as the rest of the screen does -- as if someone has rotated the screen and simply shrunk the image down to fit the new width.

I've drawn up a simple diagram to better explain: http://imgur.com/a/9B8Pc . Note how the numbers 1-15 all appear

I also tried the official Raspbian image again, just to make sure I wasn't going crazy, and it works exactly as expected

Fourdee, 2017-03-05:
@Grayda 

Thanks for the info 👍 

> I also tried the official Raspbian image again, just to make sure I wasn't going crazy, and that works exactly as expected

I'll run some tests here. See if we can find a solution.

Fourdee, 2017-03-05:
Strange, cant get rotation to work at all on my RPi 1 B. ```display_rotate=1``` and 128MB GPU split.

@Grayda 

Can you link your ```config.txt``` please?

Fourdee, 2017-03-05:
@Grayda 

Found the solution, you also need to flip values of ```framebuffer_width=``` and ```framebuffer_height=```:
eg: 720p, + 90 rotation is
```/DietPi/config.txt```
```
framebuffer_width=720
framebuffer_height=1280
```

Grayda, 2017-03-07:
Gave it a shot last night and it works great. Thanks for the tip!

closed 2017-03-07 by Grayda