MichaIng/DietPi #796: User | Create DietPi user, use for all software installs

Fourdee, 2017-03-05:
Change from ```root``` to ```dietpi``` user. 
- All software installs (- ```www-data```)
- Default login

WolfganP, 2017-03-05:
@Fourdee Thx for this enhancement. As I commented in https://github.com/Fourdee/DietPi/issues/794, as per http://dietpi.com/phpbb/viewtopic.php?f=9&t=22 I think the best will be to *add* an unprivileged user ('pi' is a good candidate as it will allow easier transition for raspbian users) rather than replace 'root' (as many existing dietpi users are already using this setup)
This way I think it will also encourage this second set of users to better user practices on avoiding root for everything (and use sudo when needed instead).

WolfganP, 2017-03-24:
@Fourdee Dan, any version target for this enhancement. Anything we can do from our side to help?
Thx, Wolf

PS: this is the "recipe" I follow to add a plain user 'pi' to a fresh dietpi install:

	mkdir /home; mkdir /home/pi
	useradd --group 100 --uid 1000 --shell /bin/bash --home /home/pi pi
	## apt-get install sudo
	chmod 4755 /usr/bin/sudo     // solved on https://github.com/Fourdee/DietPi/issues/794
	usermod -G sudo pi
	passwd pi
	chown pi:users /home/pi
added mods due my setup:
	chgrp -R users /var/lib/plexmediaserver /var/lib/rpimonitor
        smbpasswd -a pi
        smbpasswd -e pi

Fourdee, 2017-03-26:
@WolfganP 

Thanks 👍 

I'll flag this for v149

WolfganP, 2017-03-26:
Thanks @Fourdee , but maybe a two step implementation is the best way to go (ie 1st create the user without any changes to packages install and let [human] users play with it and test the environment (like I did above, just creating the user and enabling it for some services like samba), 2nd work on the packages and root restrictions (force the move to unprivileged user)

Fourdee, 2017-03-26:
> but maybe a two step implementation is the best way to go (ie 1st create the user without any changes to packages install and let [human] users play with it and test the environment (like I did above, just creating the user and enabling it for some services like samba),

@WolfganP 
Sounds good 👍

Fourdee, 2017-04-17:
notes:
```

			# mkdir -p /home
			# useradd -m amiberry
			# chpasswd <<< "amiberry:amiberry"

			#	Allow sudo without password
			# if (( ! $(cat /etc/sudoers | grep -ci -m1 '^amiberry') )); then

				# echo -e "\namiberry ALL=NOPASSWD: ALL\n" >> /etc/sudoers

			# fi

			#	Same groups as user pi
			# usermod -a -G adm amiberry
			# usermod -a -G dialout amiberry
			# usermod -a -G cdrom amiberry
			# usermod -a -G sudo amiberry
			# usermod -a -G audio amiberry
			# usermod -a -G video amiberry
			# usermod -a -G plugdev amiberry
			# usermod -a -G games amiberry
			# usermod -a -G users amiberry
			# usermod -a -G input amiberry
			# usermod -a -G netdev amiberry
			# usermod -a -G spi amiberry
			# usermod -a -G i2c amiberry
			# usermod -a -G gpio amiberry
			# usermod -a -G tty amiberry
```

WolfganP, 2017-04-17:
@Fourdee you may use the condensed syntax for usermode:

`usermod -a -G group1,group2,group3 username`

Fourdee, 2017-04-20:
@WolfganP 

Ok, heres whats done (available in testing branch https://github.com/Fourdee/DietPi/blob/master/TESTING-BRANCH.md):
- 🈯️ New user ```dietpi``` with password ```dietpi```
- 🈯️ Same permissions as ```pi``` user
- 🈯️ DietPi user data directory: user ```dietpi``` permissions applied during ```dietpi-software``` installs (this is prep for below)
- 🈺 ```root``` is still the default user account used in ```dietpi-software```

Fourdee, 2017-04-22:
Ok, done.

I'll mark this one as closed, then create another to replace default ```root``` to ```dietpi``` user. This will require a complete reset of images.

closed 2017-04-22 by Fourdee

joaofl, 2017-08-29:
@Fourdee 
Because you pointed me out this thread yesterday, I realized what you have done already towards adding the dietpi user.

So I realized I could already ssh into my dietpi using the diepi user, with default password dietpi. But even more dangerous, the sudo command for the dietpi user does not require password.
This means that whatever dietpi exposed to the internet, that did not change the password for the dietpi user is very exposed.

Maybe by default ssh should be only allowed by root user, or at least expose the need to change password for both users... And most probably require password for commands using sudo.

Fourdee, 2017-08-29:
@joaofl 

> This means that whatever dietpi exposed to the internet, that did not change the password for the dietpi user is very exposed.

Same could be said for users who dont change their root password. But yes, users may not be aware of the dietpi user.

> Maybe by default ssh should be only allowed by root user, or at least expose the need to change password for both users... And most probably require password for commands using sudo.

Yep, I believe we could add a info window during 1st run setup, inform the user to change passwords for ```root``` and ```dietpi```.
Although, users are only at any real risk if their network has:
- UPnP enabled
- UPnP disabled: SSH port is forwarded to IP address of system.