MichaIng/DietPi #797: DietPi-Software | Koel

sootstone, 2017-03-05:
Audio streaming service similar to Ampache - https://github.com/phanan/koel

elRadix, 2017-03-05:
yes, please include this.

I tried to install this, but failed

phonicmouse, 2017-03-06:
+1 for this, it would be very useful

sootstone, 2017-05-23:
[Koel Documentation](https://koel.phanan.net/docs/#/?id=installation)

Manged to get this running on an RPi 3 and also Virtual Box with Apache. Only a small issue with [Koel not able to run from a subdirectory](https://github.com/phanan/koel/issues/153) which doesn't quite fit from a DietPi perspective, but I'm sure we can work something out. Needs work, but basically - 

1. Install a LAMP stack and Node.js from dietpi-software.

2. Install [Composer](https://getcomposer.org/doc/00-intro.md#globally) globally:

```
php -r "copy('https://getcomposer.org/installer', 'composer-setup.php');"
php -r "if (hash_file('SHA384', 'composer-setup.php') === '669656bab3166a7aff8a7506b8cb2d1c292f042046c5a994c43155c0be6190fa0355160742ab2e1c88d40d5be660b410') { echo 'Installer verified'; } else { echo 'Installer corrupt'; unlink('composer-setup.php'); } echo PHP_EOL;"
php composer-setup.php
php -r "unlink('composer-setup.php');"
mv composer.phar /usr/local/bin/composer
```

3. Install [Yarn](https://yarnpkg.com/lang/en/docs/install/#linux-tab) :

```
curl -sS https://dl.yarnpkg.com/debian/pubkey.gpg | sudo apt-key add -
echo "deb https://dl.yarnpkg.com/debian/ stable main" | sudo tee /etc/apt/sources.list.d/yarn.list
apt-get update && apt-get install yarn
```

4. Get [Koel Latest Stable](https://github.com/phanan/koel/releases/tag/v3.6.2), unpack, move and change permissions:

```
wget "https://github.com/phanan/koel/archive/v3.6.2.zip" -O package.zip
unzip -o package.zip
rm package.zip
mv koel-* /var/www/koel
chmod 775 -R /var/www/koel
chown -R www-data:www-data /var/www/koel
```
5. cd to koel directory and install with Composer:

```
cd /var/www/koel
composer install
```

6. Set up database (MySQL example, [credit to bplower](https://gist.github.com/bplower/613a99156d603abac083)):

```
mysql -u root -p
Enter password:
mysql> CREATE DATABASE koel DEFAULT CHARACTER SET utf8 DEFAULT COLLATE utf8_general_ci;
mysql> CREATE USER 'koel-db-user'@'localhost' IDENTIFIED BY 'koel-pass';
mysql> GRANT ALL PRIVILEGES ON koel.* TO 'koel-db-user'@'localhost' WITH GRANT OPTION;
mysql> exit;
```
7. Modify .env with details info, should look something like this (still in the koel directory):

```
nano .env

# Database connection name, which corresponds to the database driver.
# Possible values are:
#   mysql (MySQL/MariaDB - default)
#   pgsql (PostgreSQL)
#   sqlsrv (Microsoft SQL Server)
DB_CONNECTION=mysql
DB_HOST=127.0.0.1
DB_DATABASE=koel
DB_USERNAME=koel-db-user
DB_PASSWORD=koel-pass

# Username and password for the initial admin account
# This info will be populated into the database during `php artisan db:seed`
# After that, it can (and should) be removed from this .env file
ADMIN_EMAIL=admin@example.com
ADMIN_NAME=admin
ADMIN_PASSWORD=admin-pass
```
8. init Koel instance:

`php artisan koel:init`

Optional:  _`php artisan serve --host 0.0.0.0` will allow you use koel at the DietPi systems IP on port 8000 (http://192.168.0.100:8000)_

**Apache Configuration**

1. Enable mod_rewrite.

```
a2enmod rewrite
dietpi-services restart
```

2. Configure Virtual Host 

```
nano /etc/apache2/sites-available/000-default.conf

DocumentRoot /var/www/koel

<Directory /var/www/koel>
	Options Indexes FollowSymLinks
	AllowOverride None
	Require all granted
</Directory>

dietpi-services restart
```

Visit DietPi systems IP (http://192.168.0.100) and you should be able to log in with the details set in `.env`

**Email Address**: admin@example.com
**Password**: admin-pass

Set Music Directory in Settings

e.g. /mnt/dietpi_userdata/Music

[Koel supports three streaming methods which can be configured via a STREAMING_METHOD setting in .env file](https://koel.phanan.net/docs/#/?id=streaming-music)

Ideally we would use [x-sendfile](https://tn123.org/mod_xsendfile/) for Apache. I was able to set this up, but it needs [apxs](https://httpd.apache.org/docs/2.4/programs/apxs.html), installed with `apt-get install apache2-dev` which obviously installs a lot of stuff we don't need, so maybe we can get round this somehow.

Install x-sendfile:

```
wget "https://tn123.org/mod_xsendfile/mod_xsendfile.c#hash(sha256:8e8c21ef39bbe86464d3831fd30cc4c51633f6e2e002204509e55fc7c8df9cf9)"
apxs -cia mod_xsendfile.c
dietpi-services restart
```

Configure Virtual Host:  

LoadModule xsendfile_module   /usr/lib/apache2/modules/mod_xsendfile.so

  <IfModule mod_xsendfile.c>
    XSendFile on
    XSendFilePath /var/www/koel
  </IfModule>

Change .env

```
nano /var/www/koel/.env

STREAMING_METHOD=x-sendfile
```

Fourdee, 2017-05-24:
@sootstone 

Nice one, looks good 👍 

I'll test your notes and convert it into DietPi-Software.

Fourdee, 2017-05-24:
notes:

Requires:
- nodejs
- lxmp stack
- Python/build-essential (for ```npm install``` ARM64)
- ffmpeg for transcoding

sootstone, 2017-05-24:
@Fourdee 

Nice :+1: 

([this](https://gist.github.com/jbheren/5c58b8488e0c47e6063cdcbe599c2ce1) might help)

sootstone, 2017-05-24:
Edited mistake in virtual host

> `AllowOverride None`

to 

`AllowOverride All`

sfb2, 2017-05-24:
very interesting "software/project" this Koel!!

> elRadix commented on 5 Mar
> yes, please include this.
> I tried to install this, but failed

unfortunately same here: got an "css/app-css not defined in asset manifest"
(Node.js 7.10 installed/ Composer 1.4.2))

sootstone, 2017-05-25:
@sfb2 

Did you install with [Yarn](https://yarnpkg.com/en/)? That's the only supported installation method at the moment as far as I'm aware, you'll receive a lot of errors otherwise.

sfb2, 2017-05-25:
> Did you install with Yarn? That's the only supported installation method at the moment as far as I'm aware, you'll receive a lot of errors otherwise.

I did exactly the steps you posted..many many thx btw.: great "tutorial"..!!
yes, i installed with Yarn, but maybe an old version ? 1.54 or something ? could really be the problem..
installed the steps here on a "normal" Xubuntu 16.04 (PC) and under Raspbian Jessie (RPI)..
guess, i have to make it completely/fresh again..
but many thx for your hint here !!; guess also Koel is a great piece of streamer; has to be in the dietpi-software (optional)..
many thx again ;-)

Fourdee, 2017-05-25:
- 🈯️ Lighttpd
- Nginx
- Apache2

- Online Doc: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=7305#p7305

Fourdee, 2017-05-25:
Completed.

closed 2017-05-25 by Fourdee

sootstone, 2017-05-25:
@sfb2 

No problem! Glad you could get it working.

sootstone, 2017-05-25:
Great job @Fourdee, looks awesome! :+1: 

I'm probably going to stick with my own setup for personal use so I can customize it for Apache, but yours is perfect for DietPi users. Hope you don't mind if I steal some of it?

Fourdee, 2017-05-26:
@sootstone 

> Hope you don't mind if I steal some of it?

By all means 👍 Its based on your install notes, without which, wouldn't be possible :)

sootstone, 2017-06-01:
Updated for future reference.

MasaruKato, 2017-07-05:
Hi, I followed your guide step by step and I keep stumbling on this error after `php artisan koel:init`. is there anything I can do to fix it?

```
gyp ERR! stack     at emitTwo (events.js:125:13)
gyp ERR! stack     at ChildProcess.emit (events.js:213:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_proces                                                                             s.js:197:12)
gyp ERR! System Linux 3.4.113-sun8i
gyp ERR! command "/usr/local/bin/node" "/var/www/koel/node_modules/node-gyp/bin/                                                                             node-gyp.js" "rebuild" "--verbose" "--libsass_ext=" "--libsass_cflags=" "--libsa                                                                             ss_ldflags=" "--libsass_library="
gyp ERR! cwd /var/www/koel/node_modules/sass-loader/node_modules/node-sass
gyp ERR! node -v v8.1.3
gyp ERR! node-gyp -v v3.5.0
gyp ERR! not ok
Build failed with error code: 1
info Visit https://yarnpkg.com/en/docs/cli/install for documentation about this 
```

sootstone, 2017-07-08:
@MasaruKato 

That's odd, what system are you trying to install Koel on?
Are you installing it as root?
Have you tried removing the `$HOME/.node-gyp` and `$HOME/.npmrc` directory, and rebooting/retrying? (Worked for people with similar problems)

MasaruKato, 2017-07-10:
I'm trying to install it on a Orange Pi PC.
Yes, I'm using the default Dietpi account... I will try to do a fresh installation and delete those folders before
`php artisan koel:init`. Dumb question, do I need `npm install` or `yarn `is enough?

sootstone, 2017-07-13:
> Dumb question, do I need npm install or yarn is enough?

`php artisan koel:init` will automatically use yarn, so no need to use either :)

MasaruKato, 2017-08-31:
For curiosity, I tried to install it from the `dietpi-software` list and it doesn't work all the same.  Maybe something changed in the installation process?

Fourdee, 2017-08-31:
@MasaruKato 

> For curiosity, I tried to install it from the dietpi-software list and it doesn't work all the same. Maybe something changed in the installation process?

Not that we are aware, which device tested on?

MasaruKato, 2017-08-31:
> Not that we are aware, which device tested on?

On a Orange Pi PC. These are the screenshots of the broken web page. 
[image]
[image]

sfb2, 2017-08-31:
hi there,
looks like a yarn problem..try to update it to the latest..

MasaruKato, 2017-09-03:
> hi there,
looks like a yarn problem..try to update it to the latest..

So I tried again, with yarn updated at 0.27.5... same result.

Fourdee, 2017-09-03:
@MasaruKato 

We tested install with the swapfile increase in v155 on fresh install, was functional. Are you running v155?
https://github.com/Fourdee/DietPi/issues/1126#issuecomment-326749203

Also please check swapfile size:
```
/DietPi/dietpi/func/dietpi-set_dphys-swapfile
```

MasaruKato, 2017-09-03:
> We tested install with the swapfile increase in v155 on fresh install, was functional. Are you running v155?

I tried to install it from dietpi-software and It worked. I will try a "manual" installation following the guide above and I'll let you know.