MichaIng/DietPi #801: Possible distribute arm64 kernel for RPI 3

Electron752, 2017-03-08:
I was wondering if you have any interest or plans to offer the ability to distribute a arm64 kernel for the RPI 3.

Arm64 support for the RPI 3 is still very experimental, but it's starting to approach the ability to be taken seriously.

Fourdee, 2017-03-08:
@Electron752 

Thanks for the request 👍 

> Arm64 support for the RPI 3 is still very experimental 

It really depends on how stable/reliable it is on a users system. As, we would need to also support it and the users installations. I'll make a note to compile the ARM64 kernel and test it. Then go from there.

MichaIng, 2018-01-31:
As we stopped offering own kernels for RPi, due to too hight effort for low benefit, we stay with the official kernel package, which is the one from stable branch of raspberrypi firmware repo: https://github.com/raspberrypi/firmware/tree/stable
That way we can guarantee stability and compatibility as far as RPi devs do and we can concentrate on enhancing our software related parts.

I read much about arm64 Raspbian/Kernel for RPi3 and the actual benefit really seems marginal compared to the additional effort, that would take away development time from other parts of RPi/Raspbian. Personally I would follow/agree RPi devs in this. It makes sense to address feedback/bugs/enhancements to their github repo to make them available for all RPi users and as fast as arm64 kernel/Raspbian repo is official available, we will offer it as well.

I close this issue for cleaning reasons. Reopen on demand.

closed 2018-01-31 by MichaIng

Electron752, 2018-01-31:
I basically agree.  ARM64 was always experimental to begin with and was mostly a learning experience.  The actual performance benefits for the typical RPI user is rather low actually.

It's too bad the 1GB limit is rather fundamental to the chipset architecture.  But 1GB is quite a bit of ram for a typical RPI application to begin with.