MichaIng/DietPi #803: WaveShare32 | Input axis inverted

Fourdee, 2017-03-11:
> https://github.com/Fourdee/DietPi/issues/767#issuecomment-282303510

MichaIng, 2018-10-15:
@Fourdee 
To clean up old issues, do you have a chance to retest this, also if just an issue on Jessie, or on Stretch as well?

Fourdee, 2018-10-20:
Using evdev method:
https://github.com/notro/fbtft/issues/445#issuecomment-354714529
```
root@DietPi:~# cat /var/log/Xorg.0.log | grep EE
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[   637.368] (EE) AIGLX: reverting to software rendering
[   637.580] (II) XINPUT: Adding extended input device "ADS7846 Touchscreen" (type: TOUCHSCREEN, id 6)
[   637.583] (EE) evdev: ADS7846 Touchscreen: Unable to query fd: Inappropriate ioctl for device
[   637.690] (EE) PreInit returned 2 for "ADS7846 Touchscreen"
```

Fourdee, 2018-10-20:
Resolved with setting evdev and using 270 rotation in config.txt.

However, still gives us an error, but it works 🤣 
```
root@DietPi:~# cat /var/log/Xorg.0.log | grep EE
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    15.193] (EE) AIGLX: reverting to software rendering
[    16.546] (II) XINPUT: Adding extended input device "ADS7846 Touchscreen" (type: TOUCHSCREEN, id 6)
[    16.548] (EE) evdev: ADS7846 Touchscreen: Unable to query fd: Inappropriate ioctl for device
[    16.620] (EE) PreInit returned 2 for "ADS7846 Touchscreen"
```

Fourdee, 2018-10-20:
Re-install test:
-

closed 2018-10-20 by Fourdee

MichaIng, 2018-10-21:
@Fourdee 
This is not related to: https://github.com/Fourdee/DietPi/issues/2130 right?

Fourdee, 2018-10-21:
@MichaIng 

> This is not related to: #2130 right?

Nope, XU4 is a DRM issue with driver.