MichaIng/DietPi #804: DietPi Image | HyperV

Fourdee, 2017-03-12:
>

JimLee1996, 2017-03-13:
I've converted virtual disk of VMware(VMDK) to Hyper-V disk(VHD), and it works well Hyper-V host running on Windows Server 2008 R2.
### Download link 
[Google Drive](https://drive.google.com/file/d/0B_QhPFq6vCVrNDJIaERuYjBTdTg/view?usp=sharing)

Fourdee, 2017-03-13:
@JimLee1996 

Legend 👍 

Thanks for sharing

jamescalderon, 2017-07-14:
Thanks! Works like a charm!

sytone, 2017-07-28:
I have a post on how to do this. 
http://www.sytone.com/2017/07/getting-a-dietpi-image-running-on-hyper-v/

Ruvetuve, 2017-07-29:
Will this work on Proxmox? Don't have the experience but I'am curious.

Ruvetuve, 2017-07-30:
Or maybe my understanding is wrong. I can just install proxmox and install the dietpi x64 image to a container?

sytone, 2017-08-02:
Thisi s for Hyper-V which is a windows virtualization platform available on windows since Windows 8.

Ruvetuve, 2017-08-04:
So which image should I take for proxmox?

Fourdee, 2017-11-30:
Reference:
http://dietpi.com/phpbb/viewtopic.php?f=9&t=2530&p=9781#p9781

MichaIng, 2018-01-28:
Nice, I thought Hyper-V is just to create/run Windows guests.

This can be enabled btw. within any Windows 8+ desktop, no Windows Server necessary. Will activate it and see if I can setup a fresh DietPi image (non UEFI for simplicity for now).

hoveeman, 2018-01-29:
@sytone I tried your process with the new dietpiv6 Virtual Box image and have been unsuccessful in getting the Hyper-V to boot. I didn't use your powershell script since the naming conventions and files have changed. I downloaded the .7z of the virtual box image from dietpi, extracted the .ova and then extracted the .vmdk with 7zip. I then ran this command.
```
.\qemu-img.exe convert ".\DietPi_VirtualBox-x86_64-(Stretch)-disk001.vmdk" -O vhdx -o subformat=dynamic '.\DietPi_VirtualBox-x86_64-Stretch-v6.vhdx'
```
I then create a Hyper-V Machine with the newly created .vhdx and I get the following error when the machine tries to boot. Note that I have not run into this issue on previous versions of dietpi, ie. v159.

[image]

MichaIng, 2018-01-29:
@hoveeman 
Maybe the conversion somehow changes the UUID? I am not too deep in this. I just created a Hyper-V image from Debian mini.iso.

Everything went fine. However I was not able to shrink the vhdx. zerofree, defragmentation, resize/compression function of Hyper-V, all didn't shrink the size beneath 3.35 GB. Seems to have something to do with data on disk location. Also exporting the machine just copies the disk image and does no compression at all. Quite poor, if it is about moving those images around...

After trying around and finally doing export, I suddenly faced a corrupted `/etc/network/interfaces` which didn't allow any networking. Fixed it manually, after that everything went fine. Just keep in mind, no idea what was the actual reason, or if other files might be affected.

At least 7zip was able to do the compression down to 110M. I will give @Fourdee  access to it to upload it for testing reasons to http://dietpi.com/downloads/testing/

Fourdee, 2018-01-29:
@MichaIng 

> €: Great, Hyper-V blocks exclusively the hardware virtualization features, stopping VirtualBox to function, even leading to blue screens on starting trial of one VM. How should one now effectively create/test images for both platforms? 

Micro$hit? 😉

MichaIng, 2018-01-29:
@Fourdee 
Agree, Hyper-V offers slightly less features than VirtualBox, In fact I found non that VirtualBox does not offer as well. Internet is there without any necessary configuration, but no local SSH possible 👎.

Only reason would be, that some people a familiar with Hyper-V or need it anyway for other products and, yeah, can't run any other VM software concurrent (Hyper-V does not even need to be active, nor the background service. Just the installation guarantees VT-x reservation).
At least Debian and DietPi installation (script) runs without any issues, same as on other VM platforms, so it's easy to create/update an image here, if one has a Hyper-V only system 😉.

MichaIng, 2018-01-30:
@Fourdee 
About uploading the Hyper-V testing image. I am personally, due to the mentioned VirtualBox/VMware blocking, at the moment not able/willing to regularly provide Hyper-V images. In/Unstalling and Hyper-V is also heavy compared to other software, as it is handled like a (large) Windows update, needing double reboot after switch. So this just makes sense by somebody who has a separate Hyper-V host or anyway need/want to run this. I am personally happy with VirtualBox and want to stay with this for testing, thus can create images there regularly.

If someone else is found, who is willing to provide Hyper-V images, then the question is, if you can/want to test them, before publishing as official DietPi image. Otherwise for the variety of virtualization platforms I guess `PREP_SYSTEM_FOR_DIETPI.sh` does an easy great job here:
- Install/plug any Debian base image, in case go with Debian `mini.iso` (BIOS and UEFI works both): https://deb.debian.org/debian/dists/stretch/main/installer-amd64/current/images/netboot/ (single partition, preferable no swap, no additional software) 
- Follow instructions to run conversion script into DietPi: https://github.com/Fourdee/DietPi/issues/1285
- Choose hardware model `20` (virtual machine) and everything else as desired.
- Adjust `/boot/dietpi.txt` to suite your 1st run needs.
- Optionally backup image for other machines, before 1st run.

@hoveeman @Ruvetuve @sytone 
Would this be a solution? Should work on all virtualization platforms as said, as they basically all seem to work the same.
Note that the script is still in beta state. But wider testing is anyway necessary, before we can consider this stable.

Fourdee, 2018-01-30:
@MichaIng 

Agree 👍 

I've got plans to add a box entry on http://dietpi.com/download for a ```Other``` system, which links to PREP.

k-plan, 2018-01-30:
@Fourdee 

> which links to PREP.

Hmm... who want/will support such build images/versions?
I personally not. Sorry, for me to much imponderables.
Never can be sure, what's going wrong and what provoke the error.

And, I know, what I'm talking about now ... :wink:

k-plan, 2018-01-30:
@MichaIng 

I personally think we should close section "Hyper-V" now.

One platform less is no big deal. Work here is more more than enough.

PS: Yes, I could build a [Parallels](https://www.parallels.com)  and provide it, but  ...  :see_no_evil:

hoveeman, 2018-01-30:
@MichaIng This definitely sounds like a good option. I will give this a try. Really appreciate the instructions.

I also wanted to update this thread to let others know that I downloaded the VMWare version last night, extracted the 7zip and converted the vmdk to vhdx and it worked perfectly in Hyper-V. Not sure why this one worked while the VirtualBox version did not.

This was the command I used:
`.\qemu-img.exe convert ".\dietpi_vm.vmdk" -O vhdx -o subformat=dynamic '.\DietPiv6.vhdx'`

sytone, 2018-01-31:
@hoveeman hit the same issue as you, my Debian and other Linux OS work fine. I am guessing something in the conversion. I just tried the VMWare version and able to confirm it works well. Update the PowerShell script while I was at it. 

@MichaIng I no longer use Virtual box as I get more out of hyper-v for what I need along with docker containers for Linux and windows based containers. As a note you can switch if needed and used to do that. 
To disable Hyper-V in order to use VirtualBox, open a command prompt as administrator and run the command: ``bcdedit /set hypervisorlaunchtype off``
You’ll need to reboot, but then you’ll be all set to run VirtualBox. To turn Hyper-V back on, run: ``bcdedit /set hypervisorlaunchtype auto`` and then reboot.

Ill try the prep process later this week.

MichaIng, 2018-01-31:
@sytone 
Thanks for this info. Hmm maybe I will be able to provide an Hyper-V image then. But @Fourdee would need to agree doing final tests, or we consider it as unofficial image, just providing it here via github/forum.

Interesting, that VMWare image can be converted but VirtualBox not. The latter one was provided by me also. It is running fine on VirtualBox and should be similar to WMWare. Maybe there are some differences in the virtual hardware/the controller the image was added to. IDE/SATA/SCSI? But actually never faced issues just switching images around.

@Fourdee 
Maybe we can going through the base Debian image installation steps to see if we really do everything the same, providing some contributers image creation guideline (at least for VM/native). I will open a new issue for this and add all steps I do from Debian mini.iso on.

pablopoo, 2018-02-22:
Installed PREP_SYSTEM_FOR_DIETPI.sh over latest Debian Stretch (clean hyper-v VM, gen 2 UEFI). At the second boot, after the partition resize, the system failed to boot properly because the fstab PARTUUID don't match with the ext4 partition.

`systemd-remount-fs[171] : mount: can't find PARTUUID=` (some uuid)

After edited the fstab with the right UUID the system started ok.

yumiris, 2018-04-16:
In case anyone stumbles upon here, I've made a Hyper-V equivalent of the latest `DietPi_VMware-x86_64-Stretch.vmdk` image. It's modified so it successfully boots up, while everything else has been left untouched.

Sharing it here, should anyone want to have a shot at testing it!
[`DietPi_Hyper-V-x86_64-Stretch.7z`](https://1drv.ms/u/s!AoE_a2H65pxWkmND26RUC3c-WwBv) (100MB OneDrive download)

If it works successfully for people and demand is high enough, I'm up for providing Hyper-V images of DietPi on a regular basis.

piszpanjuzef, 2018-04-29:
Hi yumiris,
Downloaded your Hyper-V virtual machine and installed. It work perfectly.
thank you very much for for working Hyper-V machine.

yumiris, 2018-05-16:
This Hyper-V image is based on the previous one, but with the DietPi 6.8 updates applied. It's a tad bigger, although it saves from having to constantly download updates whenever a new VM is deployed.

[`DietPi_Hyper-V-x86_64-Stretch - v6.8.7z`](https://1drv.ms/u/s!AoE_a2H65pxWky0Sizev1iwhpZqC)(280 MB OneDrive download)

Enjoy!

yumiris, 2018-09-19:
It's been a while. Here's the v6.15 snapshot:

[https://1drv.ms/u/s!AoE_a2H65pxWlFYHW3fF978K-v6Q](DietPi_Hyper-V-x86_64-Stretch - v6.15.7z)(165 MB OneDrive download)

veljko, 2018-09-29:
> It's been a while. Here's the v6.15 snapshot:
> 
> [`DietPi_Hyper-V-x86_64-Stretch - v6.15.7z`](https://1drv.ms/u/s!AoE_a2H65pxWlFYHW3fF978K-v6Q) (165 MB OneDrive download)

@yumiris, thanks for building these, converting vmdk/ova image never worked for me. Hope it becomes official...

MichaIng, 2018-12-25:
Added to FeatHub, feel free to leave a vote: https://feathub.com/Fourdee/DietPi/+1

yumiris, 2019-01-31:
Hyper-V image for v6.20 can be found [here](https://github.com/yumiris/DietPi.Hyper-V/releases/tag/v6.20)!

MichaIng, 2019-02-12:
@yumiris 
Wow, just noticed the wonderful GitHub page you wrapped around. Complete and clear about what it is etc. 👍.

Just one little hint as we implemented for exactly this use:
- You see the `Image` info line on login banner? Would be great if you could add yourself there.
- For this, e.g. from the machine itself:
```
cat << _EOF_ > /{DietPi,boot}/dietpi/.prep_info
yumiris
DietPi for WMware
_EOF_
```
- First line your name for credits and user info.
- Second line the base image you used, e.g. if you used the official WMware image as a basis?

But no need to hurry with this or hack into the existing release IMO, just keep in mind when doing the next update ✌️.

@Fourdee 
Shall we add it as user created image to https://dietpi.com#download, linking users to https://github.com/yumiris/DietPi.Hyper-V?

yumiris, 2019-03-24:
@MichaIng

Done it in v6.22: https://github.com/yumiris/DietPi.Hyper-V/releases/tag/v6.22

---

And for the usual announcement:

Hyper-V image for v6.22 can be found [here](https://github.com/yumiris/DietPi.Hyper-V/releases/tag/v6.22)!

MichaIng, 2019-03-24:
@yumiris 
Great work, thank you very much!

yumiris, 2019-03-25:
@MichaIng 

Heh, I've already tried minimising the image as much as possible, i.e. running `dietpi-cleaner` and whatnot, but it certainly grew. I have a question: is a copy of the entire DietPi Git repository present on DietPi installs? 🤔 With so many commits per release, I wouldn't be surprised if the repository contributes a lot to the image's total size.

MichaIng, 2019-03-25:
@yumiris 
> Heh, I've already tried minimising the image as much as possible, i.e. running dietpi-cleaner and whatnot, but it certainly grew.

Yeah it is not so much about purging files/packages, as this should be already good on a fresh image.
- It is indeed about filling empty space on the image file with zeros first (via `zerofree` from Linux system), as by default you know that removed files are still located as "inodes" on the drive, just marked as removed thus can be overwritten.
- And the next auf shrinking the actual image file size with certain tools. VirtualBox has `VBoxManage modifyhd --compact` for .vdi files and WMware has `mware-vdiskmanager -k` for .vmdk files. Ah just found: https://weblogs.asp.net/dixin/shrink-virtual-hard-disk-image-vhd-and-vhdx-files
  - So you can use the Hyper-V manager itself to "compact" the virtual drive.
- However skipping the actual image file shrinking is not too bad if it's about the final compressed archive size. If you applied `zerofree` then e.g. 7zip is able to compress the zeros very efficiently, so the download size will be nearly the same even that the contained image file is still large. But of course also end users would be happier to extract/copy/handle a smaller image file than a larger one 😉.

> I have a question: is a copy of the entire DietPi Git repository present on DietPi installs? 

Nope, we do not clone the Git repository but only download and download and apply the current zip archive: https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-update#L330-L331
All scripts with settings/flag files and such is currently 1.4M, so has only a tiny influence:
```
2019-03-25 17:42:17 root@micha:/boot# G_TREESIZE /DietPi
1.4 MB /DietPi
1.4 MB /DietPi/dietpi
```

yumiris, 2019-03-26:
A long time ago I've tried `zerofree` with the VHDX Compact feature. That didn't make any difference to the VHDX's size, so I stopped doing that. However, I gave it another shot today, and like you've mentioned, 7-Zip is excellent at compressing zeroes.

The VHDX is still 2.6GiB like before, but it can now be compressed down to ~150MB. Although the VHDX will take up a bit of quite, the saved bandwidth when downloading the 7z file is enormous.

---

I have a deja vu of making an announcement like this:

Hyper-V image for v6.22-rev02 can be found [here](https://github.com/yumiris/DietPi.Hyper-V/releases/tag/v6.22-rev02)!

MichaIng, 2019-03-26:
@yumiris 
> A long time ago I've tried zerofree with the VHDX Compact feature. That didn't make any difference to the VHDX's size, so I stopped doing that.

Jep indeed, also the zeros take full space in the image file until you compress it. But the Hyper-V image compacting feature should also be able to reduce the image size then.

> I have a deja vu of making an announcement like this:

Hehe worth it, download size 134 MB vs. 343 MB 👍.

I will try it out by times. Would you mind if we link your GitHub page from dietpi.com to raise attention for your Hyper-V images?

yumiris, 2019-03-27:
@MichaIng 

> But the Hyper-V image compacting feature should also be able to reduce the image size then.

Yup, that's what we'd expect, but the feature isn't changing the size at all, even after using `zerofree`. That's actually what made me give up using `zerofree` when I tried it out over a year ago for my Hyper-V images. Now that I compress the images, I realised that the benefits were invisible. 😛

> I will try it out by times. Would you mind if we link your GitHub page from dietpi.com to raise attention for your Hyper-V images?

For sure, go for it!

MichaIng, 2019-03-27:
@yumiris 
> Yup, that's what we'd expect, but the feature isn't changing the size at all

Okay, I guess this was also what I found above: https://github.com/MichaIng/DietPi/issues/804#issuecomment-361314353
Strange though, what for is the compacting if it does not change anything 🤔. However at least download size can be reduced, which is more important anyway.

> For sure, go for it!

Great, could take a while since I have to consult Fourdee in how to edit the dietpi.com main page. I think he has some template/software for this.

yumiris, 2019-03-27:
@MichaIng 

Yup, it seems like we've both encountered the exact same scenario. I wonder if the the filesystem might affect the compacting. A wild guess is that an NTFS VHDX would be compacted much more effectively than ext4. 🤔 

---

Not a problem -- gives me more time to refine the project if needed. 😉

MichaIng, 2019-03-27:
@yumiris 
> I wonder if the the filesystem might affect the compacting. A wild guess is that an NTFS VHDX would be compacted much more effectively than ext4. 🤔

Possible. Also possible that compacting only works if the file system is actually resized via `resize2fs` or even the whole partition via `(g)parted`. Does not make much sense on dynamically allocated space, however not sure how Hyper-V is working.

Since dynamic allocation otherwise works fine with ext4 (the image file has not 8 GiB that the drive can have), the usual assumption would be that shrinking/compacting works the same. Possible that it's about which part of drive space is written. E.g. the if space is freed in middle and Hyper-V can only compact/regain free space from the end. This would require moving used space to the beginning of the drive space. E.g. `gparted` does that automatically when you use it to reduce the partition size. However, too much testing effort for now for the minor benefit 😉.

Fourdee, 2019-03-28:
@MichaIng 

🈯️ Added download info to https://dietpi.com/download
[image]

MichaIng, 2019-03-28:
@Fourdee 
Very nice, many thanks for adding it.

@yumiris 
Okay I think we can close this issue for now, or do you want to keep it opened to inform about image updates this way as well?
Thanks to dietpi-updates the image does not need to be updated too regularly, but of course just inform us if some day you do can't/do not want to maintain these anymore.

yumiris, 2019-03-29:
@Fourdee 

Absolutely not a problem, and thank you for the shoutout!

@MichaIng 

In regard to `gparted` & Hyper-V compact, it's indeed a minor thing, but you got me curious enough to want to test it out. I'll see if I can make Hyper-V actually compact the ext4 images sometime next weekend (hopefully!)

Will keep you updated.

---

It's up to you if you want to keep it open or closed. -- I don't mind announcing releases here if the issue stays open. 😛 

I'll keep a balance between keeping the project alive and letting `dietpi-update` do must of the updates. Maybe every two versions or so, but it depends on the time I've got. I don't think I'll give up on it anytime soon, since I'm a big fan of DietPi and I found it to be invaluable back when I used Hyper-V for my servers. However, of course, if for some reason I cannot continue releasing images anymore, I will let you know!

---

In regard to the issues, I've actually written in the readme:

```
Please rely on this project's issues page for errors/bugs directly related to Hyper-V, and the official DietPi GitHub or forums for DietPi-specific issues. If uncertain, report the issues here. Any issues that are related to DietPi itself will be relayed to the official team.
```

We'll leave it up to people to decide where to report issues, and we'll forward the issues to each other when needed.

MichaIng, 2019-05-01:
Okay I close this issue here, also as a general cleanup. Bug reports should open a new issue.

closed 2019-05-01 by MichaIng