MichaIng/DietPi #805: DietPi Image | VMware x32

Fourdee, 2017-03-12:
> http://dietpi.com/phpbb/viewtopic.php?f=11&t=1437&p=5988&hilit=x86_64#p5984

MichaIng, 2018-01-30:
@Fourdee 
Before offering all different kind of VM images, x32+x64, UEFI+BIOS, WMware+VirtualBox+Hyper-V+whatever (would make already 2x2x3=12 images), I think we could just generally include x32 support to the preparation script, to allow users creating their own. This should be quite easy, adding support for new CPU architecture `x86` or `i386` (?) and all other e.g. package related stuff should be handled by APT (i386).
As VM images are generally quite painless (for my impression, compared to SBCs with all their special own needs/APT source extensions etc.), this would in the first step enable us to start some public DietPi-Prep script testing, getting valuable feedback.

Fourdee, 2018-01-30:
@MichaIng 

> I think we could just generally include x32 support to the preparation script, to allow users creating their own. This should be quite easy, adding support for new CPU architecture x86 or i386

Yep, although the main reason i've not pushed this forward is because the DietPi scripts have no var/support for the HW_ARCH of x86_32. We'd also need to go through each of the dietpi-software titles and test/add code for x32 HW_ARCH.

Its an extremely long process if we want to do it right, most CPU's of the last 5-10 years support x86_64, hence reasons for not pushing this forward.

MichaIng, 2018-01-31:
@Fourdee 
Ah yes that's true, we would need to adjust several software installations to pull the other i386 package, if available at all, and/or disable all software titles that are not available via official repo with i386 candidate. Much work, and we have more important stuff to do at the moment. I will mark as low priority.

X3msnake, 2018-02-12:
@MichaIng  @Fourdee 

Is this something I can do manually on a fork?

What would needed be done?

where are each of the diet-pi titles you mention?

Problem is with the old laptops, for example i'm using diet-pi as the base for a VM for a extension for inkscape for embroidery and there are people using older laptops that have the power to run the VM but do not support x64

MichaIng, 2018-02-20:
@X3msnake 
See here within dietpi-software script: https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L3395
All titles, not installed via APT (e.g. due to not available/outdated version in Debian repo), but via individual packages/raw download/source, would need check for available x32 versions/support. Then these need to be added to `http://dietpi.com/downloads/binaries` in case and/or relinked to x32 bit binary in case: https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L4176

If you just search e.g. `amd64` through the script, you see that there are many cases, so it's quite a lot of work. Even if we at first just block those titles for x32, adding them step by step by times, the excluding code would need to be added to several places as well: https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L360

I understand your case, as especially older systems benefit by the low resource consumption of DietPi, whether as direct native OS or via VM. But this will not be possible too soon.

X3msnake, 2018-02-20:
@MichaIng  Yes I understand.

Thanks for the info and for the great work you all have been doing with this project :)

X3msnake, 2018-04-24:
> @MichaIng <https://github.com/MichaIng> I have some old 32bit pc systems
> ready to rock DietPi. Are there any super low consumption distro's you know
> of?

Alpine

Fourdee, 2018-06-29:
Marking as closed, as per: https://github.com/Fourdee/DietPi/issues/1854#issuecomment-401457209

closed 2018-06-29 by Fourdee