MichaIng/DietPi #806: DietPi Image | ASUS Tinker Board

Fourdee, 2017-03-12:
> http://dietpi.com/phpbb/viewtopic.php?f=9&t=1218&p=5846&hilit=ASUS+tinker#p5175

Funding required for board purchase:
[image]

Fourdee, 2017-03-16:
Funded by donation (Thanks Ioan!):
On order, awaiting arrival.

Fourdee, 2017-03-20:
Arrived, time to crack on 😃

Fourdee, 2017-03-20:
Notes:

- WifI ```8723bs```

Fourdee, 2017-03-20:
Performance:
- Network throughput
- Heat eff
- SD IO

Fourdee, 2017-03-20:
GPU:
Not available via repos.
```
root@DietPi:~# apt-cache search libva-rockchip1
root@DietPi:~# apt-cache search libmali-rk
```

Fourdee, 2017-03-20:
GPS installed by default on official image??
[image]

```
dpkg -l | grep gpsd
ii  gpsd                      3.16-1~bpo8+1               armhf        Global Positioning System - daemon
```

Fourdee, 2017-03-21:
Completed:
http://dietpi.com/#download

Invictaz, 2017-03-21:
Well doen. Impressive how fast you did it. Was it because the official blob for the GPU and the CPU distro already available from Rockchip?

Fourdee, 2017-03-21:
@Invictaz 

Thanks 👍 We actually used the official ASUS v1.4 image, stripped it (and then some) then optimized for DietPi.

Once we had the info on WiFi/BT modules and various other things, DietPi is coded to allow for new boards fairly quickly. Just a case of adding new code and done.

Just bear in mind, GPU features are not support yet in DietPi for this image, due to:
https://github.com/Fourdee/DietPi/issues/806#issuecomment-287760711

mmagnetix, 2017-03-21:
@Fourdee 👍 

receiving the board this week. how does the sdcard perform? the controller is UHS-I might speed things up?

Regards,
Magnetix

Invictaz, 2017-03-22:
@Fourdee 

Still too bad there is no Allwinner A10 build. I really could use it. 
Some people said that the libvdpau and GPU support is now there partially. 

See also my links on issue #824

Fourdee, 2017-03-22:
> receiving the board this week. how does the sdcard perform? the controller is UHS-I might speed things up?

@mmagnetix 
Oh yes it does, SDIO 3.0 is really nice, Sandisk Extreme 32GB 60MB/s write rated
[image]

May get better performance on higher speed card, although, I doubt it will touch Odroid's EMMC 150MB/s

mmagnetix, 2017-03-22:
@Fourdee 

Awesome. The goal of Asus is almost there (to be a Raspberry PI killer) Hopefully the GPU driver issue is corrected in the next couple of days. Then Odroid C2 has competition. 
 
the heatsink in the package is it enough to keep the CPU <85c or is did the CPU throttle kicked in?

Fourdee, 2017-03-23:
@mmagnetix 

> Hopefully the GPU driver issue is corrected in the next couple of days.

Yep, hopefully. I will test the board on another monitor, just incase its an EDID/auto detect issue.

> the heatsink in the package is it enough to keep the CPU <85c or is did the CPU throttle kicked in?

Seems fine, I did a quick 1min CPU burn (8 threads) and only reached 70'c. So in terms of real-world server usage, should never hit thermal throttling unless at full load for 10+ mins.

>  Then Odroid C2 has competition.

For me, the C2 is still a much better board. Especially for Kodi usage:
- C2: 4K resolution
- Asus: 1080p resolution (not 4K)
- C2: 990MB/s network bandwidth
- Asus: 500MB/s network bandwidth (in send direction i believe): https://github.com/Fourdee/DietPi/issues/806#issuecomment-287755120
- C2: 150MB/s R/W with EMMC module
- C2: Cheaper
- C2: Available hi-end audio DAC HATS (HiFi Shield)
- C2: Better comminuty and support by Odroid

Asus doesn't even have an official forum for the tinkerboard (that Google can find). There is no comminuty.

Hopefully things will improve, Asus have the global resources and funds. Lets wait and see :)

Fourdee, 2017-03-23:
Interesting, on another TV, i still get the same screen blank after 5 seconds in desktop, however, it does return signal after another 5 seconds.

Chromium + Youtube performance is dreadful, 720p@30fps = 5-10fps

mmagnetix, 2017-03-23:
Looks like a hdmi resolution mode issueDid you tried all resolutions to check if it stablized on say 720p.

mmagnetix, 2017-03-26:
I heard some rumors that the Tinker Board works OK (desktop/Kodi) with mainline Kernel 4.11.

Fourdee, 2017-05-11:
Update image

bpatrik, 2017-07-30:
Is there any update on this?
v1.8 stable and v1.9 beta is out since may:
https://www.asus.com/us/Single-Board-Computer/Tinker-Board/HelpDesk_Download/

bpatrik, 2017-08-07:
I found a strange bug with Dietpi, Tiner os (1.8 stable, 19, beta) but not with armbian.
If I server some files over http, the tcp connection stops sometimes and the files won't get transferred.
I've also reported this error here too, with more details:
https://tinkerboarding.co.uk/forum/thread-493.html

p.s.: after trying out most of the OS for tinker board, this is my favorite so far. Great job! :)

Fourdee, 2017-09-07:
Image updated (based on TinkerOS 2.0.1 beta): 
http://dietpi.com/downloads/images/DietPi_AsusTinkerBoard-armv7-(Stretch).7z

teravice, 2018-03-22:
do you have any update on this? Is there any hope that the GPU issue is going to be fixed?

Fourdee, 2018-08-18:
@adamotte 

Hi thanks for the request, the current DietPi image uses 2.0.5. I'll flag this for update.

adamotte, 2018-08-20:
@Fourdee, Simple question : Does DietPi support the S version (so with the eMMC) or do we need to install it on SD card ?

Fourdee, 2018-08-20:
@adamotte 

Yes it supports S version 👍 

Marking as closed in favor of:
https://github.com/Fourdee/DietPi/issues/2026

closed 2018-08-20 by Fourdee

Fourdee, 2018-08-20:
@adamotte 

TinkerOS 2.0.7:
https://dietpi.com/downloads/testing/DietPi_ASUSTB-ARMv7-Stretch.7z

Although, this release breaks WiFi. Will leave under testing.

adamotte, 2018-08-21:
Hum i still need WiFi (remote work :beach_umbrella: :tropical_drink: :sun_with_face:) so i will wait before upgrade. Anyway thanks for the update.

ravaz, 2018-09-13:
The test link for v2.0.7 doens't work anymore. It there any update on this topic?

Fourdee, 2018-09-13:
@ravaz 

2.0.7 image on its way, see https://github.com/Fourdee/DietPi/issues/2070 for link when its completed.

MichaIng, 2018-10-21:
@Fourdee 
Just found: https://tinkerboarding.co.uk/forum/thread-69.html

Beneath DietPi is listed with the still correct note that GPU features are disabled. But since this is not a note on the official TinkerOS, perhaps something has changed there or it does work on fresh TinkerOS and we break something with DietPi preparation, perhaps accidentally auto purge some required firmware package?

Fourdee, 2018-10-21:
@MichaIng 

> break something with DietPi preparation, perhaps accidentally auto purge some required firmware package?

Possibly. I also believe the install has a custom version of xserver packages, to support the GPU. However, I was unsuccessful in getting the provided packages to work.

When i'am able, i'll take another look. But low priority for the moment due to other tickets.