MichaIng/DietPi #809: Re: Please help; Ampache not functioning

mzeichner, 2017-03-14:
Reference CODE: d58612fa-b924-40a4-8500-c236a83fd7cf-0

Fourdee, 2017-03-15:
@mzeichner 
Thanks 👍 

I'll take a look.

Fourdee, 2017-03-15:
@mzeichner 

Everything appears fine, lets try a reinstall to drop sql DB and regenerate, also log output for further debugging.

#### Please run the following commands:

Uninstall and log output:
```
dietpi-software uninstall 40  | tee -a /root/log.log
```

Install and log output:
```
dietpi-software install 40 | tee -a /root/log.log
```

#### Retry http://ip/ampache, if still failing, please upload the log file:
```
/root/log.log
```

Fourdee, 2017-03-24:
No response.

@mzeichner 

I'll mark this as closed, please re-open if required.

closed 2017-03-24 by Fourdee