MichaIng/DietPi #812: No HDMI-Audio on Pi3 while forcing Audio to HDMI via config.txt

guidol70, 2017-03-16:
On the DietPi V145 with Amiberry I didnt get any Sound via HDMI in the AMIGA Emulation.
Neither in the normal HDMI-Configuration nor in the advanced/enhanced HDMI-Configuration.
I did force in the normal settting the HDMI Conenction AND in the config.txt via 
hdmi_drive=2
hdmi_force_edid_audio=1

but I did only get sound from Amiberry via the analog 3.5mm analog connector.
As test I configured a USB-Sound Device that also did work.

Today I doublechecked against the actual Amibian 1.313-230117 and there HDMI-Audio did work on the first try if the config.txt parameters are inserted.

The problem with DietPi&Amiberry has been tested here with two different Pi3 and 2 HDMI Monitors (one 4:3 Hannspree XmS (1280:1024) and a 40" Philips HDMI-TV

Fourdee, 2017-03-16:
@guidol70 

Thanks for the report 👍 

I'll try to replicate the issue at our end.

Fourdee, 2017-03-17:
Unable to replicate on fresh install. HDMI output utilized.

@guidol70 

Just few things we can check:
- During initial install, was the HDMI connected and TV powered on? RPi will output to 3.5mm if no HDMI detected.
- With these settings, applied to ```/DietPi/config.txt``` (please check) , ```hdmi_drive=2 hdmi_force_edid_audio=1``` is the sound still not functional on HDMI?

guidol70, 2017-03-17:
The HDMI was connected and powered.
Boot RPi3 did output to the 3.5mm in several HDMI-config.
Yes I did use these 2 settting in the /DietPi/config.txt - and had success
with hdmi_drive=2 on Amibian.
Didnt need hdmi_force_edid_audio=1

Normally I copy this setting out of my blog and uncomment it:
https://lehwalder.wordpress.com/category/raspberry-pi/
And on this page is typed OK....

Hmmm - couldnt check anymore (only with a new install) because I installed
Amibian over the Amiberry Card.

Kind Regards
Guido

Fourdee, 2017-03-18:
@guidol70 

> Yes I did use these 2 settting in the /DietPi/config.txt - and had success

With these settings applied, HDMI audio is now functional?

guidol70, 2017-03-18:
Yes - but in Amibian (as I did write) - but not in DietPi&Amiberry

Fourdee, 2017-03-21:
@guidol70 

Thanks 👍 

I'll run some further tests here with your settings applied:

_Note to self:
- Test with ```hdmi_drive=2``` +? ```hdmi_force_edid_audio=1```_

Fourdee, 2017-03-22:
@guidol70 

Still unable to replicate lack of HDMI audio in my tests, even with  ```hdmi_drive=2``` + ```hdmi_force_edid_audio=1```.

Can you provide me with the modified ```config.txt``` you are using on DietPi+AmiBerry, that fails with HDMI output please?

Also, can you confirm if you used the special DietPi+AmiBerry image your installation?
- http://dietpi.com/downloads/images/DietPi-AmiBerry_RPi-armv6-(Jessie).7z

guidol70, 2017-03-23:
>    armv6-(Jessie).7z

No I cant provide it anymore :( I did re-use the SD-Card with Amibian.

I can confirm I did use the special DietPi+Amiberry Image weeks
ago....after update it did show me V145

If no one else got this problem....close it. If it was a real problem to
more users they had left also a info....

Thanks for the search....

Fourdee, 2017-03-26:
@guidol70 

Thanks for the info.

Unfortunately, as I can't replicate this issue on 2 of my TV's with fresh/untouched DietPi-AmiBerry installs, there is not much I can do in regards to offering assistance at this stage.

I'll mark this as closed, however, if you have any more info on the issue, please reopen and let us know.

Just encase, I'd also recommend trying a fresh install of the DietPi-AmiBerry image, ensure HDMI connected and powered during the install stage.

closed 2017-03-26 by Fourdee