MichaIng/DietPi #813: Failed to start reboot.target: on update to 146

9H5G, 2017-03-16:
Hi - the update ran fine but this message appeared after the reboot command:

Failed to start reboot.target: Connection timed out

Broadcast message from root@AG on pts/0 (Thu 2017-03-16 12:42:23 UTC):

The system is going down for reboot NOW!

After the reboot, I'm encouraged to run the update again.

Fourdee, 2017-03-16:
@9H5G 

Thanks for the report 👍 

Appears something (SystemD service) is preventing system from rebooting gracefully.

Which device is this on? 

Lets also check status of DietPi-RAMlog/disk:
```
systemctl status dietpi-ram* -l
```

9H5G, 2017-03-16:
Hi Dan,

Its a Pi3 with one USB drive configured and a USB serial in use.

The RAMlog:

root@192.168.43.7's password:

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.

 ───────────────────────────────────────
 DietPi     | Update available
 ───────────────────────────────────────
 V145       | RPi 3 Model B (armv7l)
 ───────────────────────────────────────
 IP Address | 192.168.43.7
 ───────────────────────────────────────

 Created by : Daniel Knight
 Web        : http://DietPi.com
 Twitter    : http://twitter.com/dietpi_
 Donate     : http://goo.gl/pzISt9
 DietPi's web hosting is powered by: MyVirtualServer.com

 dietpi-launcher  = All the DietPi programs in one place.
 dietpi-config    = Feature rich configuration tool for your device.
 dietpi-software  = Select optimized software for installation.
 dietpi-update    = Run now to update DietPi (from V145 to V146).
 htop             = Resource monitor.
 cpu              = Shows CPU information and stats.

root@AG:~# systemctl status dietpi-ram* -l
● dietpi-ramlog.service - DietPi-RAMlog
   Loaded: loaded (/etc/systemd/system/dietpi-ramlog.service; enabled)
   Active: active (exited) since Thu 2017-03-16 12:43:58 UTC; 9h ago
  Process: 535 ExecStart=/bin/bash -c /boot/dietpi/dietpi-ramlog 0
(code=exited, status=0/SUCCESS)

Mar 16 12:43:57 AG bash[535]: DietPi-Ramlog: Starting
Mar 16 12:43:58 AG bash[535]: chown: invalid user: ‘debian-tor:nogroup’
Mar 16 12:43:58 AG bash[535]: DietPi-Ramlog: Completed
Mar 16 12:43:58 AG systemd[1]: Started DietPi-RAMlog.

● dietpi-ramdisk.service - DietPi-RAMdisk
   Loaded: loaded (/etc/systemd/system/dietpi-ramdisk.service; enabled)
   Active: active (exited) since Thu 2017-03-16 12:44:02 UTC; 9h ago
  Process: 691 ExecStart=/bin/bash -c /boot/dietpi/dietpi-ramdisk 0 &>>
/etc/dietpi/logs/dietpi-ramdisk.log (code=exited, status=0/SUCCESS)
  Process: 537 ExecStartPre=/bin/mkdir -p /etc/dietpi/logs (code=exited,
status=0/SUCCESS)

Mar 16 12:44:02 AG systemd[1]: Started DietPi-RAMdisk.

Kind Regards
John

Fourdee, 2017-03-18:
@9H5G 

> USB serial

Just quickly, can you test reboot with this unplugged? Have a hunch. If problems still persist, we'll need to debug systemD further.

9H5G, 2017-03-18:
I think we'll need to do some more debugging :/ Tried with the USB serial
cable disconnected but with the 2 USB drives still plugged in and it failed
exactly the same Dan.

Kind Regards
John

Fourdee, 2017-03-18:
@9H5G 

Gutted, ok lets try rsyslog 1st, see if that shows up anything. We'll need to change logging mode to ```Full```:
```dietpi-software``` > Log System > Full > ```Go >> Start install```

Once it "reboots", "reboot" again, then lets see logfile:
```
cat /var/log/daemon.log
```

9H5G, 2017-03-18:
Here you go!

Mar 18 14:28:13 AG systemd[1]: Started Synchronise Hardware Clock to System
Clock.
Mar 18 14:28:13 AG systemd[1]: Started Turns off Raspberry Pi display
backlight on shutdown/reboot.
Mar 18 14:28:13 AG systemd[1]: Unmounting /mnt/usb_2...
Mar 18 14:28:13 AG systemd[1]: Stopping system-ifup.slice.
Mar 18 14:28:13 AG systemd[1]: Removed slice system-ifup.slice.
Mar 18 14:28:13 AG systemd[1]: Stopping system-systemd\x2dfsck.slice.
Mar 18 14:28:13 AG systemd[1]: Removed slice system-systemd\x2dfsck.slice.
Mar 18 14:28:13 AG systemd[1]: Stopping System Time Synchronized.
Mar 18 14:28:13 AG systemd[1]: Stopped target System Time Synchronized.
Mar 18 14:28:13 AG systemd[1]: Stopping Graphical Interface.
Mar 18 14:28:13 AG systemd[1]: Stopped target Graphical Interface.
Mar 18 14:28:13 AG systemd[1]: Stopping Multi-User System.
Mar 18 14:28:13 AG systemd[1]: Stopped target Multi-User System.
Mar 18 14:28:13 AG systemd[1]: Stopping OpenBSD Secure Shell server...
Mar 18 14:28:13 AG systemd[1]: Stopping Login Prompts.
Mar 18 14:28:13 AG systemd[1]: Stopped target Login Prompts.
Mar 18 14:28:13 AG systemd[1]: Stopping Getty on tty1...
Mar 18 14:28:13 AG systemd[1]: Stopping Login Service...
Mar 18 14:28:13 AG systemd[1]: Stopping LSB: Start or stop the Webmin
server...
Mar 18 14:28:13 AG systemd[1]: Stopping LSB: Autogenerate and use a swap
file...
Mar 18 14:28:13 AG systemd[1]: Stopping LSB: Brings up/down network
automatically...
Mar 18 14:28:13 AG systemd[1]: Stopping D-Bus System Message Bus...
Mar 18 14:30:32 AG systemd-modules-load[135]: Inserted module 'fuse'
Mar 18 14:30:32 AG systemd-modules-load[135]: Inserted module 'i2c_bcm2708'
Mar 18 14:30:32 AG systemd-modules-load[135]: Inserted module 'i2c_dev'
Mar 18 14:30:32 AG systemd[1]: Mounted FUSE Control File System.
Mar 18 14:30:32 AG systemd[1]: Mounted Configuration File System.
Mar 18 14:30:32 AG systemd[1]: Started File System Check on Root Device.
Mar 18 14:30:32 AG systemd[1]: Started Apply Kernel Variables.
Mar 18 14:30:32 AG systemd[1]: Started Create Static Device Nodes in /dev.
Mar 18 14:30:32 AG systemd[1]: Starting udev Kernel Device Manager...
Mar 18 14:30:32 AG systemd-fsck[110]: e2fsck 1.42.12 (29-Aug-2014)
Mar 18 14:30:32 AG systemd-fsck[110]: rootfs: clean, 68059/872032 files,
321920/3872384 blocks
Mar 18 14:30:32 AG fake-hwclock[118]: Sat 18 Mar 14:30:22 UTC 2017
Mar 18 14:30:32 AG systemd[1]: Started udev Kernel Device Manager.
Mar 18 14:30:32 AG systemd[1]: Starting Copy rules generated while the root
was ro...
Mar 18 14:30:32 AG systemd[1]: Starting LSB: Tune IDE hard disks...
Mar 18 14:30:32 AG systemd[1]: Starting LSB: Set preliminary keymap...
Mar 18 14:30:32 AG systemd[1]: Started Copy rules generated while the root
was ro.
Mar 18 14:30:32 AG hdparm[193]: Setting parameters of disc:  /dev/sda.
Mar 18 14:30:32 AG systemd[1]: Started LSB: Tune IDE hard disks.
Mar 18 14:30:32 AG keyboard-setup[196]: Setting preliminary keymap...done.
Mar 18 14:30:32 AG systemd[1]: Found device /dev/mmcblk0p1.
Mar 18 14:30:32 AG systemd[1]: Started LSB: Set preliminary keymap.
Mar 18 14:30:32 AG systemd[1]: Starting Remount Root and Kernel File
Systems...
Mar 18 14:30:32 AG systemd[1]: Starting File System Check on
/dev/mmcblk0p1...
Mar 18 14:30:32 AG systemd[1]: Started Remount Root and Kernel File Systems.
Mar 18 14:30:32 AG systemd[1]: Started Various fixups to make systemd work
better on Debian.
Mar 18 14:30:32 AG systemd[1]: Starting Load/Save Random Seed...
Mar 18 14:30:32 AG systemd[1]: Starting Local File Systems (Pre).
Mar 18 14:30:32 AG systemd[1]: Reached target Local File Systems (Pre).
Mar 18 14:30:32 AG systemd[1]: Mounting /DietPi...
Mar 18 14:30:32 AG systemd[1]: Mounting /tmp...
Mar 18 14:30:32 AG systemd[1]: Mounted /DietPi.
Mar 18 14:30:32 AG systemd[1]: Mounted /tmp.
Mar 18 14:30:32 AG systemd[1]: Started Load/Save Random Seed.
Mar 18 14:30:32 AG systemd[1]: Found device Cruzer_U primary.
[... 259 lines cut ...]
Mar 18 14:32:13 AG node-red[1231]: 18 Mar 14:32:13 - [info] Settings file
 : /mnt/dietpi_userdata/node-red/settings.js
Mar 18 14:32:13 AG node-red[1231]: 18 Mar 14:32:13 - [info] User directory
: /mnt/dietpi_userdata/node-red
Mar 18 14:32:13 AG node-red[1231]: 18 Mar 14:32:13 - [info] Flows file
: /mnt/dietpi_userdata/node-red/flows_AG.json
Mar 18 14:32:13 AG node-red[1231]: 18 Mar 14:32:13 - [info] Starting flows
Mar 18 14:32:13 AG node-red[1231]: 18 Mar 14:32:13 - [warn] [udp
in:a0fc2006.31de7] udp: port already in use
Mar 18 14:32:13 AG node-red[1231]: 18 Mar 14:32:13 - [info] Started flows
Mar 18 14:32:13 AG node-red[1231]: 18 Mar 14:32:13 - [info] [udp
out:608e8a58.58cdc4] udp broadcast ready: 2394 -> 192.168.43.255:2394
Mar 18 14:32:14 AG node-red[1231]: 18 Mar 14:32:14 - [info] [udp
in:a0fc2006.31de7] udp listener at 0.0.0.0:2394
Mar 18 14:32:14 AG node-red[1231]: 18 Mar 14:32:14 - [info] Server now
running at http://127.0.0.1:1880/
Mar 18 14:32:14 AG node-red[1231]: 18 Mar 14:32:14 - [info] serial port
/dev/ttyUSB0 opened at 38400 baud 8N1
Mar 18 14:32:14 AG node-red[1231]: 18 Mar 14:32:14 - [info]
[mqtt-broker:dc5d5c79.f3a42] Connected to broker: mqtt://127.0.0.1:1883
Mar 18 14:32:15 AG dhclient: DHCPDISCOVER on eth0 to 255.255.255.255 port
67 interval 4

Kind Regards
John

Fourdee, 2017-03-19:
@9H5G 

Thanks for the log. Cant see anything out of the ordinary, SystemD reports no issues:
```
Mar 18 14:28:13 AG systemd[1]: Stopping D-Bus System Message Bus...

#REBOOT

Mar 18 14:30:32 AG systemd-modules-load[135]: Inserted module 'fuse'
```

Did the system reboot successfully on its own?

Ok, think its best we get a bug report to allow us to check system configuration/settings. Please complete the following:
http://dietpi.com/phpbb/viewtopic.php?f=8&t=499#p2184

9H5G, 2017-03-20:
Sorted - cd245aa7-6883-413c-b41e-adee2a86a612-0

9H5G, 2017-03-24:
The update to 147 ran without error. Rebooting shows v147. I think we can close this one.

Fourdee, 2017-03-24:
@9H5G 

Thanks. 👍 

Still unsure as to what the cause of this was. But if it reoccurs, please reopen ticket and we'll investigate it further.

closed 2017-03-24 by Fourdee