MichaIng/DietPi #814: DietPi Image | NanoPi Neo 2

Fourdee, 2017-03-16:
- 1GBit eth
- 64bit H5 ARM CPU

http://www.friendlyarm.com/index.php?route=product/product&product_id=180

Awaiting delivery.

Fourdee, 2017-03-28:
ARMbian dev branch, lets hold packages:
```
apt-mark hold armbian-firmware armbian-tools-jessie linux-u-boot-nanopineo2-dev
```

Fourdee, 2017-03-28:
Notes:
- 🈯️ H5 is much more responsive than the H3. Doesn't have a noticeable "delay" in processing/command exec the H3 has.

Fourdee, 2017-03-29:
Completed, image live on http://dietpi.com

closed 2017-03-29 by Fourdee

southwolf, 2017-07-17:
[Wiki page](http://wiki.friendlyarm.com/wiki/index.php/NanoPi_NEO2)

[Official Images](https://www.mediafire.com/folder/ah4i6w029912b/NanoPi-NEO2)

AFAIK, `/sys/devices/system/cpu/cpufreq` is not working. [Ref](http://linux-sunxi.org/Cpufreq)