MichaIng/DietPi #815: DietPi Image | OPi Lite

Fourdee, 2017-03-17:
Dev sample en route (thanks Steven), awaiting arrival.

joaofl, 2017-03-17:
@Fourdee 

Do you think you can share some of the steps on porting and creating an image from dietpi to a new board?

I got a sample from this Roseapple pi, which I would like to give it a try. It has USB3 at least.

http://roseapplepi.org/

Fourdee, 2017-03-17:
@joaofl 

> Do you think you can share some of the steps on porting and creating an image from dietpi to a new board?

Yep, you'll need to start with a minimal Debian Jessie image, or, build one with ARMbian build tools (https://github.com/igorpecovnik/lib). I can compile this if needed, just let me know.

Then you can prep system for DietPi:
https://github.com/Fourdee/DietPi/blob/master/PREP_SYSTEM_FOR_DIETPI.sh

joaofl, 2017-03-19:
@Fourdee 
I'll give it a try on my spare time then. I'm really interested on learning more about..

For this reason I started another thread for that only: https://github.com/Fourdee/DietPi/issues/822
Thanks for the support!

Fourdee, 2017-03-24:
Board arrived.

Fourdee, 2017-03-24:
- Excellent signal strength/range with onboard WiFi, impressive:
```
wlan0     IEEE 802.11bgn  ESSID:"Mi6"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:2.447 GHz  Access Point: E8:DE:27:4A:D0:66
          Bit Rate:150 Mb/s   Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:****-****-****-****-****-****-****-****  
          Power Management:off
          Link Quality=98/100  Signal level=94/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```

Fourdee, 2017-03-24:
Completed.

closed 2017-03-24 by Fourdee