MichaIng/DietPi #818: RPi Zero W | onboard WiFi not found

rulenma, 2017-03-18:
Hi,

hopefully Iam right here

#787

doesn`t work for me.
Iam running a fresh installation of 146 on the RPi zero w and it says "wlan0 No such device"
But in the config Wifi is enabled.
Anything else that I can do?
Iam searching for the right command to see if I can see that device, but I don't know the command and the name of the device.

Anyone has experience with the wifi modul on the zero w?
They say it`s the same modul as on the RPi3 but I compared both and on the zero w its just bad.
The Access Point can be in the same room and the zero w is 10m away and the signal quality is worst or it can't see the wifi.

Any ideas?
regards :)

Fourdee, 2017-03-19:
@rulenma 

Thanks for the report 👍 

We identify onboard RPi WiFi capability by hardware revision code. I'am just wondering if your Zero W has a new/unknown code to us.

Please can you run the following command, and paste result:
```
cat /proc/cpuinfo | grep 'Revision' | awk '{print $3}'
```

rulenma, 2017-03-19:
Thanks for the reply.

`root@DietPi:~# cat /proc/cpuinfo | grep 'Revision' | awk '{print $3}'`
`9000c1`

Fourdee, 2017-03-19:
@rulenma 

Thanks 👍 

> 9000c1

Appears to be correct, must be something else throwing this out. We'll need to debug.
We can check some of the settings/debug commands with DietPi bug report. 

Please see the following to send me a bug report:
http://dietpi.com/phpbb/viewtopic.php?f=8&t=499#p2184

Fourdee, 2017-03-19:
@rulenma 

Ok, lets try re enabling WiFi modules again, please run the following commands:
```
/DietPi/dietpi/func/dietpi-set_hardware wifi onboard_enable
/DietPi/dietpi/func/dietpi-set_hardware wifi enable
reboot
```

Now try dietpi-config again and attempt to enter WiFi menu:
```
dietpi-config
```

If this still reports no hardware found, please take a screenshot/picture of the following commands:
```
ifconfig -a
lsmod
iwconfig
```

rulenma, 2017-03-19:
The eth0 is an usb ethernet adapter.
Here the screenshots:

[image]
[image]
[image]

Fourdee, 2017-03-19:
@rulenma 

Thanks. 

> The eth0 is an usb ethernet adapter.

Please unplug this and reboot system. then check ```dietpi-config``` again.

rulenma, 2017-03-19:
@Fourdee 

Thanks, but same results.
Now I see some errors with the eth0, that the device could not be found...
Is there any way to install it fresh but with 146?
Didn`t find the image yet, so I needed to use the usb ethernet adapter.

Any more ideas?
In raspbian it works fine. But I want to use it with DietPi because its so great <3 and I almost have it on all my devices :-)

Fourdee, 2017-03-20:
@rulenma 

> Thanks, but same results.
Is there any way to install it fresh but with 146?
Didn`t find the image yet, so I needed to use the usb ethernet adapter.

Ok thanks.

Our updated RPi image (DietPi_v145_RPi-armv6-(Jessie).img) contains the Zero W WiFi support and "should" be functional. Just to make sure latest image is used, please can you re-download this image and write it again?

Direct link to image DL: http://dietpi.com/downloads/images/DietPi_RPi-armv6-(Jessie).7z

If problems persist:
- SD card used
- PSU used
- Zero W revision number (on PCB)

> In raspbian it works fine. But I want to use it with DietPi because its so great <3 and I almost have it on all my devices :-)

We'll get it working. I did run a test on my Zero W with the DietPi image above, no issues at our end, but i'll take another look just incase.

rulenma, 2017-03-20:
@Fourdee 

Thanks a lot for that.
Now it`s working. But I don`t know why it didn`t work in the other version...

What is your experience with the wifi modul in the zero w, if I may ask.
My zero is about 1m away from the WiFi AP and Iam having lots of ping timeouts. The signal is also bad. Better not talk about the rest :-)
Regards

Fourdee, 2017-03-20:
@rulenma 

> Now its working. But I dont know why it didn`t work in the other version...

Zero W support was added to the v145 image with modified date 8th May. Is it possible you used an image that was older than this? In which case WiFi would not be supported on Zero W.

If not, may just be a bad image write.

> My zero is about 1m away from the WiFi AP and Iam having lots of ping timeouts. The signal is also bad. Better not talk about the rest :-)

Mine runs really well, no issues so far and similar range results to RPi 3 onboard. 

May be worth checking ```iwconfig``` for signal stats on the Zero W.

Fourdee, 2017-04-12:
@rulenma 

> Now its working.

I'll mark this as resolved. Please reopen if required.

esturniolo, 2018-04-21:
Hi.
I'm with the same problem here.
Pi Zero, last version download from the web and no luck.
I tried to download the version that you posted but it's unavailable.

This command shows:
```
# cat /proc/cpuinfo | grep 'Revision' | awk '{print $3}'
900093
```

And I can't find the commands that you suggest

```
/DietPi/dietpi/func/dietpi-set_hardware
```

I take "screenshots" with my cell phone (I don't have any dongle to connect it) from the commands that you asked before.
Please, take a look at http://emi.st/i/album/B8V too see all the images.

I hope that you can't find how to solve this.

Thanks for your patience.

Fourdee, 2018-04-22:
@esturniolo 

Thanks for the report 👍 

HW revision code checks out. We'll try to replicate in local testing.

Fourdee, 2018-04-24:
Test:
- RPi Zero W
- Fresh install
- Local keyboard
- Prompt to setup connection > WiFi
- Hardware found
- Scan + connects fine

```
root@DietPi:~# cat /proc/cpuinfo | grep 'Revision' | awk '{print $3}'
9000c1
```

@esturniolo 
```900093``` is not a Pi Zero W (https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md), which WiFi dongle are you using? ```lsusb```

Jelmer-H, 2018-04-24:
Hi, I also have an issue with my freshly arrived RPI Zero W and the wifi adapter. In flashed the 6.6 image and upgraded to 6.7 using a USB ethernet adapter.

 Strange thing is that my unit gives 000f as revision.

```
root@DietPi:/DietPi/dietpi# cat /proc/cpuinfo 
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 898.66
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2835
Revision	: 000f
Serial		: 000000007b412a42
```

I checked the back of my device and it reads: Raspberry Pi Zero W V1.1 Any idea what might be wrong here?

Fourdee, 2018-04-25:
@Jelmer-H 

> Pi Zero W's incorrectly programmed at manufacturing, and the 000f Revision is one of the symptoms.

Interesting!

Seems RPi manufacturing quality control has recently dropped. My RPi 3 B+ ethernet has a hardware fault on day one:
https://github.com/Fourdee/DietPi/issues/1627#issuecomment-374669173

esturniolo, 2018-04-25:
Ops.

> @esturniolo
> 900093 is not a Pi Zero W (https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md), which WiFi dongle are you using? lsusb

My bad!!! I was test the wrong Zero! I feel like a stupid. Sorry @Fourdee for wasting your time.

Fourdee, 2018-04-26:
@esturniolo 

> I was test the wrong Zero! 

No worries, i've done the same thing in the past 😃

Fourdee, 2018-04-26:
Marking as closed, unable to replicate (https://github.com/Fourdee/DietPi/issues/818#issuecomment-383854092).

Please reopen if required.

sapdemon, 2018-07-25:
I`ve tried switch on onboard wifi, and i done this by next commands

modprobe -rf cfg80211
modprobe -rf brcmfmac
modprobe cfg80211
modprobe brcmfmac

I`ve added the screen with the result of this operations

Fourdee, 2018-07-25:
@sapdemon 

Hi,

Thanks for the report 👍 

I'll run a local test here, see if we can replicate.

Would it also be possible for you to upload the edited ```dietpi.txt``` file used? **Please remove any WiFi keys before hand.**

Fourdee, 2018-07-25:
🈯️ Unable to replicate any issues with WiFi enable in ```dietpi.txt``` on RPi Zero W, id of ```9000c1```.

```dmesg``` output
```
[   33.212510] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[   33.255021] brcmfmac: brcmf_fw_map_chip_to_name: using brcm/brcmfmac43430-sdio.bin for chip 0x00a9a6(43430) rev 0x000001
[   33.255399] usbcore: registered new interface driver brcmfmac
[   33.860445] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Oct 23 2017 03:55:53 version 7.45.98.38 (r674442 CY) FWID 01-e58d219f
[   33.861746] brcmfmac: brcmf_c_preinit_dcmds: CLM version = API: 12.2 Data: 7.11.15 Compiler: 1.24.2 ClmImport: 1.24.1 Creation: 2014-05-26 10:53:55 Inc Data: 9.10.39 Inc Compiler: 1.29.4 Inc ClmImport: 1.36.3 Creation: 2017-10-23 03:47:14
```

@sapdemon 

Same as below, could be firmware related:
https://github.com/raspberrypi/linux/issues/2274#issue-274863888

Please answer:
- DietPi image version used (once you extract the image, it should contain a v6.X)?
- Paste results of ```ls -lha /lib/firmware/brcm```
- Any changes made to ```config.txt```?
- PSU used
- SD card used

Fourdee, 2018-07-26:
@sapdemon 

Thanks, firmware looks good, so that is not the issue.

Appears to be some solder burn: https://photos.app.goo.gl/VvTiuwX9FUfJzT9XA May of caused a hardware failure on the board.
To verify this, do you experience the same issue with official Raspbian Images?

sapdemon, 2018-07-27:
@Fourdee 
Thx.
I will install official Raspbian Image in the next week and write here the result.

Fourdee, 2018-08-10:
No response, unable to replicate, marking as closed. Please reopen if required.

closed 2018-08-10 by Fourdee