MichaIng/DietPi #822: DietPi Image | Roseapple Pi

joaofl, 2017-03-19:
Some time ago I got a sample of the [Roseapple Pi](http://roseapplepi.org/index.php/spec/) board, which is another Pi kind of board, but counts with  USB3 and 2Gb DDR3 RAM. They were given away many samples of their board, in a attempt to get to the market.

In this sense, would be nice to have DietPi for it, and I would like to contribute on it if I can.

From: https://github.com/Fourdee/DietPi/issues/815
> @joaofl
> 
> Do you think you can share some of the steps on porting and creating an image from dietpi to a new board?
> 
> Yep, you'll need to start with a minimal Debian Jessie image, or, build one with ARMbian build tools (https://github.com/igorpecovnik/lib). I can compile this if needed, just let me know.
> 
> Then you can prep system for DietPi:
> https://github.com/Fourdee/DietPi/blob/master/PREP_SYSTEM_FOR_DIETPI.sh

joaofl, 2017-03-19:
@Fourdee 

> Yep, you'll need to start with a minimal Debian Jessie image, or, build one with ARMbian build tools (https://github.com/igorpecovnik/lib). I can compile this if needed, just let me know.

He actually already have it compiled, with kernel v3.10. Not sure if there are newer releases by the manufacturer, and if it is worth trying anything like that, and if there is any need of recompiling it?

Let me see how far can I go on those steps from PREP_SYSTEM file,  and in case of doubts, I bother you here :)

joaofl, 2017-03-20:
With current Armbiam release, I could only get 18.4Mb/s transfer through USB3 using the command:
sync; dd if=/dev/zero of=tempfile bs=1M count=2048; sync

It also constantly produces the following warning:

> [ 5186.626714] drivers/video/owl/backlight/owl_pwm_bl.c/owl_backlight_is_on: get backlight failed!

It seems that some changes were made in the kernel to get rid of this error. Haven't seen anything about the USB limitation so far.

This is the repo with the latest kernel source: https://github.com/xapp-le/kernel

Current release:
ARMBIAN 5.26 stable Debian GNU/Linux 8 (jessie) 3.10.105-s500

I'll do some further research and report here.

Fourdee, 2017-03-20:
@joaofl 

Looks like your on a roll, keep us updated 👍 

I did at purchasing one of these boards (specs look really good on paper), however, page not found lol :frowning_face:
http://roseapplepi.org/index.php/purchase/
[image]

joaofl, 2017-03-20:
@Fourdee 
They seem to be a bit decadent. Like Igor was saying, some of this companies rely on community development for their business to work... Pity.

I can provide you the email I asked the sample for, and I'm sure they provide you one in case they are still giving some away.

joaofl, 2017-03-21:
@Fourdee 
Could you get USB3 to work full speed on this Sparky Allo board? https://github.com/Fourdee/DietPi/issues/699

Because they use the same chip as Roseapple Pi. The issue is that at the armbian distro uses the kernel from lemaker, which does not support USB3.
https://github.com/LeMaker/linux-actions

It seems that the kernel by sparky (the one used on Sparky Allo) support USB3... At least there are some commits supposed to fix that.
https://github.com/sparkysbc/Linux/commit/3867b3fd05d2998a5503bb6c72a4946b5580a563

However the sparky does not work on the Roseapple Pi out of the box. Ethernet works, but none of the USB ports work, but I still can't understand why. Do you have any clue?

Fourdee, 2017-03-21:
@joaofl 

> I can provide you the email I asked the sample for, and I'm sure they provide you one in case they are still giving some away.

Please if you could, worth a shot daniel.knight@dietpi.com

> Could you get USB3 to work full speed on this Sparky Allo board?

I dont have any USB3 devices that are capable of testing anywhere near full potential.  I've ordered a USB3 flash drive 150MB/s, should be here tomorrow and i'll test then :)

> However the sparky does not work on the Roseapple Pi out of the box. Ethernet works, but none of the USB ports work, but I still can't understand why. Do you have any clue?

Different ARM chips and most likely different USB device paths on both SBC? although, no idea if i'am honest, you probably know more than I do when it comes to kernel code/modification 😉

joaofl, 2017-03-21:
@Fourdee 
> you probably know more than I do when it comes to kernel code/modification
This isn't true... I'm pretty much venturing into this area in order to learn. My deal is with low level embedded systems (no OS, or few Kb ones), but I want to get to know linux more in depth.

But in any case, If I cant figure out how to make the USB3 to work any bit faster, I'll try to get the dietpi image done for the board it as it is by armbian.

Another doubt: How did you get the sparky kernel compiled? Since armbian does't use it, I found way too hard to get it done.

Thanks!

Fourdee, 2017-03-22:
@joaofl 

> How did you get the sparky kernel compiled? Since armbian does't use it, I found way too hard to get it done.

We used a premade uboot/kernel provided by Allo on their Ubuntu image, then replaced RootFS with Jessie and added systemD init to ```uenv.txt```:
```init=/lib/systemd/systemd```

joaofl, 2017-03-22:
@Fourdee 

I was trying to build the DietPi image yesterday, but I had an issue when I got to:

`#FSTAB
cp /boot/dietpi/conf/fstab /etc/fstab
systemctl daemon-reload
mount -a`

`/dev/mmcblk0p1  /boot           auto    defaults,noatime,discard	0 2
/dev/mmcblk0p2 / auto defaults,noatime,discard  0 1`

In this code, you assume I already have 2 partitions, one with the boot, right? But in fact, from Armbian's image, I only have the mmc1 mounted at the root "/".

What should I do in this case? Keep with one partition with the "/boot" folder in it, or make a second partition with that content?

If I have to make the partition, is there any specific procedure you recommend?

Another thing is: The board ID... Should I make a pull request with it, or can you do it?

Thanks.

joaofl, 2017-03-23:
> ARMbian premade (Jessie server) image: http://www.armbian.com/banana-pi-m2-plus/ (1 partition)
> ARMbian build tools: https://github.com/igorpecovnik/lib#how-to-build-my-own-image-or-kernel . Add BOOTSIZE=64 to lib/configuration.sh before you run ./compile.sh to create a 2 partition image. Theres also a bug in the first run script of ARMbian when building with 2 partitions (igorpecovnik/lib#290 (comment)). You'll need to do the following before powering on:
> 
> #After compile
> ```
> - write image.
> - mount sd/image 2nd parition to modify files
> - Disable (comment) do_expand_rootfs() from /etc/init.d/firstrun
> - Disable (comment) ln -sf /boot/bin/*.bin /boot/script.bin from /etc/init.d/firstrun
> - Copy the correct .bin if required: eg: cp /boot/bin/orangepipc.bin /boot/script.bin
> ```
> Power on

@Fourdee 
I saw that you had this problem before in https://github.com/Fourdee/DietPi/issues/345.

The thing is, I can not find `do_expand`, neither  `ln -sf /boot/bin/*.bin /boot/script.bin `

It seems to have moved to /etc/init.d/resize2fs. So I commented this two lines:

```
	# first stage - resize the partition
	#[[ ! -f /var/lib/armbian/resize_second_stage ]] && do_expand_partition

	# second stage - resize the filesystem
	#[[ ! -f /var/run/resize2fs-reboot ]] && do_expand_filesystem
```
It made the image (sub optimal shrink) and it booted fine. Now I'm installing some applications to test.

I used the ID=100, since the last added board group is 90.

The colors are a bit greener than normal. No colors at all. Is it some sort of theme? It looks like Armbian, including boot logo etc... Running stable for now. Want me to share the image with you?
Recommend testing or doing anything else?

Any reply regarding the sample?

joaofl, 2017-03-23:
I keep getting this error:

[   30.600191] drivers/video/owl/backlight/owl_pwm_bl.c/owl_backlight_is_on: get backlight failed!

root@DietPi:~# lsmod 
Module                  Size  Used by
ethernet               46159  0 

I'm using ID 70 from sparky for now. All dietpi tools I've tested so far work.

joaofl, 2017-03-23:
Got some surprising performance numbers:

[image]

After all, USB3 works, for reading only! Does it make any sense?

Fourdee, 2017-03-23:
@joaofl 

Great work, looking good 👍 

> I used the ID=100, since the last added board group is 90.

I assigned 100 for the Asus tinker board, so next group would be 110 for this board: https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-obtain_hw_model#L105-L109

https://github.com/Fourdee/DietPi/pull/828 also needs to be sent to the ```testing``` branch.

As ```HW_MODEL``` 110 is to be used, you'll also need to increase the array size for DietPi-Software to ```110```:
https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-software#L266. This is mainly to allow us to define if a peice of software is available, or not, for specific ```HW_MODEL```: https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-software#L517

> Any reply regarding the sample?

No word yet, will keep an eye for it.

> [ 30.600191] drivers/video/owl/backlight/owl_pwm_bl.c/owl_backlight_is_on: get backlight failed!

Looks like a LCD addon module, but appears ARMbian has built this driver into the kernel, rather than the module, so a blacklist module would probably have no effect?
It may need to be edited at kernel stage during ARMbian compile, to disable: https://github.com/igorpecovnik/lib#how-to-change-a-kernel-configuration

Although, I wouldn't spend too much time on this personally, looks like a warning because expected device is not available, not critical to operation.

We actually had a similar issue with Sparky and flaky module causing havoc in ```dmesg```:
https://github.com/Fourdee/DietPi/blob/testing/PREP_SYSTEM_FOR_DIETPI.sh#L494-L497

> Should I share the image with you?

Yep, if you can make those changes to ```HW_MODEL```, please send it over 👍

joaofl, 2017-03-24:
Thanks @Fourdee 

I'll give one last try on disabling the backlight thing on the kernel, then I'll send you the image. I had a bit of trouble resizing the image. Can you give a tip o this?

Using gparted (UI), but it does not allow resizing the image too small, to keep only 50MB free. 
This because it uses resize2fs without the -f option.

So I did it manually `sudo resize2fs -p -f /dev/sdb2 485000K`

But then, the file system is resized, but not the logical partition. I know there is a simple way of resizing it, but I could not find it straight on.

Fourdee, 2017-03-25:
@joaofl 

> Using gparted (UI), but it does not allow resizing the image too small, to keep only 50MB free.
This because it uses resize2fs without the -f option.

Yep, sometimes you need to resize it a few times in Gparted. eg: reports 500MB min, try 1000MB, then try 800MB etc.
This normally occurs when the image has multiple small files. I believe Gparted is simply reading the size of each file, instead of calculating the actual file size on disk, although I could be wrong.

I made a little script for image resize/finalization, it:
- Requires a ```.img``` file (runs in loopback)
- Copies all data from p1/p2 to ```/root```, allowing for partition recreation (avoids potential partition corruptions on image read)
- Copies data back to new partitions
- Allows for resize in Gparted 

```
#!/bin/bash
{

	#check packages are installed
	if (( ! $(dpkg -l | grep -ci -m1 'gparted') )); then
		apt-get install gparted zerofree -y
	fi

	#-------------------------------------------------------------------------------
	#MUST LEAVE 50MB+ space or automation autologin may fail due to 0 free space!!!
	#-------------------------------------------------------------------------------

	#-------------------------------------------------------------------------------
	#ARMbian recreating partitions. Boot failure due to UUID change. must also update /boot/armbianEnv.txt
	#-------------------------------------------------------------------------------

	#XU4
	IMAGE_FP='/mnt/samba/#Backups/_Daniel/Projects/DietPi'

	IMAGE_NAME='DietPi_v147_OrangePiLite-armv7-(Jessie)'
	IMAGE_NAME+='.img'

	if [ ! -f "$IMAGE_FP/$IMAGE_NAME" ]; then

		read -p "No file found..."
		exit

	fi

	modprobe loop
	losetup -f
	losetup /dev/loop2 "$IMAGE_FP/$IMAGE_NAME"
	partprobe /dev/loop2

	e2fsck -f /dev/loop2p1
	e2fsck -f /dev/loop2p2

	#Mount loopback.
	mkdir -p /mnt/loop2p1
	mkdir -p /mnt/loop2p2
	mount /dev/loop2p1 /mnt/loop2p1
	mount /dev/loop2p2 /mnt/loop2p2

	echo -e "\ncopying data to $HOME\n"
	rm -R $HOME/loop2p1
	rm -R $HOME/loop2p2

	cp -R /mnt/loop2p1 $HOME/
	cp -R /mnt/loop2p2 $HOME/
	rm -R $HOME/loop2p1/lost+found
	rm -R $HOME/loop2p2/lost+found

	sync

	umount /mnt/loop2p1
	umount /mnt/loop2p2

	#recreate partitions
	echo -e "\nRecreate partitions\n"
	gparted /dev/loop2

	sync

	#transfer back
	mount /dev/loop2p1 /mnt/loop2p1
	mount /dev/loop2p2 /mnt/loop2p2

	echo -e "\nCopying data back\n"
	cp -R $HOME/loop2p1/* /mnt/loop2p1/
	cp -R $HOME/loop2p2/* /mnt/loop2p2/

	read -p "Data copied back to /mnt/loop2px, make changes to files if required..."

	sync

	umount /mnt/loop2p1
	umount /mnt/loop2p2

	#recreate partitions
	echo -e "Resize partitions"
	gparted /dev/loop2

	sync

	read -p "If failed press CTRL+C to exit and prevent further action, else, press any key to continue..."

	zerofree -v /dev/loop2p1
	zerofree -v /dev/loop2p2

	losetup -d /dev/loop2

	END_PARTITION=$(( $(fdisk -l "$IMAGE_FP/$IMAGE_NAME" | grep ".img2" | awk '{print $3}') + 1 ))
	END_PARTITION=$(( $END_PARTITION * 512 ))

	truncate --size=$END_PARTITION "$IMAGE_FP/$IMAGE_NAME"

	echo -e "\nCleaning up $HOME/loop2px\n"
	rm -R $HOME/loop2p1
	rm -R $HOME/loop2p2

	read -p "Done, press any key to continue..."

	exit

}
```

joaofl, 2017-03-28:
@Fourdee :+1:  Your script works nice. 

If installed before the user interface (tasted with MATE), the GPU wont work.
One have to remove it from the blacklist, and reinstall it (from http://wiki.lemaker.org/LeMaker_Guitar:GPU_hardware_acceleration).
After enabled the GPU driver, MATE loads fine, but not KODI. However, I did not test if it is working fine yet. So it was not included in the image and have to be installed manually later.

Disk image file:
https://mega.nz/#!oBMSnDQA!djOYQpYxdIfl6b-13vtqJ0SR6mqZCZ_Heo71TG1q7b8

Compressed:
https://mega.nz/#!YFFGTLrD!eRlF3efC6A0Jm_PA-DEYxRPWwIHXp9XDg9IplwVxuH4

Fourdee, 2017-03-28:
@joaofl 

Excellent, great work on this 👍 

Unfortunately, I cant test it here yet (no word on sample board). Although, i've check the image contents, looks spot on.

Is the image good to go? I'll get this hosted on dietpi.com and update site with this release.

joaofl, 2017-03-28:
@Fourdee 
I guess they are bankrupt now, and can not afford it :(
But it may be useful for people who got this board somehow. I hope it is.

Yes, It seems to be good to go. Maybe I forgot to delete bash history, but thats not a big deal anyway.

joaofl, 2017-03-28:
I intend to add the correct CPU temperature info, which is still not shown by "cpu" command.
Maybe for the next release? Show I keep this issue open?

Fourdee, 2017-03-29:
@joaofl 

> But it may be useful for people who got this board somehow. I hope it is.

👍 It will be.

> Yes, It seems to be good to go. I have tasted the installation of MATE desktop and a couple of other scripts, and worked straight on. Maybe I forgot to delete bash history, but thats not a big deal anyway.

Excellent, i'll get this uploaded and dietpi.com updated

> I intend to add the correct CPU temperature info, which is still not shown by "cpu" command.
Maybe for the next release? Show I keep this issue open?

Probably best to create another ticket for this. I'll reopen this ticket to remind me for dietpi.com update

Fourdee, 2017-03-29:
@joaofl 

Image is live 👍 :
http://dietpi.com/download

Stats bar, same CPU as Sparky SBC. Is this clocked at 1.1GHz also? And, whats heat like idle/load?

joaofl, 2017-03-29:
@Fourdee 

> this clocked at 1.1GHz also
Yes. I was always touching the heatsink to feel the temperature, and honestly, it runs really cool. I read some people having overheating problems by running it at 1.3 GHz.

> Stats bar, same CPU as Sparky SBC
I'll fix it and push to test asap.

Fourdee, 2017-03-31:
@joaofl 

I'll mark this as closed (image is now live) 👍 

For reference:
https://github.com/Fourdee/DietPi/issues/848

closed 2017-03-31 by Fourdee