MichaIng/DietPi #823: OPi Plus | Image > EMMC issues

Fourdee, 2017-03-20:
As reported by user, run local test to confirm:
http://dietpi.com/phpbb/viewtopic.php?f=11&t=1507&p=6259#p6259

Fourdee, 2017-03-21:
```
root@DietPi:~# blkid
/dev/mmcblk0: PTUUID="f8208de9" PTTYPE="dos"
/dev/mmcblk0p1: SEC_TYPE="msdos" LABEL="BOOT" UUID="DB92-1874" TYPE="vfat" PARTUUID="f8208de9-01"
/dev/mmcblk0p2: UUID="bce783b9-63cf-42c3-bf1c-ace6ec327f03" TYPE="ext4" PARTUUID="f8208de9-02"
/dev/mmcblk1: PTUUID="f8208de9" PTTYPE="dos"
/dev/mmcblk1p1: SEC_TYPE="msdos" LABEL="BOOT" UUID="DB92-1874" TYPE="vfat" PARTUUID="f8208de9-01"
/dev/mmcblk1p2: UUID="bce783b9-63cf-42c3-bf1c-ace6ec327f03" TYPE="ext4" PARTUUID="f8208de9-02"
```

DenisFromHR, 2017-03-21:
My isue is with the OrangePi PC Plus, not the OrangePi Plus - so, not quite the same hardware. Perhaps that is the core of the problem?

Fourdee, 2017-03-21:
@DenisFromHR 

Yep, that would be it. Apologies, so many OrangePi variants with similar names, hard to keep track.

Ok, as we do not have this board available to us, or an image, we cant official support it, however, i believe applying the correct script.bin should do it:

On the DietPi image, before EMMC write:
```
wget http://dietpi.com/downloads/binaries/fex_bin/orangepipcplus.bin -O /boot/script.bin
reboot
```

Then check ```blkid``` for EMMC, if its there
- complete the process then power off.
- Download http://dietpi.com/downloads/binaries/fex_bin/orangepipcplus.bin and place into ```/boot```
- Power on and test

DenisFromHR, 2017-03-22:
that did it, thanks :)
We can mark this as solved...

Fourdee, 2017-03-22:
@DenisFromHR 
Excellent, apologies for the incorrect initial support on this one. 
Thanks for letting us know results 👍

closed 2017-03-22 by Fourdee