MichaIng/DietPi #824: Diet Pi Image for Banana Pi

Mister-XY, 2017-03-21:
Hello, is it possible to get an Image from Dietpi for Banana Pi?
Here is an ARMBian Version for it
https://www.armbian.com/banana-pi/

Invictaz, 2017-03-22:
Yeah, I hope Allwinner A10 / A20 support is coming. Not all hardware acceleration is working yet but at least it's a start:

https://github.com/linux-sunxi/linux-sunxi
http://linux-sunxi.org/Cedrus
http://linux-sunxi.org/Sunxi-cedrus
http://linux-sunxi.org/Cedrus/libvdpau-sunxi
https://github.com/linux-sunxi/libvdpau-sunxi
http://linux-sunxi.org/Mainlining_Effort
http://www.phoronix.com/scan.php?page=news_item&px=Linux-4.8-Sound
http://forum.kodi.tv/showthread.php?tid=254202
https://www.phoronix.com/scan.php?page=news_item&px=Allwinner-A10-DRM-Display

Mister-XY, 2017-03-31:
And now?

Invictaz, 2017-04-17:
@Mister-XY we wait for @Fourdee 
Since there is a ton of other work to do it might take a while. Since there are already ARMBIAN packages available for A10/A20 it might be possible. The LIBVDPAU-SUNXI is the hardest to get right since there is a lot of repo's and info but no real mainlining. In order to not violate the GPL DietPi can only use the open sourced efforts and not the closed source binaries from Allwinner (I think).

Fourdee, 2017-05-10:
https://github.com/Fourdee/DietPi/issues/938#issue-227327655

Fourdee, 2018-02-22:
Closing as we no longer support or provide images for BPi devices.

closed 2018-02-22 by Fourdee