MichaIng/DietPi #827: BUG reported: Reference CODE: 7a714c54-a3c8-4328-99cb-dd694b2cc498-0

mmagnetix, 2017-03-22:
Hi,

No system log messages are created on the Odroid XU4.

I was working on building an secure Email server with MTA Postfix TLS and tried to troubleshoot postfix and discovered that there were no system logs created.

mmagnetix, 2017-03-22:
rsyslog service is not running and can't be started files/scripts are missing

Running from the Odroid XU4 (software release 1.46)

root@mx:~# systemctl status rsyslog
● rsyslog.service
   Loaded: not-found (Reason: No such file or directory)
   Active: inactive (dead)

root@mx:~# systemctl start rsyslog
Failed to start rsyslog.service: Unit rsyslog.service failed to load: No such file or directory.

mmagnetix, 2017-03-22:
Problem does not occur on Pine64 (software release 1.46)

root@mx:/var/log# systemctl status rsyslog
● rsyslog.service - System Logging Service
   Loaded: loaded (/lib/systemd/system/rsyslog.service; enabled)
   Active: active (running) since Sat 2017-03-18 20:17:05 CET; 4 days ago
     Docs: man:rsyslogd(8)
           http://www.rsyslog.com/doc/
 Main PID: 592 (rsyslogd)
   CGroup: /system.slice/rsyslog.service
           └─592 /usr/sbin/rsyslogd -n

Fourdee, 2017-03-23:
@mmagnetix 

Hi,

syslog is not installed on DietPi by default, this is to save on excessive logging to RAM. Aimed at the majority of our users who do not require intense logging.

### You can install it by either:
- Apt | ```apt-get install rsyslog```
- Change logging mode to full ```dietpi-software``` > Log System > Full. More info here: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=20#p68

mmagnetix, 2017-03-23:
Ok.rsyslog package is by default installed on the DietPI image of the Pine64.Therefor I assumed that the package was broken. Thanks for the heads up.Regards,Magnetix.

Fourdee, 2017-03-24:
@mmagnetix 

> Ok.rsyslog package is by default installed on the DietPI image of the Pine64

Ah, must of slipped the net on this one, thanks for letting me know we'll get that fixed 👍

Fourdee, 2017-04-16:
Does not exist on current image (v145):
```
root@DietPi:~# dpkg -l | grep syslog
root@DietPi:~#
```

Marking as resolved.

closed 2017-04-16 by Fourdee