MichaIng/DietPi #830: No TLS?

shellshocker, 2017-03-24:
There is no TLS-cert on the website and also no checksum to verify the image. From a security perspective -> this is bad. ;-)

Fourdee, 2017-03-24:
@shellshocker 

Thanks for the feedback 👍 

- We'll get the TLS cert and https redirect added to dietpi.com with cerbot. We need to migrate our server anyway (upgrade to Jessie from Wheezy), will do that at the same time.
- CRC/MD5 hash for images: https://github.com/Fourdee/DietPi/issues/831

trajano, 2017-03-29:
Please fix the title so it shows TLS not TSL.

JonasDralle, 2018-03-03:
any update on this?  
It can't take nearly a year to upgrade to HTTPS...

Fourdee, 2018-04-24:
Ok, lets get the ball rolling on this:
- Server upgrade/transition to Stretch
- HTTPS via certbot

Email sent to MVS, requesting a new VPS we can use for the transition. Old one can then be dropped once completed.

MichaIng, 2018-04-24:
@Fourdee 
This means also phpbb update, right? I am excited like a little boy a day before Christmas eve 😃.

Fourdee, 2018-04-27:
New server, 
- Dual core 3GHz
- 2GB RAM
- 20GB SSD

[image]

MichaIng, 2018-04-27:
@Fourdee 
And after upgrade, to have a clean phpBB and phpMyAdmin instance, export again from new server, server, reflash clean image and import again 😆. No I think the effort is too high 😋.

dennisvanderpool, 2018-04-27:
I just visited the site and noticed your remark. Just let me know if i can help (testing).

JonasDralle, 2018-04-28:
[image]
some parts still feel unsafe

Fourdee, 2018-04-29:
@JonasDralle 

Thanks for the report 👍 

Appears Firefox doesnt like http links used on the site, or it doesn't like the http redirect, we'll try to resolve:
> For your own domain, serve all content as HTTPS and fix your links.  Often, the HTTPS version of the content already exists and this just requires adding an "s" to links - http:// to https://.

Fourdee, 2018-04-29:
Completed.

closed 2018-04-29 by Fourdee

MichaIng, 2018-04-29:
SSL Labs result looks good. Cipher suites could be a bid stricter, but some old clients would be excluded.
Mozilla SSL generator with intermediate settings always provided 95%/A+ as far as I remember, but hmm, actually this looks exactly like what you are using (besides disabled OCSP stapling: https://mozilla.github.io/server-side-tls/ssl-config-generator/?server=apache-2.4.25&openssl=1.0.2l&hsts=yes&profile=intermediate
Maybe SSL Labs become stricter lately 😉.