MichaIng/DietPi #831: DietPi Images | CRC/MD5 hash for all images

Fourdee, 2017-03-24:
> https://github.com/Fourdee/DietPi/issues/830#issue-216664900

shellshocker, 2017-03-24:
Please use SHA-256 or higher for the checksum. MD5 is old and broken.

k-plan, 2017-03-26:
> Please use SHA-256 or higher for the checksum. MD5 is old and broken.

To check, if a downloaded and extracted image is corrupt? No.
SHA-256 is very bulky for human reading.
So please offer both checksum in one file, so human or software can decide, which one to use.
Most will do so.

Fourdee, 2017-04-04:
[hash.txt](https://github.com/Fourdee/DietPi/files/892852/hash.txt)

Fourdee, 2017-04-08:
Marking as completed.

Until we update all image downloads, and the zip contains a ```hash.txt```, see https://github.com/Fourdee/DietPi/issues/831#issuecomment-291467022 for current hashes.

closed 2017-04-08 by Fourdee