MichaIng/DietPi #834: DietPi-Set_Hardware | RPi OpenGL hangs startx

Fourdee, 2017-03-24:


Fourdee, 2017-03-24:
Temp solution, disable openGL:
```
/DietPi/dietpi/func/dietpi-set_hardware rpi-opengl disable
```

Fourdee, 2017-03-24:
Thanks to @midwan 👍  confirmed on latest firmware:

> Also with the latest firmware update on the Pi, all the SDL2 apps that used OpenGLES2 stopped working. Rolled back to the previous version and the work again

Fourdee, 2017-04-04:
tests:
- 🈴 RPi Official via apt: ```4.4.50```
- 🈯️ rpi-update: ```4.9.20```

Fourdee, 2017-04-04:
### Workaround/Fix: Update RPi kernel to ```4.9```
NB: This will remove support for Allo Piano 2.1 and BOSS DAC's.

```
apt-get install rpi-update
SKIP_BACKUP=1 rpi-update
reboot
```

Fourdee, 2017-04-04:
@midwan

Looks like ```4.9``` kernel install is solution for OpenGL functionality: https://github.com/Fourdee/DietPi/issues/834#issuecomment-291501770

midwan, 2017-04-04:
@Fourdee 
However, I found that going to kernel 4.9 broke the ability to start SDL2 applications from the console (i.e. not from within X11). Not just Amiberry, but any test code I tried. They all failed to find a render driver, even when run with xinit.

Maybe you'll have better luck and I did something wrong, but I tried it several times with the same result always... :(

Fourdee, 2017-04-05:
@midwan 

> However, I found that going to kernel 4.9 broke the ability to start SDL2 applications from the console (i.e. not from within X11). Not just Amiberry, but any test code I tried. They all failed to find a render driver, even when run with xinit.

Thanks for the heads up 👍, very strange. I'll reference this to ticket:
https://github.com/Fourdee/DietPi/issues/850

Fourdee, 2017-12-08:
Confirmed working on our Stretch image with our latest 4.9.62 kernel: https://github.com/Fourdee/DietPi/issues/475

Marking as completed, Stretch image will be released at weekend.

closed 2017-12-08 by Fourdee