MichaIng/DietPi #836: After dietpi-update i lost /Dietpi/*

mikolaj24, 2017-03-24:
-bash: /DietPi/dietpi/login: No such file or directory

i only run dietpi-update

System working but not stability (i lost www too)

Fourdee, 2017-03-24:
@mikolaj24 

> System working but not stability (i lost www too)

Most likely a filesystem corruption has occurred. This is usually linked to unstable hardware PSU/SDcard.

Which?:
- PSU used (eg: 5v/1A)
- SD card used (eg: Toshiba class 8)

mikolaj24, 2017-03-24:
my power is 1A

i use poweroff to close server.

Fourdee, 2017-03-24:
@mikolaj24 

> my power is 1A, Orange Pi zero 512

Boardline sufficient I believe for that device. Its most likely an unstable 5v that is causing the problem.
We highly recommend the Official RPi 5v/2A PSU, for all MicroUSB powered devices.

> is any chance to copy back this files? Server working but www and other services are down
/Dietpi is tempfs, meybe i can mount it manualy?

I personally wouldnt recommend attempting to fix this installation. The filesystem corruption appears to be in a bad state, and any "fix" could be short lived, especially if the PSU is the cause.

I'd recommend formatting the SD card (to reset all cells on SD card to 0: https://github.com/Fourdee/DietPi/issues/769#issuecomment-283030267) and writing the DietPi image again. Ideally, with a more sufficient PSU to power the system.

closed 2017-03-24 by mikolaj24