MichaIng/DietPi #838: wifi Country

mikolaj24, 2017-03-24:
I cant set my country in diet-config wifi
I get

Country code (PL) could not been applied. Please    │ 
                       │ check the country code and try again.

Fourdee, 2017-03-25:
@mikolaj24 

```iw reg set``` does not work in all instances, depends on kernel version and WiFi chipset used.

Please try this alternative method:
```
echo -e "options cfg80211 ieee80211_regdom=PL" > /etc/modprobe.d/cfg80211.conf
reboot
```

Then check with:
```
iw reg get
```

Fourdee, 2017-03-26:
We'll apply to ```ieee80211_regdom=```, as well as ```iw reg set```.

Fourdee, 2017-04-04:
@mikolaj24 

Did this work for you? https://github.com/Fourdee/DietPi/issues/838#issuecomment-289210613

Fourdee, 2017-04-05:
I'll mark this as closed. 
We now apply ```iw reg set``` and ```ieee80211_regdom=```. Aside from any module params, this is it in terms of applying country code, then, its up to kernel support for the WiFi chipset.

symbios24, 2018-01-19:
Hello, i`m getting the Country code (US) could not been applied.
I tried your fix (echo -e "options cfg80211 ieee80211_regdom=US" > /etc/modprobe.d/cfg80211.conf) 
its not working 
Any solution?

symbios24, 2018-01-19:
Hello, i`m getting the Country code (US) could not been applied.
I tried your fix (echo -e "options cfg80211 ieee80211_regdom=US" > /etc/modprobe.d/cfg80211.conf)
its not working
Any solution?

Plus i cannot the the wifi hotspot from any device computer/smartphone is this normal?
From Dietpi setting i see that its enabled.

symbios24, 2018-01-19:
Hello, i`m getting the Country code (US) could not been applied.
I tried your fix (echo -e "options cfg80211 ieee80211_regdom=US" > /etc/modprobe.d/cfg80211.conf)
its not working
Any solution?

Plus i cannot see the wifi hotspot from any device computer/smartphone is this normal?
From Dietpi setting i see that its enabled.

symbios24, 2018-01-19:
Hello, i tried the DietPi stretch  for my raspberryPi 3 i`m getting the Country code (US) could not been applied.
I tried your fix (echo -e "options cfg80211 ieee80211_regdom=US" > /etc/modprobe.d/cfg80211.conf)
its not working
Any solution?

Plus i cannot see the wifi hotspot from any device computer/smartphone is this normal?
From Dietpi setting i see that its enabled.

After that i tried the DietPi Jessie and i can see the hotspot from my devices i did not try to change the country thought.

Thanks

Fourdee, 2018-01-19:
@symbios24  

Thanks for the report 👍  we'll investigate/test

k-plan, 2018-01-19:
@Fourdee 

on a running v6.-1 Odroid C1 (jessie):

[image]

[image]

[image]

```
root@Odroid-C1:~# cat /etc/wpa_supplicant/wpa_supplicant.conf
country=JP
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

network={
    ssid="MySuperDooperWiFi"
    key_mgmt=WPA-PSK
    psk="0123456789"
}

root@Odroid-C1:~# cat /etc/modprobe.d/cfg80211.conf
options cfg80211 ieee80211_regdom=GB
```

Fourdee, 2018-01-20:
We'll need to bring back ```iw reg set```:
```
root@DietPi:~# iw reg get
country 00: DFS-UNSET
        (2402 - 2472 @ 40), (6, 20), (N/A)
        (2457 - 2482 @ 40), (6, 20), (N/A), PASSIVE-SCAN
        (2474 - 2494 @ 20), (6, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 160), (6, 20), (N/A), PASSIVE-SCAN
        (5250 - 5330 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN
        (5490 - 5730 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN
root@DietPi:~# iw reg set GB
root@DietPi:~# iw reg get
country GB: DFS-ETSI
        (2402 - 2482 @ 40), (N/A, 20), (N/A)
        (5170 - 5250 @ 80), (N/A, 20), (N/A)
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS
        (5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)

```

Fourdee, 2018-01-20:
@k-plan 
https://github.com/Fourdee/DietPi/issues/838#issuecomment-359170668
Resolved, i've tested C1 (your image, works a treat 👍 ) in ```dietpi-config``` and also sets during boot.

Marking as completed.

closed 2018-01-20 by Fourdee

symbios24, 2018-01-20:
What i have to do now to be fixed on me? 
i only have to update DietPi stretch?

Thanks

Fourdee, 2018-01-20:
@symbios24 

You'll need the updated v6.0 image, will be released next week, you can try it below:
http://dietpi.com/downloads/testing/v6.0/

symbios24, 2018-01-21:
Thanks
another thing now if its possible
i removed the card with DietPi from my raspberry pi3 and insert it in my raspberry pi Model B that has a usb WIfi Adapter(realtek) is it possible the WifiHotspot to see and configure automatically any usb/or internal wifi adapter so it works at any time without any setup if you change wifi adapter or board?? i`m asking because the hospot is not active like this.

symbios24, 2018-02-04:
Can you do the wifi to work automatic i mean if you have a raspberry pi 3 with internal wifi only to use it, or if you have a usb wifi to use that instead without the need to configure it from network adapters menu.
thanks