MichaIng/DietPi #840: DietPi-Software | NTPD wait tweaks

Fourdee, 2017-03-26:


k-plan, 2017-03-26:
@Fourdee 

occur this only, if WiFi (WLAN) configured to use DHCP client?

If so, it will be worth to test a option `raspi-config` use:

[image]

Please take a look here:

https://github.com/raspberrypi-ui/rc_gui/blob/bf829a6cd4a130eaa421e1a39cee6388301f8414/raspi-config#L854

Fourdee, 2017-03-26:
@k-plan 

Thanks.

Notes:
- ```route | grep -m1 -ci '[[:space:]]UG[[:space:]]'```

Fourdee, 2017-03-27:
@k-plan 

This already exists in our boot script: https://github.com/Fourdee/DietPi/blob/master/dietpi/boot#L235-L260

Although, it will time out after 10 seconds of waiting. I'll add an option in ```dietpi-config``` to allow infinite wait: https://github.com/Fourdee/DietPi/issues/842

Fourdee, 2017-03-27:
Implemented changes as per: https://github.com/Fourdee/DietPi/issues/840#issue-217061769

Completed.

closed 2017-03-27 by Fourdee