MichaIng/DietPi #841: DietPi-Software | DietPi-Cam (RPi Cam Control v6.2.29): No timelaps video convert

k-plan, 2017-03-26:
Tested Hardware:

- RPi Zero W
- [RPi Cam v1.3](http://www.waveshare.com/product/rpi-camera-c.htm) 
- or [Raspberry Zero mini Camera (JT-ZERO-V2.0)](http://www.waveshare.com/rpi-zero-v1.3-camera.htm)

Software:

- DietPi v1.47
- DietPi Cam (RPi Cam Control v6.2.29) installed via `dietpi-software`
- Web-server: lighttp

Both cams are up and running fine with DietPi Cam.
Can take pictures, timelaps and videos with this cams.
But converting timelaps pictures into a undercranking via web interface, it will not work.
Converting into a video, preview will shown as "Busy" and nothing will happened.

[image]

Web interface show Convert using: `gst-launch-1.0  -v multifilesrc location=i_%05d.jpg caps=\"image/jpeg,framerate=\(fraction\)30/1\" ! jpegdec ! queue ! videoscale ! video/x-raw ! queue ! omxh264enc target-bitrate=15000000 control-rate=variable ! video/x-h264, profile=high ! h264parse ! mp4mux ! filesink location=`

Fiddling around, I noticed, `gst-launch-1.0` isn't installed.

`root@RPi-Zero-Cam:~# apt-get install gstreamer1.0-tools`

But this didn't help. Something more seemingly will be missing.

Fourdee, 2017-04-02:
Notes:
```
apt-get install gstreamer1.0-tools #gstreamer1.0-plugins-base libav-tools
```

```
cd /var/www/dietpicam/media
gst-launch-1.0  -v multifilesrc location=i_%06d.jpg caps=\"image/jpeg,framerate=\(fraction\)30/1\" ! jpegdec ! queue ! videoscale ! video/x-raw ! queue ! omxh264enc target-bitrate=15000000 control-rate=variable ! video/x-h264, profile=high ! h264parse ! mp4mux ! filesink location=

WARNING: erroneous pipeline: no element "multifilesrc"

```

k-plan, 2017-04-02:
Fiddle around a little bit with [motioneye](https://github.com/ccrisan/motioneye).

So I noticed, where is a newer `precompiled ffmpeg_3.1.1-1_armhf.deb`  for Raspian.
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian

Nice peace of software with a slick web interface.
Focal point are surveillance camera, but it will support Raspberry PI camera module, most USB cameras, IP (network) cameras and direct streams. 
Timelapse movies works as well. (tested)

Running on my  RPi Zero W, as log as you don't use that much high video resolution and frame rate. 

Can be a option or alternative.

Fourdee, 2017-04-02:
@k-plan 

Interesting, thanks 👍 

Seems we already have a request for this aswell:
https://github.com/Fourdee/DietPi/issues/718

Fourdee, 2017-04-05:
Cant get gstreamer to convert images to video, not sure if syntax is correct, but this everytime:
```
Failed to open file 'i_00000.jpg': No such file or directory
```

```
tl_0000_0001_20170405_121012.jpg
```

Nope:
````
gst-launch-1.0  -v multifilesrc location=tl_%04d_%04d_%08d_%06d.jpg caps=\"image/jpeg,framerate=\(fraction\)30/1\" ! jpegdec ! queue ! videoscale ! video/x-raw ! queue ! omxh264enc target-bitrate=15000000 control-rate=variable filesink location=output.avi
```

Fourdee, 2017-04-08:
Lets try using FFmpeg and conversion instead:
```
dietpi-software install 7

cd /var/www/dietpicam/media
ffmpeg -pattern_type glob -i 'tl_*.jpg' -vcodec libx264 output.mp4

rm tl_*.jpg
```

Works, but runs out of memory after 6 images.

k-plan, 2017-04-09:
> Works, but runs out of memory after 6 images.
>
> Don't have the time to fiddle with this further at the moment. Motion Eye is now available as an alternative with working timelapse.
>
> Marking this as a known issue for now.

Yes, let well alone! 👍 

Motion Eye? Yes! 😃 

... close.

closed 2017-04-09 by k-plan