MichaIng/DietPi #844: apt-get install python error: "files list file for package 'libisccfg-export90' is missing final newline"

blamoo, 2017-03-29:
Happened on a Raspberry Pi 3 fresh installation.

Steps:
- Installed DietPi_v145_RPi-armv6-(Jessie).img on a 2GB SD
- Configured wireless networking using keyboard + HDMI monitor
- Moved to SSH to continue installation
- Installed and configured no-ip client (haven't tested without it, maybe isn't related to the problem)
- Changed APT mirror to "Global"
- Completed the initial setup and restarted
- Ran 'apt-get install python-pip'

Output: 
```
root@DietPi:~# apt install python-pip
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following extra packages will be installed:
  binutils build-essential cpp cpp-4.9 dpkg-dev fakeroot file g++ g++-4.9 gcc gcc-4.9 libalgorithm-diff-perl libalgorithm-diff-xs-perl libalgorithm-merge-perl libasan1 libatomic1 libc-dev-bin libc6-dev libcloog-isl4 libdpkg-perl
  libfakeroot libfile-fcntllock-perl libgcc-4.9-dev libgomp1 libisl10 libmagic1 libmpc3 libmpfr4 libpython-stdlib libpython2.7-minimal libpython2.7-stdlib libsqlite3-0 libstdc++-4.9-dev libtimedate-perl libubsan0 linux-libc-dev make
  manpages manpages-dev mime-support patch python python-chardet python-colorama python-distlib python-html5lib python-minimal python-ndg-httpsclient python-openssl python-pkg-resources python-pyasn1 python-requests python-setuptools
  python-six python-support python-urllib3 python-wheel python2.7 python2.7-minimal
Suggested packages:
  binutils-doc cpp-doc gcc-4.9-locales debian-keyring gcc-4.9-doc libstdc++6-4.9-dbg gcc-multilib autoconf automake libtool flex bison gdb gcc-doc libgcc1-dbg libgomp1-dbg libitm1-dbg libatomic1-dbg libasan1-dbg liblsan0-dbg
  libtsan0-dbg libubsan0-dbg libcilkrts5-dbg libquadmath-dbg glibc-doc libstdc++-4.9-doc make-doc man-browser diffutils-doc python-doc python-tk python-genshi python-lxml python-openssl-doc python-openssl-dbg python-distribute
  python-distribute-doc doc-base python2.7-doc binfmt-support
Recommended packages:
  python-dev-all
The following NEW packages will be installed:
  binutils build-essential cpp cpp-4.9 dpkg-dev fakeroot file g++ g++-4.9 gcc gcc-4.9 libalgorithm-diff-perl libalgorithm-diff-xs-perl libalgorithm-merge-perl libasan1 libatomic1 libc-dev-bin libc6-dev libcloog-isl4 libdpkg-perl
  libfakeroot libfile-fcntllock-perl libgcc-4.9-dev libgomp1 libisl10 libmagic1 libmpc3 libmpfr4 libpython-stdlib libpython2.7-minimal libpython2.7-stdlib libsqlite3-0 libstdc++-4.9-dev libtimedate-perl libubsan0 linux-libc-dev make
  manpages manpages-dev mime-support patch python python-chardet python-colorama python-distlib python-html5lib python-minimal python-ndg-httpsclient python-openssl python-pip python-pkg-resources python-pyasn1 python-requests
  python-setuptools python-six python-support python-urllib3 python-wheel python2.7 python2.7-minimal
0 upgraded, 60 newly installed, 0 to remove and 4 not upgraded.
Need to get 41.7 MB of archives.
After this operation, 133 MB of additional disk space will be used.
Do you want to continue? [Y/n]
Get:1 http://mirrordirector.raspbian.org/raspbian/ jessie/main libmagic1 armhf 1:5.22+15-2+deb8u3 [244 kB]
Get:2 http://mirrordirector.raspbian.org/raspbian/ jessie/main libsqlite3-0 armhf 3.8.7.1-1+deb8u2 [377 kB]
[blah blah blah]
Get:60 http://mirrordirector.raspbian.org/raspbian/ jessie/main python-wheel all 0.24.0-1 [47.5 kB]
Fetched 41.7 MB in 3min 6s (224 kB/s)
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
        LANGUAGE = (unset),
        LC_ALL = (unset),
        LANG = "0.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
Extracting templates from packages: 100%
Selecting previously unselected package libmagic1:armhf.
dpkg: unrecoverable fatal error, aborting:
 files list file for package 'libisccfg-export90' is missing final newline
E: Sub-process /usr/bin/dpkg returned an error code (2)
```

To solve the issue i had to `find / | grep libisccfg-export90` and delete all the files found.

The same problem happened when trying to install deluged manually, but with `libdns-export100` instead of `libisccfg-export90`.

blamoo, 2017-03-29:
nvm, i couldn't boot the system after that irresponsible "delete all the files" thing.

i'll recover them from a backup, analyze the deleted files and update this issue later.

blamoo, 2017-03-29:
For some reason, the file `/var/lib/dpkg/info/libisccfg-export90.list` was filled with 0x00 bytes. I've mounted the original image and the file is okay. 
It looks like something happened after the installation.. I'll test a bit more tomorrow, maybe the SD card isn't healthy or whatever...

BTW, this fixes the problem:
```
rm /var/lib/dpkg/info/libisccfg-export90.list
apt install --reinstall libisccfg-export90
```

Fourdee, 2017-03-29:
@blamoo 

> mounted the original image and the file is okay.
Edit: It happened again, now with my .bashrc file. It's filled with 0x00 bytes.

Hi, Thanks for the report.

If this is occurring on random files, I believe its due to filesystem corruption. Usual causes are:
- Unstable PSU
- Unstable SD CARD

Which of the above are you using on this system? 

With some SD cards, users have reported completing a full format on the SD card, prior to image write, prevents similar issues. Clears all SD card cells to 0 (eg: full reset). Might be worth a try:
https://github.com/Fourdee/DietPi/issues/769#issuecomment-283030267

Fourdee, 2017-03-29:
@blamoo 

> The PSU is the original charger from a Samsung S3, 5V@1A. I'll test it with a different PSU too.

1A is not sufficient, you need at least 5V/2A for RPi 3. Also, phone chargers are not recommended as they most lack the stable 5V under load.

I think another PSU and fresh write of image should resolve this.

blamoo, 2017-03-29:
I'll test with the 5v line of a computer PSU then...

Fourdee, 2017-03-29:
@blamoo 

via USB ? still wont be enough unfortunately, USB 2 0.5A, USB 3 0.9A.

Molex 5V? Worth a shot 😉

blamoo, 2017-03-29:
Bingo! 16 amps :D
[image]

Fourdee, 2017-03-29:
@blamoo 

> Bingo! 16 amps :D

Epic 👍 

Can you read volts on that device aswell?

blamoo, 2017-03-29:
Sure:

[image]

blamoo, 2017-03-29:
The same problem happened with a different file now: /var/lib/dpkg/info/libirs-export91.list

blamoo, 2017-03-30:
Tested again today, the same problem happened with the latest Minibian.

It looks like a hardware failure. I'll test again with a different SD card...

blamoo, 2017-03-31:
Okay, it looks like it was just because of a bad SD Card... Thanks anyways :)

closed 2017-03-31 by blamoo