MichaIng/DietPi #846: DietPi-Config | Soundcard: Support pHAT BEAT for Raspberry Pi's

k-plan, 2017-03-29:
Product:

https://shop.pimoroni.com/products/phat-beat

https://shop.pimoroni.com/products/pirate-radio-pi-zero-w-project-kit

Code on Github:

https://github.com/pimoroni/phat-beat

Fourdee, 2017-03-29:
@k-plan 

Looks great, i'll get one ordered. Had my eye on this one aswell for a while: https://shop.pimoroni.com/products/speaker-phat

Lets get them both added :)

Fourdee, 2017-03-29:
Worth a shot, lets see:
https://twitter.com/DietPi_/status/847139159672324096

k-plan, 2017-04-09:
- pHAT BEAT Web-Radio based on [Mopidy](https://learn.pimoroni.com/tutorial/sandyj/streaming-spotify-to-your-pi) with ODROID-SHOW2 as display:

[image]

- autoplay special playlist, when device boot
- display shows radio station, track info and some device infos. Auto refresh every 15 sec.
- overclocked to: `arm_freq=1050
core_freq=500
sdram_freq=500
gpu_freq=500
sd_clock=100.000 MHz`
- current consumption: ca. 390 mA via USB powerbank  for RPi-Zero W (using WiFi) with pHAT BEAT (with VU Meters), stereo speaker 8 Ohm, Odroid-show2 display

😃 

cons: 
- have to disable [clean-shutdown](https://github.com/pimoroni/clean-shutdown) because give rise to 100% cpu load. (normal running: 45 -50%)

Fourdee, 2017-04-10:
@k-plan 

> pHAT BEAT Web-Radio based on Mopidy with ODROID-SHOW2 as display:

Epic 👍 

> have to disable clean-shutdown because this script give rise to 100% cpu load

Try moving this line:
https://github.com/pimoroni/clean-shutdown/blob/master/daemon/usr/bin/cleanshutd#L65

To, below the ```if``` line:
https://github.com/pimoroni/clean-shutdown/blob/master/daemon/usr/bin/cleanshutd#L66

Invictaz, 2017-05-11:
@k-plan Great project

Fourdee, 2017-05-12:
#### Speaker-phat:
- 🈯️ Audio fine with hifiberry-dac overlay
- 🈴 Fails with ALSA dmix for VU meter https://github.com/pimoroni/pivumeter
```
root@DietPi:~# speaker-test -c2

speaker-test 1.0.28

Playback device is default
Stream parameters are 48000Hz, S16_LE, 2 channels
Using 16 octaves of pink noise
ALSA lib pcm_direct.c:896:(snd1_pcm_direct_initialize_slave) slave plugin does not support mmap interleaved or mmap noninterleaved access
ALSA lib pcm_dmix.c:1034:(snd_pcm_dmix_open) unable to initialize slave
Playback open error: -22,Invalid argument
```

closed 2018-02-22 by Fourdee