MichaIng/DietPi #848: DietPi Software | CPU temperature not shown on Roseapple Pi

joaofl, 2017-03-30:
CPU temperature is not shown by the "cpu" command.

Have to take a look at that. Must be the same as for the sparky board.

Fourdee, 2017-03-30:
@joaofl 

Sparky CPU temp is a little different, has a special case:
https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-cpuinfo#L73-L82

https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-cpuinfo#L59

joaofl, 2017-03-30:
@Fourdee 
Thanks for the link
I'll also check out how it is done on the Armbian distribution. It works there.

joaofl, 2017-04-05:
@Fourdee 
Your suggestions worked right on. I've made a pull request to fix that: https://github.com/Fourdee/DietPi/pull/858

closed 2017-04-05 by joaofl

joaofl, 2017-04-05:
I've run a 5 min strees test from dietpi, and here is what I've got. It kind of runs cool.

It seems to stabilize around 60, but it is probably worth testing it for longer.

```
Wed Apr  5 04:04:48 BST 2017 | 45'c
...
Wed Apr  5 04:06:09 BST 2017 | 53'c
...
Wed Apr  5 04:08:17 BST 2017 | 56'c
...
Wed Apr  5 04:09:46 BST 2017 | 58'c
```

Fourdee, 2017-04-05:
@joaofl 

Great work. Temps look good 👍