MichaIng/DietPi #850: DietPi-Software | AmiBerry SDL2

Fourdee, 2017-04-01:


Fourdee, 2017-04-11:
Compiles:
- SDL2 ARMv6
- SDL2 ARMv7

Fourdee, 2017-04-13:
@midwan

Cannot get SDL window to init.
Which system + setup have you run with SDL2 + RPi and its working? eg: RetroPie image + SDL2 install? I can then replicate install and see whats needed.

midwan, 2017-04-13:
@Fourdee 
I've tested it in Raspbian, RetroPie and DietPi. :)

The key ingredients are:
1) install the SDL2 package from the RetroPie repo, as per the instructions in the Amiberry SDL2 readme
2) have the "Legacy" driver enabled in raspi-config, not OpenGL. Unfortunately I haven't found a way around this one, but OpenGL causes all SDL2 windows to fail to initialize from the console - they do work if launched from within X11, but not with xinit.

Fourdee, 2017-04-13:
@midwan 

> install the SDL2 package from the RetroPie repo, as per the instructions in the Amiberry SDL2 readme

Legend 👍  thanks.

Fourdee, 2017-04-17:
Note to self:
- update http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=64#p64 replacing all ```uae4arm``` references with ```amiberry```

Fourdee, 2017-04-17:
@midwan 

Finally sussed it, original SDL2 compiles were fine. User error forgetting to increase GPU mem split lol.

x2/3 framerate with SDL 2 (Elite / Indy 500), great work, smooth as a baby's bum 👍

Fourdee, 2017-04-17:
AmiBerry SDL2 compile:
```
apt-get install -y \
git build-essential \
libsdl2-image-dev libsdl2-ttf-dev  \
libxml2-dev libflac-dev libmpg123-dev libpng-dev google-perftools libgoogle-perftools-dev
```

midwan, 2017-04-17:
@Fourdee 
The neon_helper is a version for newer ARM processors only, but I'm working towards phasing it out completely since I didn't see any noticeable speed improvements when using that vs C++ code optimized by the compiler.

The alternative is the "arm_helper" file from the SDL1 archive, it contains versions of the same functions for the older CPU of the Pi 1/Zero.

Fourdee, 2017-04-18:
@midwan 

> The alternative is the "arm_helper" file from the SDL1 archive, it contains versions of the same functions for the older CPU of the Pi 1/Zero.

Thanks, although i have no idea how to go about achieving this, any pointers/steps? :)

Or even better, do you have a rpi1 compiled binary available you could upload? Only one i'am missing.

Fourdee, 2017-04-18:
Notes:
If SDL2 is complied with OpenGL enabled:
```
Apr 18 10:18:52 DietPi systemd[1]: Started AmiBerry Amiga Emulator.
Apr 18 10:18:53 DietPi bash[1978]: Amiberry-SDL2 by Dimitris (MiDWaN) Panokostas
Apr 18 10:18:53 DietPi bash[1978]: Unable to create window Could not initialize OpenGL / GLES library
```

Must disable x11 and OpenGL in compile for Amiberry SDL2.

Fourdee, 2017-04-18:
SDL2 available from ```dietpi-autostart```:
[image]

And
```
systemctl start amiberry-sdl2
```

Fourdee, 2017-04-18:
Weird, this just appeared, i'll add font to our zip.
```
root@DietPi:/etc/amiberry# ./amiberry-sdl2 -f conf/autostart.uae
Amiberry-SDL2 by Dimitris (MiDWaN) Panokostas
starting sound thread..
SDLTrueTypeFont::SDLTrueTypeFont. Couldn't open data/Topaznew.ttf
stopping sound thread..
```

midwan, 2017-04-18:
@Fourdee 
It's a recent change, the font has been replaced with something more Amiga-like. The color scheme in the GUI is also updated, and I am currently working on enabling custom mapping of any key to functions like "enter GUI", "Quit" and eventually Custom Controls.

The next step will then be to allow custom mapping of any controller button to those functions as well.

Things are moving rather quickly, so I missed notifying you about this change, sorry. :)

Fourdee, 2017-04-18:
@midwan 

Aye, no worries, great to see Amiberry progress quickly 👍 

We'll most likely have to stick with ```xinit``` for this, this ensures input grab, regardless of tty. I've tried various bash/systemD options and "hacks" to control input, no joy.

midwan, 2017-04-18:
@Fourdee 
Forgot to mention that there are 2 more files added in the `data `directory:

1. cursor.bmp
2. cursor-x2.bmp

Fourdee, 2017-04-19:
@midwan 

>  2 more files added in the datadirectory:

Thanks, updated 👍 

SDL2 leaks to desktop aswell. Think we may have to recompile with the patch.

Fourdee, 2017-04-21:
ToDo:
- Update online docs with reference to ```/etc/amiberry``` instead of ```uae4arm```
- ^^ SDL2: add info to launch from desktop (must be run by user due to SDL2 tty leak ```cd /etc/amiberry; ./amiberry-sdl2 -f conf/autostart.uae```

Fourdee, 2017-04-22:
Completed.

@midwan 
I'll keep an eye on SDL2 devolpment. But if you can give me a heads up on major releases, we can roll that out to our DietPi users during updates.

Great work on SDL2, Amiberry smooth as butter now 👍

closed 2017-04-22 by Fourdee

midwan, 2017-04-22:
@Fourdee 
Thanks, of course I'll notify you of any major changes/releases. For now I'm focusing of fixing some bugs, since I wouldn't want to release something that has annoying issues.

Good work integrating this in DietPi! :)