MichaIng/DietPi #852: DietPi-Software | VirtualHere

Fourdee, 2017-04-02:
> https://www.virtualhere.com/home

tatoosh, 2017-04-02:
Nice, it works very good with deitpi by installing manually and adding systemd service.

phonicmouse, 2017-04-02:
+1 for this

Fourdee, 2017-04-09:
- ARMv6+: https://virtualhere.com/sites/default/files/usbserver/vhusbdarm
- ARM64: https://virtualhere.com/sites/default/files/usbserver/vhusbdarm64
- AMD64: https://virtualhere.com/sites/default/files/usbserver/vhusbdx86_64

Fourdee, 2017-04-09:
Virtual Here is excellent, really impressed.
[image]

Fourdee, 2017-04-09:
Ok, there is a minor issue with attached USB drives. 
- 🈺 Virtual here will unmount and take over all USB drives on system, excluding those in-use  (eg: DietPi user data)
- 🈴 Using the USB drive in client, will unmount the drive, regardless of if its in-use or not.

Will make users aware in online doc.

Fourdee, 2017-04-09:
Completed.

closed 2017-04-09 by Fourdee