MichaIng/DietPi #854: DietPi-Software | OcToPrint

Fourdee, 2017-04-04:
> http://dietpi.com/phpbb/viewtopic.php?f=12&t=1594#p6533
https://github.com/foosel/OctoPrint

Yury-MonZon, 2017-05-10:
Sorry, forgot to reply. I'm in!

kurtjcu, 2017-06-20:
sounds interesting, I have a few printers and pi3/pi2/orangePi to test with

tobyrobb, 2017-07-07:
I'm in also , have pi zero w's and a Pi 3 running Diet-pi, and printers.

Fourdee, 2017-07-07:
Thanks guys, i'll pop this into v154 and take a look.

Fourdee, 2017-07-07:
Reqs:
- Pyhton
- Python-Pip
- Python-dev
- Build-Essential

Does not run under root, simply terminates:
```
root@DietPi:/mnt/dietpi_userdata/octoprint# octoprint serve
2017-07-07 16:04:09,663 - octoprint.server - INFO - ******************************************************************************
2017-07-07 16:04:09,664 - octoprint.server - INFO - Starting OctoPrint 1.3.4 (HEAD -> master branch)
2017-07-07 16:04:09,664 - octoprint.server - INFO - ******************************************************************************
2017-07-07 16:04:09,715 - octoprint.plugin.core - INFO - Loading plugins from /usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins, /root/.octoprint/plugins and installed plugin packages...
2017-07-07 16:04:11,133 - octoprint.plugins.discovery - INFO - pybonjour is not installed, Zeroconf Discovery won't be available
2017-07-07 16:04:11,136 - octoprint.plugin.core - INFO - Found 7 plugin(s) providing 7 mixin implementations, 4 hook handlers
You should not run OctoPrint as root!
root@DietPi:/mnt/dietpi_userdata/octoprint#
```

- ```You should not run OctoPrint as root!``` should be ```OctoPrint is not designed to run under root. You can however override this at your own risk with --iknowwhatimdoing```

```
octoprint serve --iknowwhatimdoing
```

Fourdee, 2017-07-07:
- Online Doc: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=7958#p7958

Yury-MonZon, 2017-07-08:
>  You may need to add the pi user to the dialout group and tty so that the user can access the serial ports:
> 
> sudo usermod -a -G tty pi
> sudo usermod -a -G dialout pi
from [https://github.com/foosel/OctoPrint/wiki/Setup-on-a-Raspberry-Pi-running-Raspbian](url)

Works fine, even autoupdates intself.

Fourdee, 2017-07-08:
@Yury-MonZon 
> Works fine, even autoupdates intself. In out case we could create another user i.e. octo.

Thanks 👍 

We dont have a ```pi``` user in DietPi. We do have a ```dietpi``` user which has same permissions as ```pi``` user:
https://github.com/Fourdee/DietPi/blob/4f7f27f49dd2329dc4b8a670896af188d3bf7037/dietpi/func/dietpi-set_software#L171-L197

Although, ```dietpi``` user is still under testing, is not implemented as the user for any of our software installations at the moment:
https://github.com/Fourdee/DietPi/issues/896https://github.com/Fourdee/DietPi/issues/896

Root for now, plans to change that in the future :)

Fourdee, 2017-07-09:
Available for testing in our testing branch, please see link below for instructions on how to setup an installation with this branch:
https://github.com/Fourdee/DietPi/blob/master/TESTING-BRANCH.md

Online doc for OctoPrint with DietPi:
http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=7958#p7958

---

@Yury-MonZon @kurtjcu @tobyrobb Would appreciate any assistance with testing this software, please see section above for instructions.

Yury-MonZon, 2017-07-09:
Great news! Thanks!
Any plans to add mjpeg-streamer for octoprint? It is much more useful with it included, ie online viewing etc...
Installation details are in the doc I mentioned before.

kurtjcu, 2017-07-09:
Awesome. I will have a try with some orange Pi Zero's.

:)

Fourdee, 2017-07-10:
@Yury-MonZon 
> Any plans to add mjpeg-streamer for octoprint? It is much more useful with it included, ie online viewing etc...
Installation details are in the doc I mentioned before.

👍 I'll take a look :)

kurtjcu, 2017-07-11:
Have been testing on my OrangePi Zero (h2) and is working well. So far only small prints (up to 2 hours) no problem. Interface runs well (including sync layer view for whole print).

I did try to install curaengine to be able to slice on the Zero, however i failed. 
I used https://github.com/foosel/OctoPrint/wiki/Cura-Integration as a guide. The first section(use pre compiled file) seemed to be going well until i tried to load a config file. I presumed this was because the verification process was failing due to compiled file not matching my architecture. I tried to follow the make instructions, and made sure I had the 15.06 branch, however make was complaining about no target, and I am afraid that is as far as I got, my compiler Foo is not strong enough for this one... 

Things to note: 
* python-wxgtk2.8 is not available from repo so I changed to python-wxgtk3.0. 
* make sure have cura engine <= 15.06 as octoprint does not support higher versions.

+1 on the capability to stream, I think video is what makes octoprint a complete package.

P.S. Love the project over all, keep up the good work.

Fourdee, 2017-07-16:
No joy on ```mjpg-streamer``` unfortunately. Won't make it for v154. Although not a solution, we do offer software titles for RPi camera which could be used in tandem:
http://dietpi.com/phpbb/viewtopic.php?f=8&t=5#p48
http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=100#p6610

I'll mark this as completed for now. If we get enough requests to add ```curaengine```, i'll create a separate ticket.

Yury-MonZon, 2017-07-18:
mjpeg-streamer is working on my orange pi zero with ordinary webcams. It wasn't easy though. What if I'll post my installation logs - will it help? Or it is abandoned completely? Thanks for your work!

Fourdee, 2017-07-18:
@Yury-MonZon 

> What if I'll post my installation logs - will it help?

Worth a shot 👍 
Please post them here and we'll test it.

Yury-MonZon, 2017-07-20:
I've checked my .bash_history and it looks like I installed mjpeg-streamer-experimantal according to this guide https://github.com/foosel/OctoPrint/wiki/Setup-on-a-Raspberry-Pi-running-Raspbian
 without raspicam module - I needed it for ordinary usb webcams (Logitech at the moment).

Please, help me on which files do you need from my system in order to help. I can take the whole system image if you like.

Fourdee, 2017-07-21:
@Yury-MonZon 

Thanks 👍 

> mjpeg-streamer-experimantal according to this guide https://github.com/foosel/OctoPrint/wiki/Setup-on-a-Raspberry-Pi-running-Raspbian

Thats fine, i'll check the above and attempt install again.

Fourdee, 2017-07-21:
Reopened, attempt: https://github.com/foosel/OctoPrint/wiki/Setup-on-a-Raspberry-Pi-running-Raspbian#webcam

Fourdee, 2017-07-23:
@Yury-MonZon 

Ok, still unable to get this to work for RPi camera module (https://github.com/Fourdee/DietPi/issues/854#issuecomment-317025315). Plan was to automatically install for that device.

For non-RPi camera module cameras, the following should do it:
```
apt-get install -y build-essential git subversion libjpeg8-dev imagemagick libav-tools cmake
git clone https://github.com/jacksonliam/mjpg-streamer.git --depth=1
cd mjpg-streamer/mjpg-streamer-experimental
export LD_LIBRARY_PATH=.
make -j $(nproc --all)
```

As we can't get the RPi cam to compile, i'll leave this for now and mark as closed.

Win-doze, 2019-01-20:
I don't know if this will help someone but I managed to compile mjpg-streamer for the raspberry pi camera. I came across this webpage https://github.com/AmedeeBulle/octoprint-containers/issues/1 and found if I installed libraspberrypi-dev using 'apt-get install libraspberrypi-dev' before trying to 'make' I got the missing input file and could stream video from the raspberry pi camera.

MichaIng, 2019-01-20:
@Win-doze 
Thanks for figuring this out

@Fourdee 
Worth to give it another try by times?

Win-doze, 2019-01-20:
@MichaIng 
Still working on it but I have put this into rc.local to get mjpg-streamer to start up on booting my pi zero w:
/root//mjpg-streamer/mjpg-streamer-experimental/mjpg_streamer -o "/usr/local/lib/mjpg-streamer/output_http.so -p 8090 -w ./www" -i "/usr/local/lib/mjpg-streamer/input_raspicam.so -x 1280 -y 720 -fps 15"
and in octoprint used the following:
http://LOCAL.LAN.IP.ADDRESS:8090/?action=stream for stream
http://LOCAL.LAN.IP.ADDRESS:8090/?action=snapshot for snapshot

I am not skilled enough to get it to work using localhost or 127.0.0.1, so if my IP address changes it breaks!
If anyone knows how to do this more elegantly feel free to chip in!

MichaIng, 2019-02-27:
@Win-doze 
Instead of `rc.local`, as this runs as a daemon (right) we should create a separate systemd unit that is started by DietPi-Services right before OctoPrint.

Hmm strange that localhost or loopback IP does not work in OctoPrint to access mjpg_streamer. If we don't find a way, we could add a StartExecPre job to the mjpg_streamer systemd unit that updates the IP in OctoPrint settings. But I need to know where exactly this can be done.

I will update the topics header to mjpg_streamer feature request.

soemarko, 2019-08-20:
I've sort of cobble together a bunch of ideas to solve this for myself. Similar to @Win-doze, I added `/root/mjpg-streamer/mjpg-streamer-experimental/mjpg_streamer -i "/root/mjpg-streamer/mjpg-streamer-experimental/input_uvc.so" -o "/root/mjpg-streamer/mjpg-streamer-experimental/output_http.so -p 5001"` to `rc.local` -- let me know where should I put this instead, but this works for now.

For the URL part, I've locked down the ip address from the router side for my pi as it's running pi-hole as well, but I don't even use it, I just use the local name as the url, so for me, the streaming URL is `http://printerpi.local:5001/?action=stream` (also, I can confirm neither 127.0.0.1 nor localhost works).

Hopefully DietPi will be able to natively support this soon.

MichaIng, 2019-09-02:
@soemarko 
Thanks for sharing. I think the localhost/127.0.0.1 issue is something we need to forward to mjpg-streamer devs. Did anyone try to access the stream via browser from the Pi itself, using either `localhost:5001` or `127.0.0.1:5001`? E.g. Chromium should be able to view it.
Perhaps there is some settings switch that disables local access by default, as it is expected to be an overhead to access a stream that is produces by a local camera, via TCP instead of directly or something?

MichaIng, 2021-02-06:
PR open to implement mjpg-streamer as standalone install option: #4091
On RPi, it will build with RPi camera support by default.
As standalone install it will simply stream to `your.ip:8082?action=stream`, password protected with `dietpi:<globalPassword>`.
If it is installed along with OctoPrint, it will be configured automatically to listen on localhost only and OctoPrint to use it accordingly.

MichaIng, 2021-02-07:
Hmm, I think I found the reason why mjpg-streamer cannot be accessed through OctoPrint via localhost/127.0.0.1: https://community.octoprint.org/t/mjpg-streamer-password-protected-or-bound-to-loopback-ip/30370

The browser console shows that the connection is actually done from the browser directly. So when using localhost, it tries to access itself. So this implies the fundamental problem that using mjpg-streamer in such setup does only work, when it is exposed to the network where you access OctoPrint from, i.e. for remote connections, also the stream needs to be accessible remotely. Together with the fact that browsers deny to access embedded URLs with basic authentication like `http://user:password@domain.org`, one cannot even password-protect the webcam stream.

That's a real privacy concern when setting up such by default. At least we would need to warn users about it, when setting up `mjpg-streamer` that way, which means that one should not forward/block access to the port (`8082` by default) from www, to not expose the stream to the world, but consequently won't be able to access it remotely either. I wonder how OctoPi sets this up.

MichaIng, 2021-02-07:
Okay, it's clear now, even the web UI states this fact (info below the settings input box for the webcam URL). I think the best we can do then is to by default have mjpg-streamer bind to the local IP, so that it is at least not accessible from www, when no NAT/firewall is in place. OctoPrint remote connections, if required, would need to be done via VPN then. I mean one can manually remove the IP binding, but the whole world can then freely access that webcam stream, which must be clear.

Joulinar, 2021-02-07:
but this still make your webcam available to everyone who at least knows how to type `/webcam/` inside a browser. Would it be an option to protect thinks using `.htaccess`. Probably a stupid idea

MichaIng, 2021-02-07:
> but this still make your webcam available to everyone who at least knows how to type `/webcam/` inside a browser.

Exactly, at least I couldn't find any sort of authentication on it. But I didn't actually flash OctoPi to simply test it 😉.

> Would it be an option to protect thinks using `.htaccess`. Probably a stupid idea

`.htaccess` is an Apache2 feature, and neither HAProxy nor mjpg-streamer nor OctoPrint, Nginx or Lighttpd will understand it. I never really used HAProxy, but I'd guess it has own authentication features that could be used. But actually, I'm not keen to setup all this by default. If there is no simpler way, currently I would keep the stream unprotected and simply warn users on install about this fact. In 99% of cases, port 8082 is not accessible remotely, otherwise a firewall can be used. And if remote access is required, I'd leave it to the user how to setup it best for the personal needs. A VPN is a secure option we could document in the docs.

MichaIng, 2021-02-07:
POST requests are actually possible to access the camera stream: `curl -X POST http://127.0.0.1:8082/stream` (it of course cannot print it to console, but can be accessed like that)
So it is principally possible to embed the camera stream via pure local 127.0.0.1 POST request done by the OctoPrint web UI itself, not the browser. I didn't test HTTP authentication with this, but it should then work as well since the browser (which currently blocks it) is not involved. I think that would be a great enhancement, but no idea how much effort it would be.

closed 2021-02-28 by MichaIng