MichaIng/DietPi #855: DietPi-Software | CloudPrint

Fourdee, 2017-04-04:
> http://dietpi.com/phpbb/viewtopic.php?f=12&t=1624#p6617

Invictaz, 2017-04-04:
If it works on the Zero / Zero W would be great. Hopefully also with HTTPS and low memory footprint.

Fourdee, 2017-04-04:
Notes:
- Stage 2 for test: https://davesteele.github.io/raspberrypi/2016/04/23/raspberry-pi-cloudprint/
user = root
pw = dietpi

Fourdee, 2017-04-04:
- 🈴 ARM64:
```
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA256

Suite: stable
Codename: cloudprint-jessie
Date: Sat, 27 Aug 2016 16:00:42 UTC
Architectures: amd64 i386 armhf armel
```

Fourdee, 2017-04-04:
Needs testing by user with printer:
```
root@DietPi:~# systemctl status cloudprintd -l
● cloudprintd.service - Google Cloud Print proxy service
   Loaded: loaded (/lib/systemd/system/cloudprintd.service; disabled)
   Active: inactive (dead)

Apr 04 14:40:07 DietPi systemd[1]: Starting Google Cloud Print proxy service...
Apr 04 14:40:07 DietPi systemd[1]: Started Google Cloud Print proxy service.
Apr 04 14:40:08 DietPi cloudprintd[6426]: No printers found
```

Fourdee, 2017-04-05:
Confirmed working, thanks Sciamano  :
http://dietpi.com/phpbb/viewtopic.php?f=12&t=1624&p=6644#p6636

Fourdee, 2017-04-05:
Completed.

Fourdee, 2017-04-06:
cloudprint package is available in Jessie repos, investigate and convert from Dave Steele's repo if functional:

Fourdee, 2017-04-08:
Jessie repo versions:
```
root@DietPi:~# dpkg -l | grep cloudprint
ii  cloudprint                     0.11-5                       all          Google Cloud Print proxy
ii  cloudprint-service             0.11-5                       all          provide a service for sharing printers on Google Cloud Print
```

Fourdee, 2017-04-08:
Ok, so we'll use Dave Steele's repo for Jessie. Stretch will use Debian repos for install and support ARM64.

Fourdee, 2017-04-08:
Completed.

Fourdee, 2017-04-08:
Reference, Stretch support: https://github.com/Fourdee/DietPi/issues/855#issuecomment-292712002

closed 2017-04-08 by Fourdee