MichaIng/DietPi #856: Pi zero W hotspot instructions

9H5G, 2017-04-04:
I tried the software install for wireless hotspot but of course there's no Ethernet. Is there an alternative method please. If so, please update software web page. The 😁

Fourdee, 2017-04-04:
@9H5G 

Our WiFi hotspot install is designed for internet routing and requires:
- 1x WiFi
- 1x Ethernet

If you want a hotspot without network routing, you'll need to manually install it and set it up as needed.

9H5G, 2017-04-04:
Ok. Thx. Will your install work as a local router if the ethernet is
removed after installation?

Fourdee, 2017-04-04:
> Ok. Thx. Will your install work as a local router if the ethernet is
removed after installation?

@9H5G 

Should be fine, give it a whirl :)

Fourdee, 2017-06-17:
No updates/responses, assuming resolved. Please reopen if required.

closed 2017-06-17 by Fourdee