MichaIng/DietPi #859: DietPi-Config | Option to rotate RPi LCD touchscreen

midwan, 2017-04-05:
It would be great if we had the option to rotate the RPI LCD (touch)screen.
Some cases require that you flip them 180 degrees, and that can be accomplished by entering the following line in /boot/config.txt:

lcd_rotate=2

Note that this takes care of rotating the touch input as well, in contrast to:

display_rotate = 2

which would rotate the display, but not the touch input layout.

Fourdee, 2017-04-08:
@midwan 

Thanks 👍 

We'll get this added for v149.

midwan, 2017-04-08:
Great! :)

Fourdee, 2017-04-08:
@midwan 
On the RPi Touchscreen, when using ```lcd_rotate=1``` or ```3```, do also need to flip the ```framebuffer_width``` and ```framebuffer_height``` values to get full screen (as per HDMI and ```display_rotate```)?

midwan, 2017-04-08:
@Fourdee I never had to use those values, so I'm not sure. I can run a test though a bit later and let you know.

I usually have lcd_rotate=2 to flip the screen 180 degrees, which is required for several cases ;)

Fourdee, 2017-04-08:
@midwan 

>  never had to use those values, so I'm not sure. I can run a test though a bit later and let you know.

Thanks, would be a great help 👍

midwan, 2017-04-10:
@Fourdee 

Had a chance to test this:
lcd_rotate only has an effect if specified with value "2" (rotate 180 degrees).
For other scenarios (e.g. Portrait mode) we have to use display_rotate with any extra options.
Relevant thread: https://www.raspberrypi.org/forums/viewtopic.php?f=108&t=120793

Fourdee, 2017-04-10:
@midwan 

Legend, thank you 👍

Fourdee, 2017-04-10:
Completed.

closed 2017-04-10 by Fourdee