MichaIng/DietPi #862: How do img file flash to eMMC of NanoPi Neo Air?

aesirteam, 2017-04-07:
How do img file flash to eMMC of NanoPi Neo Air?

Fourdee, 2017-04-07:
@aesirteam 

Install DietPi on the eMMC (you must have DietPi up and running on a SDcard prior to this):
```
#7Zip prereqs
apt-get install -y p7zip

wget 'http://dietpi.com/downloads/images/DietPi_NanoPiNEOAir-armv7-(Jessie).7z' -O image.7z
p7zip -d image.7z

#Write to EMMC
# - Following assumes you have no other files in this directory with name 'DietPi_v*.img' 
#   (other than the one we downloaded above)
#   Run BLKID to check index of EMMC (should be entry for /dev/mmcblk1, but may be /dev/mmcblk2 etc). Change /dev/mmcblk1 in command below accordingly
dd if="$(ls | grep -m1 'DietPi_v'*'.img')" of=/dev/mmcblk1 

#Power off, unplug SD card, power on.
```

aesirteam, 2017-04-10:
Step 1: Copy  .img file to SD  /opt path;
Step 2: Power On, login, fdisk eMMC
  fdisk /dev/mmcblk2
  d->n->p->...->w
  mkfs.fat /dev/mmcblk2p1
Step 3: blkid /dev/mmcblk2
Step 4:  dd if="$(ls | grep -m1 'DietPi_v'*'.img')" of=/dev/mmcblk2

but tty console output error:

Starting kernel ...

Uncompressing Linux... done, booting the kernel.
Loading, please wait...
Begin: Loading essential drivers ... done.
Begin: Running /scripts/init-premount ... done.
Begin: Mounting root file system ... Begin: Running /scripts/local-top ... done.
Begin: Running /scripts/local-premount ... done.
Begin: Will now check root file system ... fsck from util-linux 2.25.2
[/sbin/fsck.ext4 (1) -- /dev/mmcblk2p2] fsck.ext4 -a -C0 /dev/mmcblk2p2 
/dev/mmcblk2p2 contains a file system with errors, check forced.
/dev/mmcblk2p2: Inode 1223 has an invalid extent node (blk 52271, lblk 0)

/dev/mmcblk2p2: UNEXPECTED INCONSISTENCY; RUN fsck MANUALLY.
        (i.e., without -a or -p options)
fsck exited with status code 4
done.
Failure: File system check of the root filesystem failed
The root filesystem on /dev/mmcblk2p2 requires a manual fsck
Rebooting automatically due to panic= boot argument

aesirteam, 2017-04-10:
According to your instructions， operation succeeded!!! 
thank you very mach!

closed 2017-04-10 by aesirteam