MichaIng/DietPi #865: DietPi-Software | SabNZBd

Fourdee, 2017-04-07:
> https://sabnzbd.org/

https://github.com/Fourdee/DietPi/issues/731#issue-202979307

Invictaz, 2017-04-08:
Don't forget to include SABYenc, a module for CPU limited devices like the Raspberry Pi. It's new in the V2.0 RC version.

https://sabnzbd.org/wiki/installation/sabyenc.html

Fourdee, 2017-04-11:
Notes:
- Jessie apt = ```sabnzbdplus (0.7.18-1)```
-  http://192.168.0.105:8080/sabnzbd
- ```.sabnzbd/sabnzbd.ini``` https://sabnzbd.org/wiki/advanced/command-line-parameters

Invictaz, 2017-04-12:
@Fourdee isn't 0.7 a bit old?

Fourdee, 2017-04-13:
> One question isn't 0.7 a bit old?

@Invictaz 

Yep, we wont be using apt-get for this one, we'll pull in from git (v2.0.0):
https://github.com/sabnzbd/sabnzbd/releases

Fourdee, 2017-04-13:
Completed.

closed 2017-04-13 by Fourdee

Invictaz, 2017-05-06:
@Fourdee It installed now. Working great.

How to update to 2.0.1 RC1? It installs 2.0.0

There isnt a setting in Sabnzbd to update and 2.0.0 seems to be hardcoded?

Fourdee, 2017-05-06:
> There isnt a setting in Sabnzbd to update and 2.0.0 seems to be hardcoded?

@Invictaz 

Yes its hard coded to install 2.0.0 

RC = release candidate. Potentially unstable.

We'll stick with 2.0.0 for now (stable), and update our install code when a future stable version of Sabnzbd is released.

Invictaz, 2017-05-06:
Is there any way I can install it myself? If I really want to? Just unzip the contents?
Especially on Kodi RC's give vast improvements.

Invictaz, 2017-06-01:
@Fourdee 2.01 final was released.

Can you update it please?

https://github.com/sabnzbd/sabnzbd/releases/tag/2.0.1

Fourdee, 2017-06-01:
@Invictaz 

Thanks for heads up 👍 

Done. Test install fine aswell.

hydrouk, 2017-07-31:
2.1.0 is out, please update

Fourdee, 2017-08-06:
@hydrouk 

Thanks 👍 

Done for v155. This won't effect existing installs of Sabnzbd, only new installs.

To try a manual upgrade (via reinstall), backup system 1st and run:
```
dietpi-software reinstall 139
```

Invictaz, 2017-08-09:
@Fourdee Why 'reinstall 139' and not the newest version? Or is 139 the number of SabNZBd in dietpi-software?

k3anton, 2017-08-09:
@Invictaz The number 139 refers to the software numbering within DietPi and has nothing to do with versions of that specific software.

hydrouk, 2017-08-09:
I have run the reinstall a couple of times but it is still installing version 2.0.1

k3anton, 2017-08-09:
@hydrouk I can also confirm what you are saying.

hydrouk, 2017-08-09:
Ahh i just noticed it is done for v155, current version is v154, my bad @Fourdee

Invictaz, 2017-08-10:
Better do 2.2.0 instead of 2.1.0 for v155. 2.2.0 is now in it's last release candidate. Wont be long until final. 
Has some major improvements.

Fourdee, 2017-08-10:
@Invictaz 

2.2.0 is currently RC2, we'll stick with stable 2.1.0 for now.

Invictaz, 2017-08-10:
Yeah but by the time 155 comes I mean. Or is that not too far away?

Invictaz, 2017-08-28:
There it is 

https://sabnzbd.org/downloads?v=2.2.1

hydrouk, 2017-09-26:
Update Available! 2.3.0

Just so you know :) Thanks @Fourdee

Invictaz, 2017-10-29:
2.3.1 released. Thanks @Fourdee

clarky0000, 2018-03-22:
Im getting this in sabnzbd GUI and .rar archives are not being unzipped...

Your UNRAR version is 0.00, we recommend version 5.00 or higher.<br />

When i try to install unrar-free
-------------------------------
root@DietPi:~# apt-get install unrar-free
Reading package lists... Done
Building dependency tree
Reading state information... Done
unrar-free is already the newest version (1:0.0.1+cvs20140707-1).
0 upgraded, 0 newly installed, 0 to remove and 20 not upgraded.
--------------------------------

Help me out - what am I missing?

Fourdee, 2018-03-22:
@clarky0000 

Hi, ```unrar``` (non-free) is not available in the Raspbian repos. Try manually install it from Debian:
```
G_AGP unrar-free

wget http://ftp.us.debian.org/debian/pool/non-free/u/unrar-nonfree/unrar_5.3.2-1+deb9u1_armhf.deb -O package.deb
dpkg -i package.deb
```

clarky0000, 2018-04-12:
Thanks for you help on that. Im also getting these messages in sabnzbd:

_yenc module... NOT found!
SABYenc disabled: no correct version found! (Found v3.3.2, expecting v3.3.1)

Can I disregard those?

Fourdee, 2018-04-12:
https://sabnzbd.org/wiki/installation/sabyenc.html

@clarky0000 

Seems your current install requires a specific version, run the following command to achieve it:
```
pip install sabyenc==3.3.1
dietpi-services restart
```

verify with:
```
python -c "import sabyenc ; print sabyenc.__version__ "
```

clarky0000, 2018-04-13:
Thats fixed it - thanks! 

root@DietPi:~# python -c "import sabyenc ; print sabyenc.__version__ "
3.3.1

I was just seeing it in the UI.  Thanks for your help - much appreciated. 

Final question - I promise - I can see there is a newer version of Sabnzbd available. Can you tell me the best command to perform the upgrade?

Cheers,
Clarky

Fourdee, 2018-04-13:
@clarky0000 

> Final question - I promise - I can see there is a newer version of Sabnzbd available. Can you tell me the best command to perform the upgrade?

We only provide installs for stable releases. RC is release candidate (potentially unstable). Once 2.3.3 becomes stable release, we'll update and patch via ```dietpi-update```.