MichaIng/DietPi #866: DietPi-Software | NAA daemon Odroid C2

lucser68, 2017-04-08:
On ODROID C2 with DietPi v147 diet-software NAA demon not available.

Fourdee, 2017-04-08:
@lucser68 

Hi,

NAA Daemon is not available for ARM64 devices at this time, as we lack the ARM64 binaries offered by HQPlayer devs.
https://community.roonlabs.com/t/hqp-naa-raspberry-pi/6140/180?u=dan_knight

I did try running ARMv7 binaries, but ended up in a messy dependency war:
https://github.com/Fourdee/DietPi/issues/674#issuecomment-269810358

lucser68, 2017-04-08:
Thank you for information. At the moment is running in a mixed environment. On Odroid C2 i have installed dietPi v147, then add arch armhf but trying to install NAA from Signalyst repository ended as you said in a list of broken dependencies. Next I've changed jessie to stretch in apt list and update, upgrade...... install NAA. for the moment is running but in future I think that any update could end in a problem and then I need to start over again. Hopefully the NAA binaries for ARM64 will be released...
Thank you again

closed 2017-04-08 by lucser68