MichaIng/DietPi #869: DietPi-Software | FFmpeg: Enable HW encoding (RPi)

Fourdee, 2017-04-09:
Requires binary compiles:
https://github.com/Fourdee/DietPi/issues/475#issuecomment-292734524

Programs such as Emby, should benefit vastly from this.

Fourdee, 2017-04-09:
RPi devices will now install our compiled FFmpeg binary with OpenMAX enabled, including Stretch.

Fourdee, 2017-04-09:
Tests:
- ARMv6
- ARMv7
- Stretch

Fourdee, 2017-04-10:
CC with x264 + FDK-aac, fail.
```
cd /root
git clone https://github.com/raspberrypi/tools.git --depth=1

export CCPREFIX="/root/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin/arm-linux-gnueabihf-"

apt-get install git checkinstall build-essential libomxil-bellagio-dev libmp3lame-dev libvorbis-dev libtheora-dev libspeex-dev yasm pkg-config libflac-dev libopenjpeg-dev libx264-dev -y
git clone https://github.com/FFmpeg/FFmpeg.git --depth=1
cd FFmpeg

#LIBFDK-AAC
apt-get install -y libfftw3-dev pkg-config autoconf automake libtool libtool-bin unzip
git clone https://github.com/mstorsjo/fdk-aac.git --depth=1
cd fdk-aac
./autogen.sh
./configure --enable-shared --enable-static --host=arm-unknown-linux-gnueabi --prefix=${CCPREFIX}
make -j 9
cd ..

#x264
git clone git://git.videolan.org/x264 --depth=1
cd x264
./configure --host=arm-linux --cross-prefix=${CCPREFIX} --enable-static --disable-asm --extra-cflags='-march=armv6' --extra-ldflags='-march=armv6'
make
cd ..

#OpenMax headers
wget https://www.khronos.org/registry/OpenMAX-IL/api/1.1.2/OpenMAX_IL_1_1_2_Header.zip -O package.zip
unzip package.zip -d IL
rm package.zip

./configure --arch=armel --target-os=linux --enable-cross-compile --pkg_config=$(which pkg-config) \
--extra-cflags="-Ifdk-aac/" --extra-ldflags="-ldl -Lfdk-aac/" \
--extra-cflags="-Ix264/" --extra-ldflags="-ldl -Lx264/"  --extra-cflags="-IIL/" --extra-ldflags="-ldl -LIL/" --cross-prefix=${CCPREFIX} --enable-gpl --enable-libx264 --enable-pthreads --enable-nonfree --enable-omx --enable-omx-rpi --enable-libmp3lame --enable-libfdk-aac
```
```
ERROR: libfdk_aac not found
```

Fourdee, 2017-04-16:
```libmp3lame libfdk-aac libx264```
```
root@DietPi:~# ffmpeg
ffmpeg version git-2017-04-14-3f1c527 Copyright (c) 2000-2017 the FFmpeg developers
  built with gcc 4.9.2 (Raspbian 4.9.2-10)
  configuration: --enable-static --enable-gpl --enable-libx264 --enable-pthreads --enable-nonfree --enable-omx --enable-omx-rpi --enable-libfdk-aac --enable-libmp3lame
  libavutil      55. 61.100 / 55. 61.100
  libavcodec     57. 92.100 / 57. 92.100
  libavformat    57. 72.101 / 57. 72.101
  libavdevice    57.  7.100 / 57.  7.100
  libavfilter     6. 84.101 /  6. 84.101
  libswscale      4.  7.101 /  4.  7.101
  libswresample   2.  8.100 /  2.  8.100
  libpostproc    54.  6.100 / 54.  6.100
```

Completed.

Fourdee, 2017-04-17:
Needs recompile with prefix ```--prefix=/usr```, else ```/usr/local/lib``` and ```export LD_LIBARY_PATH``` required.

WolfganP, 2017-04-18:
@Fourdee could you please share how can we install this custom version for testing? Thx in advance.

Fourdee, 2017-04-18:
@WolfganP 
Once re-compiled (should be done tomorrow), testing branch install can be used. I'll let you know when its ready + instructions.

Fourdee, 2017-04-19:
@WolfganP 

Ready for testing, you could install the updated FFmpeg binaries with simply:
```
INSTALL_URL_ADDRESS='http://dietpi.com/downloads/binaries/rpi/ffmpeg_rpi.7z'
wget "$INSTALL_URL_ADDRESS" -O package.7z
7z x -y package.7z -offmpeg_rpi
dpkg -i ffmpeg_rpi/*.deb
rm -R ffmpeg_rpi
rm package.7z
```

Fourdee, 2017-04-21:
Marking as completed.

@WolfganP 
If you run any tests, please let me know if any issues.

WolfganP, 2017-04-22:
Will do @Fourdee

MichaIng, 2018-08-02:
@Fourdee 
Maybe it's time to compile a new version? Even Stretch repo is some subversions further via security patches: https://packages.debian.org/stretch/ffmpeg

V4.0.2 available on Buster repo (for testing) and GitHub sources of course: https://github.com/FFmpeg/FFmpeg/releases

Fourdee, 2018-08-03:
@MichaIng 

Yep, agree.

Heres the old compile code I used, here for reference. I'll take a look at this when I can:
```
#FFmpeg ---------------------------------------------
#RPi tool chain
# cd /root
# git clone https://github.com/raspberrypi/tools.git --depth=1

# export CCPREFIX="/root/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/bin/arm-linux-gnueabihf-"

apt-get install -y git checkinstall build-essential libomxil-bellagio-dev libmp3lame-dev libvorbis-dev libtheora-dev libspeex-dev yasm pkg-config libflac-dev libopenjpeg-dev -y
git clone https://github.com/FFmpeg/FFmpeg.git --depth=1
cd FFmpeg

#LIBFDK-AAC
apt-get install -y libfftw3-dev pkg-config autoconf automake libtool libtool-bin unzip
git clone https://github.com/mstorsjo/fdk-aac.git --depth=1
cd fdk-aac
./autogen.sh
./configure --enable-shared --prefix=/usr
make -j $(nproc --all)
checkinstall --nodoc --fstrans=yes #libfdk-aac
mv *.deb /root/
cd ..

#lame
wget http://downloads.sourceforge.net/project/lame/lame/3.99/lame-3.99.tar.gz
tar xzvf lame-3.99.tar.gz
cd lame-3.99
./configure --enable-shared --prefix=/usr #--enable-static --host=arm-linux --prefix=${CCPREFIX}
make -j $(nproc --all)
mkdir -p /usr/share/doc/lame
checkinstall --nodoc --fstrans=yes #libmp3lame
mv *.deb /root/
cd ..

#x264
git clone git://git.videolan.org/x264 --depth=1
cd x264
./configure --disable-asm --enable-shared --prefix=/usr #--host=arm-linux --cross-prefix=${CCPREFIX} --enable-static --disable-asm --extra-cflags='-march=armv6' --extra-ldflags='-march=armv6'
make -j $(nproc --all)
checkinstall --nodoc --fstrans=yes #libx264
mv *.deb /root/
cd ..

#OpenMax headers
wget https://www.khronos.org/registry/OpenMAX-IL/api/1.1.2/OpenMAX_IL_1_1_2_Header.zip -O package.zip
unzip package.zip -d IL
rm package.zip

#REQUEST:  --enable -libass --enable-libx265

./configure --prefix=/usr --enable-static --enable-gpl --enable-libx264 --enable-pthreads --enable-nonfree --enable-omx --enable-omx-rpi --enable-libfdk-aac --enable-libmp3lame
make -j $(nproc --all)

mkdir -p /usr/share/ffmpeg
checkinstall --nodoc --install=no --fstrans=yes
mv *.deb /root/

#Install
#export LD_LIBRARY_PATH=/lib:/usr/lib:/usr/local/lib
apt-get install libx264-142 libmp3lame0

#FFmpeg ---------------------------------------------
```

MichaIng, 2018-12-29:
I guess Debian armhf packages do not (fully) support RPi?
E.g. https://packages.debian.org/buster/libx265-165

MichaIng, 2019-05-07:
The ffmpeg package from archive.raspberrypi.org repo got HW acceleration:
```
ffmpeg (7:3.2.12-1~deb9u1+rpt1) stretch; urgency=medium

  [ Dave Stevenson ]
  * avcodec/omx: Fix handling of fragmented buffers
  [ Serge Schneider ]
  * Add libraspberrypi0 dependency
  * Set --arch=armhf
  [ Arindam Chaudhuri ]
  * Enable Raspberry Pi HW acceleration

 -- Serge Schneider <serge@raspberrypi.org>  Fri, 25 Jan 2019 09:56:06 +0000
```

Same for the package from Buster branch:
```
ffmpeg (7:4.1-1+rpt1) buster; urgency=medium

  [ Dave Stevenson ]
  * avcodec/omx: Fix handling of fragmented buffers
  [ Serge Schneider ]
  * Add libraspberrypi0 dependency
  * Set --arch=armhf
  [ Arindam Chaudhuri ]
  * Enable Raspberry Pi HW acceleration

 -- Serge Schneider <serge@raspberrypi.org>  Wed, 06 Feb 2019 14:17:15 +0000
```

These packages are preferred by APT over the ones from Raspbian repo (identical to Debian repo), so I suggest we revert to APT install on RPi as well.

However some quick test would be great, including x264/x265 if possible.

MichaIng, 2019-05-30:
Included and merged: https://github.com/MichaIng/DietPi/pull/2875

Actually since the repo version is newer, an APT upgrade must have upgraded the package already (present since end of January), so it's just to avoid installing our own compiled package first.

closed 2019-07-03 by MichaIng