MichaIng/DietPi #870: DietPi-Software | Docker

Fourdee, 2017-04-09:
> http://dietpi.com/phpbb/viewtopic.php?f=12&t=640&p=6708#p6663

telamon, 2017-05-01:
DietPi would've been a nice dist if it had better docker-support as the availability of containers on the hub really does outperform dietpi's custom menus and software install scripts.
 
I tried using hypriot's docker packages but my Xu4 + dietpi kernel-panics whenever i try to start a container that uses --net=host priviledge. Had a few kernel-panics related to AuFS it seems as well but sadly I never get the chance to inspect the errors since the logs never get flushed to disk and are gone without a trace after reboot. :(  :+1:  for docker support.

bduisenov, 2017-07-24:
Hi @Fourdee, do you have this in the roadmap already?

Fourdee, 2017-07-25:
@bduisenov 

Do now, i'll pop this into v156 milestone. Will take a look during that phase.

bartaspoz, 2017-09-23:
v156 released and I see that you moved docker to v157 milestones. Did you encountered problems or just this is so time consuming? :)

Fourdee, 2017-09-23:
@bduisenov 

> Did you encountered problems or just this is so time consuming? :)

Lack of available time from my end, and, we needed to release v156 to resolve bugs which would effect our users.

I will get round to it, although, no ETA yet.

telamon, 2017-09-25:
Hey, a little status update. I recently upgraded my diet pi version and also purged out the docker provided by hypriot and replaced it with the official installation via get.docker.com which now contains arm distributions. And so far the old kernel panic issue has been resolved. I cannot reproduce it ant more.

Fourdee, 2017-10-14:
Added in by @techdabbler for v157 (seems he deleted his account) 👍 

Marking as completed.

closed 2017-10-14 by Fourdee

bartaspoz, 2017-10-21:
Hi, firstly, I couldn't find Docker in gui in optimized software list. Found ID and tried to install by:
dietpi-software install 162

And recieved bellow errors:
# Executing docker install script, commit: 490beaa 
Either your platform is not easily detectable or is not supported by this 
installer script. 

And next:
#               processing | D i e t  P i             #

Failed to stop docker.service: Unit docker.service not loaded.
sed: can't read /lib/systemd/system/docker.service: No such file or directory

Test performed on: Odroid C2

bartaspoz, 2017-10-22:
A llittle followup. Managed to install docker manually from repo, but containers don't want to work, error during start as below:

could not create session key: function not implemented

Is there anything we can do?

Test platform: Odroid C2

bduisenov, 2017-10-22:
Hi @bartaspoz, I'm facing the same issue with docker installation. Neither can see the docker item in the list of dietpi-software

techdabbler, 2017-10-22:
@bartaspoz - This is as expected and designed.  The official docker script doesn't support ARM8 so dietpi does not display the menu item on ARM8 devices and if you force install it will fail. It appears the Odroid C2 is ARM8.

@bduisenov ate you also on ARM8 or on a different arch?

bduisenov, 2017-10-22:
@techdabbler Armv8-A, Quad-Core ARM Cortex A53 64-Bit Processor

techdabbler, 2017-10-22:
@bduisenov - That makes sense then too. Def it is due to docker not officially supporting raspbian and arm8. I didn't do anything fancy but package up the official installer.  If anyone out there knows steps to get docker to work on raspbian arm8 I can add the steps to dietpi-software installer.

bduisenov, 2017-10-23:
@techdabbler got it, thanks for the explanation.

bartaspoz, 2017-10-23:
People say that probably enabling keychain functionality in the kernel is needed in order to work (error: could not create session key: function not implemented). The keychain is enabled by default on 3.18+ Kernel, and latest available for DietPI is 3.16+. I'm not an linux expert, can you tell us if updating kernel to 3.18+ is possible on DietPI?

Additionally, do you know why I'm receiving this strange notification?:
dietpi-update    = Run now to update DietPi (from V157 to V156).

bartaspoz, 2017-10-23:
@techdabbler you're right. But there is a possibility to install armhf on arm64, which I have done. You have to add new repo as below
deb [arch=armhf] https://download.docker.com/linux/debian jessie stable

And Install docker by:
apt-get install docker-ce

And it installs, daemon starts without any problems, but containers won't work :( Probably because of the session keychain, as I wrote before. Here's link to identical problem that someone had:
https://github.com/moby/moby/issues/33976

techdabbler, 2017-10-23:
@bartaspoz - This starts to sound like it is outside my wheel house.  I don't know anything about the kernel or how to replace it or update it in dietpi.  

Nor do I know about the weird v157 to v156 message you are getting.  

I'd recommend maybe you open up another current issue with arm8 and docker and see if anyone else is able to help you.  Continuing on this thread since it is marked closed seems less likely someone else might see it and be able to help you. 

Just my 2 cents worth.

j2l, 2019-01-23:
> 
> 
> Anyone installed Docker on ROCK_64? The chip is a _Rockchip RK3328 Quad-Core ARM Cortex A53 64-Bit Processor_

Docker is available from the dietpi-software menu, but I didn't find any working orchestrator/router.

Did someone successfully run one?

lpninja, 2019-01-27:
@j2l yes we ran docker on it.