MichaIng/DietPi #871: Install fails after first boot with timeout from server

lavadrop, 2017-04-10:
I downloaded the latest Amiberry image and restored it to a brand new Samsung EVO Select MicroSD. I put it on my RPi 3 and turned it on. The script went fine until the first reboot; after rebooting, the script timed out connecting to the ox.ac.uk server after 20 tries and halted. I rebooted it from the CLI and same thing happened. 
Here's the error:

/DietPi/dietpi/dietpi-software: line 10586: /DietPi/dietpi/func/check_connection: No such file or directory
[Info] Testing connection to http://mirror.ox.ac.uk/sites/archive.raspian.org/archive/raspbian
[Info] Attempt (1/20), please wait...

Fourdee, 2017-04-10:
@lavadrop 

Thanks for the report 👍  i'll see if we can replicate here.

Fourdee, 2017-04-10:
@lavadrop 

Test install fine.

Did you connect ethernet prior to powering on? If so, could be a slow dhcp/connection failing before the time runs out.

If the time runs out, you can retry launching the installer by running:
```
dietpi-software
```

If problems persist, try changing the APT mirror:
```
dietpi-config
```
- Network Options: NAS/Misc
- Apt Mirror

lavadrop, 2017-04-10:
The RPi is connected before powering up. The timeouts are instant. After it drops me back to the shell, I can ping ox.ac.uk from the Pi. I'll try changing the mirror later tonight.

lavadrop, 2017-04-11:
I tried 2 different cards several times, changing the txt files for wifi and locale and 3-4 times the script kept failing, either by timing out or by failing to expand the file system. I'm on a mac so I used ApplePi-Baker; apparently the program is failing to write directories and/or correct permissions when writing an image compressed with 7zip. I unzipped the Amiberry image and wrote the img with ApplePie-Baker and the issue is gone. The Pi is updating from source as I write this.

Fourdee, 2017-04-11:
@lavadrop 

> changing the txt files for wifi and locale and 3-4 times the script kept failing,

Thanks, under WiFi with slow establishing connection, 20 seconds may not be enough. We do have an option that will force an endless wait in ```/boot/dietpi.txt```:
```
boot_wait_for_network=2
```

This will prevent slow establishing connections from failing the 20 second test.

lavadrop, 2017-04-11:
Thanks, will have that in mind if I use wifi instead

Fourdee, 2017-04-13:
@lavadrop 

I'll mark this as resolved. Please reopen if required.

closed 2017-04-13 by Fourdee

DannyCork, 2017-12-14:
Just to mention that I encountered the exact same issue on a Mac with PiBaker. I unzipped the AmiBerry zip first and the problem went away.

thanks @lavadrop

jedmao, 2018-04-06:
@lavadrop I can't find your email address anywhere. I have a question for you. Will you please contact me at jedmao@outlook.com?