MichaIng/DietPi #880: Force reboot after Hostname change

9H5G, 2017-04-12:
Its possible to change the hostname in -config and go straight on to install LAMP/LASP etc. Theresult can be this: make-ssl-cert: Could not get FQDN, using "AG".
make-ssl-cert: You may want to fix your /etc/hosts and/or DNS setup and run
make-ssl-cert: make-ssl-cert generate-default-snakeoil --force-overwrite
make-ssl-cert: again.

Which is undesireable for a beginner (or many of us who won't know how to fix it) :)

Solution as title? :)

Fourdee, 2017-04-13:
@9H5G 

> Its possible to change the hostname in -config and go straight on to install LAMP/LASP etc. Theresult can be this: make-ssl-cert: Could not get FQDN, using "AG".

Thanks for the report and suggestion. The reboot prompt is disabled on 1st run install in ```dietpi-config```, so this could be an issue.
I'll add a reboot prompt straight after changing hostname. It will be optional yes/no choice, but i'll make it highly recommended.

Fourdee, 2017-04-13:
[image]

Fourdee, 2017-04-13:
Completed.

closed 2017-04-13 by Fourdee