MichaIng/DietPi #885: DietPi-Config | Add support for VC1 MPEG2 RPi key entry

Fourdee, 2017-04-16:
> https://twitter.com/W00fer/status/853414918586028032

Fourdee, 2017-04-29:
```
decode_MPG2=0x46de3941
decode_WVC1=0xf402b3ea
```

Fourdee, 2017-04-29:
Completed.

pzman89, 2017-06-01:
I realize this is closed and if you prefer me to go to dietpi forums I can but when I enter my mpg codec (I don't have a wvc1),  I still get "codec disabled" when running the "vcgencmd codec_enabled MPG2" command.  I triple checked with raspberrypi.com to make sure I have them the right serial and they said yes.  Any thoughts on what might be the issue?  The ramdisk DietPi/Config.txt definitely has the serial in it.

Fourdee, 2017-06-01:
@pzman89 

Thanks for the report 👍 

I'll reopen this ticket and see if we can replicate.

Invictaz, 2017-06-01:
For me the WVC1 seems disabled in Movian (self compiled) MPEG2 does work.

pzman89, 2017-06-01:
I was thinking of running a fresh copy of dietpi to see if it gets resolved but let me know if theres something you want me to check before trying.

pzman89, 2017-06-01:
@Fourdee still a no go after a fresh DietPi install.  It doesn't even save my key through the interface, I have to manually enter it in /Dietpi/config.txt.  Even then, it still reports disabled.

Fourdee, 2017-06-02:
@pzman89 

> It doesn't even save my key through the interface, I have to manually enter it in /Dietpi/config.txt. Even then, it still reports disabled.

Thanks, i'll try to replicate.

If its still reporting disabled, might be an issue with current kernel (4.9). I'll check with 4.4.

Fourdee, 2017-06-02:
- Ok, possible bug with adding the entries, not generating the required line in ```config.txt```. I'll get that fixed. 
- Updated RPi kernel/firmware through apt, currently 4.9, same issue with status disabled.

pzman89, 2017-06-02:
@Fourdee You might be right about the kernel.  I installed raspbian, which has kernel 4.4.50 and added the key via config editor in NOOBS and its reporting as enabled.

Fourdee, 2017-06-02:
@pzman89 

> You might be right about the kernel. I installed raspbian, which has kernel 4.4.50 and added the key via config editor in NOOBS and its reporting as enabled.

You beat me to it 👍  just installed Jessie lite to run the same tests lol. I'll see if I can confirm.

Fourdee, 2017-06-02:
Raspian lite:
- 🈯️ ```4.4.50-v7+```
- 🈯️ ```apt-get upgrade``` = ``` 4.9.24-v7+```

Strange.

Fourdee, 2017-06-04:
I'll mark this as closed and finish up with new ticket:
https://github.com/Fourdee/DietPi/issues/996

closed 2017-06-04 by Fourdee